espressif/esp-idf #8488: nvs_flash_init() crashes when debug-without-optimizations is chosen as of release/v4.4 02.03.2022

moefear85, 2022-03-02:
```
entry 0x4008064c
W (430) wifi:age timer already deinit
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x40158184  PS      : 0x00060434  A0      : 0x80084c14  A1      : 0x3ffbaea0  
0x40158184: xPortSetInterruptMaskFromISR at /home/moefear85/Source/esp-idf-v4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:562

A2      : 0x00000001  A3      : 0x15122500  A4      : 0x00060420  A5      : 0x00000000  
A6      : 0x0000cdcd  A7      : 0x3ffbaea0  A8      : 0x80085d0b  A9      : 0x3ffbae60  
A10     : 0x00000000  A11     : 0x15122500  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x3ffbae60  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace:0x40158181:0x3ffbaea00x40084c11:0x3ffbaee0 0x40085158:0x3ffbaf10 0x40084dcc:0x3ffbafb0 0x400d7dce:0x3ffbb010 0x400d7bbd:0x3ffbb0f0 0x400d7c06:0x3ffbb120 0x400d7d3d:0x3ffbb160 0x400e6602:0x3ffbb1a0 0x400e308f:0x3ffbb1e0 0x400dfd0a:0x3ffbb220 0x400dfd28:0x3ffbb260 0x400d8af4:0x3ffbb280 0x400d851b:0x3ffbb3a0 0x4015d74b:0x3ffbb410 
0x40158181: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >* std::__addressof<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) at /home/moefear85/Software/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/move.h:48

0x40084c11: Cache_Flush at /home/moefear85/Source/esp-idf-v4.4/components/esp_rom/include/esp32/rom/cache.h:141

0x40085158: spi_flash_mmap_pages at /home/moefear85/Source/esp-idf-v4.4/components/spi_flash/flash_mmap.c:279

0x40084dcc: spi_flash_mmap at /home/moefear85/Source/esp-idf-v4.4/components/spi_flash/flash_mmap.c:156

0x400d7dce: load_partitions at /home/moefear85/Source/esp-idf-v4.4/components/spi_flash/partition.c:184

0x400d7bbd: ensure_partitions_loaded at /home/moefear85/Source/esp-idf-v4.4/components/spi_flash/partition.c:74 (discriminator 6)

0x400d7c06: esp_partition_find at /home/moefear85/Source/esp-idf-v4.4/components/spi_flash/partition.c:87

0x400d7d3d: esp_partition_find_first at /home/moefear85/Source/esp-idf-v4.4/components/spi_flash/partition.c:140

0x400e6602: nvs::partition_lookup::lookup_nvs_partition(char const*, nvs::NVSPartition**) at /home/moefear85/Source/esp-idf-v4.4/components/nvs_flash/src/nvs_partition_lookup.cpp:14

0x400e308f: nvs::NVSPartitionManager::init_partition(char const*) at /home/moefear85/Source/esp-idf-v4.4/components/nvs_flash/src/nvs_partition_manager.cpp:45

0x400dfd0a: nvs_flash_init_partition at /home/moefear85/Source/esp-idf-v4.4/components/nvs_flash/src/nvs_api.cpp:135

0x400dfd28: nvs_flash_init at /home/moefear85/Source/esp-idf-v4.4/components/nvs_flash/src/nvs_api.cpp:171

0x400d8af4: mesh_setup at /media/moefear85/BACKUP/Projects/Programming/ESP32/Tests_and_Examples/sniffer_mesh/build/../main/common/mesh/mesh.cpp:55

0x400d851b: app_main at /media/moefear85/BACKUP/Projects/Programming/ESP32/Tests_and_Examples/sniffer_mesh/build/../main/main.cpp:21

0x4015d74b: main_task at /home/moefear85/Source/esp-idf-v4.4/components/freertos/port/port_common.c:129

ELF file SHA256: 66e68bd8fe4f0fe1

CPU halted.
```

happens when I choose debug without optimizations. I'm using the latest release/v4.4 branch as of 2.3.2022. My main.cpp is:

```
esp_event_loop_create_default();
mesh_setup();
```

my mesh_setup() function is:
```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
wifi_promiscuous_filter_t wifi_filter={.filter_mask=WIFI_PROMIS_FILTER_MASK_DATA};
wifi_promiscuous_filter_t wifi_ctrl_filter={.filter_mask=WIFI_PROMIS_CTRL_FILTER_MASK_ALL};

ESP_ERROR_CHECK( esp_wifi_init(&cfg)                                               );
ESP_ERROR_CHECK( esp_wifi_start()                                                  );
ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA)                                  ); 

ESP_ERROR_CHECK( esp_wifi_set_promiscuous_filter(&wifi_filter)                     );
//ESP_ERROR_CHECK( esp_wifi_set_promiscuous_ctrl_filter(&wifi_ctrl_filter)         );
ESP_ERROR_CHECK( esp_wifi_set_promiscuous_rx_cb(wifi_sniffer_cb)                   );
ESP_ERROR_CHECK( esp_wifi_set_channel(CHANNEL, WIFI_SECOND_CHAN_NONE)              );
ESP_ERROR_CHECK( esp_wifi_config_80211_tx_rate(WIFI_IF_STA,WIFI_PHY_RATE_MCS7_SGI) );
ESP_ERROR_CHECK( esp_wifi_set_promiscuous(true)                                    );
ESP_ERROR_CHECK( esp_base_mac_addr_get(mac_addr_self)                              );

ESP_ERROR_CHECK( nvs_flash_init()                                                  );
ESP_ERROR_CHECK( nvs_open("MESH",NVS_READWRITE,&my_nvs_handle)                     );
```

igrr, 2022-03-02:
I think this is the same issue as https://github.com/espressif/esp-idf/issues/7835. It has been fixed in master in https://github.com/espressif/esp-idf/commit/8220b2bb34ca27cf5eaa1872ba492dde134558f0, the fix will be backported to release/v4.4. Please subscribe to the linked issue to get notified when the fix is available.

closed 2022-03-02 by igrr

moefear85, 2022-03-02:
ok I made the assumption that the fix was already included in the latest release/v4.4 . I don't mind upgrading to latest master or to commit 8220b2b in particular, but then arduino-as-component complains about esp-idf version being higher than 4.99. Is there a specific arduino commit that would work with 8220b2b?

ginkgm, 2022-04-17:
Target Branch | Merged by commit | Supported since IDF ver. | Last   N.A. IDF ver.
-- | -- | -- | --
master | 06d8a5bf33fbc6d94bbcea75b3eb4623ff130275 | N.A. | v5.0-dev
release/v4.4 | c9ce438466c1bdffc1e683f34f12296bb9c77f61 | v4.4.1 | N.A.
release/v4.3 | cd598bd7c00b01e9385b199e3729f4d6cf22bf7d | N.A. | v4.3.2