espressif/esp-idf #8489: Idf.py building python ImportError: cannot import name 'rest_of_line' from 'pyparsing'...

Ampaex, 2022-03-02:
Before submitting a new issue, please follow the checklist and try to find the answer.

- [ x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: None
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-1762-g6e0308d352
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:  Linux
- Using an IDE?: No, visual studio code. (I run idf.py command through bash console)
- Power Supply:   None

## Problem Description

Lately, when I tried to compile, I got to the point where I get the later mentioned error. This could be previously "solved" by doing a fullclean and compiling again.
Right now this error is not being "solved" anymore and I cannot continue building my project without errors, it's not an error that indicates easily where is the problem or any way of fixing it.

### Expected Behavior
To build the project
### Actual Behavior
Gives the error near the end of compilation.
### Steps to reproduce

1. I really don't know what is causing it.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

My code is really large and I don't know what is causing the problem.

## Debug Logs

```
Executing action: all (aliases: build)
Running ninja in directory /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build
Executing "ninja all"...
[9/861] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[280/861] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
[... 132 lines cut ...]
esptool.py v3.3-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/bootloader/bootloader.bin
[97/97] cd /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/bootloader/esp-idf/esptool_py && /home/antonio/.espressif/python_env/idf4.3_py3.9_env/bin/python /home/antonio/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/bootloader/bootloader.bin
Bootloader binary size 0x6390 bytes. 0xc70 bytes (11%) free.
[857/861] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/esp-idf/esp_system/ld/sections.ld 
cd /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/esp-idf/esp_system && /home/antonio/.espressif/python_env/idf4.3_py3.9_env/bin/python /home/antonio/esp/esp-idf/tools/ldgen/ldgen.py --config /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/sdkconfig --fragments-list "/home/antonio/esp/esp-idf/components/esp_ringbuf/linker.lf;/home/antonio/esp/esp-idf/components/driver/linker.lf;/home/antonio/esp/esp-idf/components/esp_pm/linker.lf;/home/antonio/esp/esp-idf/components/spi_flash/linker.lf;/home/antonio/esp/esp-idf/components/esp_phy/linker.lf;/home/antonio/esp/esp-idf/components/esp_system/linker.lf;/home/antonio/esp/esp-idf/components/esp_system/app.lf;/home/antonio/esp/esp-idf/components/esp_rom/linker.lf;/home/antonio/esp/esp-idf/components/hal/linker.lf;/home/antonio/esp/esp-idf/components/esp_event/linker.lf;/home/antonio/esp/esp-idf/components/esp_wifi/linker.lf;/home/antonio/esp/esp-idf/components/lwip/linker.lf;/home/antonio/esp/esp-idf/components/log/linker.lf;/home/antonio/esp/esp-idf/components/heap/linker.lf;/home/antonio/esp/esp-idf/components/soc/linker.lf;/home/antonio/esp/esp-idf/components/esp_hw_support/linker.lf;/home/antonio/esp/esp-idf/components/xtensa/linker.lf;/home/antonio/esp/esp-idf/components/esp_common/common.lf;/home/antonio/esp/esp-idf/components/esp_common/soc.lf;/home/antonio/esp/esp-idf/components/freertos/linker.lf;/home/antonio/esp/esp-idf/components/newlib/newlib.lf;/home/antonio/esp/esp-idf/components/newlib/system_libs.lf;/home/antonio/esp/esp-idf/components/app_trace/linker.lf;/home/antonio/esp/esp-idf/components/bt/linker.lf;/home/antonio/esp/esp-idf/components/esp_gdbstub/linker.lf;/home/antonio/esp/esp-idf/components/espcoredump/linker.lf" --input /home/antonio/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in --output /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/esp-idf/esp_system/ld/sections.ld --kconfig /home/antonio/esp/esp-idf/Kconfig --env-file /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/config.env --libraries-file /home/antonio/Escritorio/Proyectos_GitHub/Autopiloto/LibelulaX/LibelulaXpilot/build/ldgen_libraries --objdump /home/antonio/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/home/antonio/esp/esp-idf/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/home/antonio/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/antonio/.espressif/python_env/idf4.3_py3.9_env/lib/python3.9/site-packages/pyparsing.py)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dobairoland, 2022-03-03:
Hi @Ampaex. Are you switching between ESP-IDF versions or have multiple copies in your computer?

The above issue is caused by the fact that v5.0 is compatible with a different version of `pyparsing` package. If you run `install.sh` then it should install the right version.

dobairoland, 2022-03-03:
In case it is still not working then please share the output of `install.sh` and `. export.sh`.

dobairoland, 2022-03-03:
Ok, I see. So I assume you don't want to experiment anymore.

Just for future reference the following error indicates that you are running the environment of ESP-IDF v4.3 but using the code of ESP-IDF v5.0:

> ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/antonio/.espressif/python_env/idf4.3_py3.9_env/lib/python3.9/site-packages/pyparsing.py)

This error happen usually when the IDF_PATH environment variable is set to a different directory with a different version of ESP-IDF.

dobairoland, 2022-03-03:
The issue could be in your `get_idf` alias which could be linked to the wrong directory.

Ampaex, 2022-03-03:
The thing is I only have one repository in "$HOME/esp/esp-idf" and didn't tried any version before, just cloned the master repository two weeks ago.
My export is: export IDF_PATH=~/esp/esp-idf

Ampaex, 2022-03-03:
Sorry for not being able to provide more information, I'm in a project and need the compiler working.

dobairoland, 2022-03-03:
Ok, I understand. We can get back to this if and when you will experience this issue again.

closed 2022-03-03 by dobairoland

ferbar, 2022-03-08:
python 3.6:

```
 ~/bin/esp/esp-idf   master ●  ./install.sh esp32
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.6.15
"python3" has been detected
Traceback (most recent call last):
  File "/home/chris/bin/esp/esp-idf/tools/install_util.py", line 70, in <module>
    main()
  File "/home/chris/bin/esp/esp-idf/tools/install_util.py", line 55, in main
    subparsers = parser.add_subparsers(dest='action', required=True)
  File "/usr/lib64/python3.6/argparse.py", line 1707, in add_subparsers
    action = parsers_class(option_strings=[], **kwargs)
TypeError: __init__() got an unexpected keyword argument 'required'

```

with virtualenv for python 3.9:
`ERROR: This script was called from a virtual environment, can not create a virtual environment again
`

python 3.6:
removing the `required=True` in esp-idf/tools/install_util.py", line 55 fixes the install.sh
` ./install.sh esp32`
=> ok

`rm -rf build` in the project folder

`idf.py -p /dev/ttyUSB0 flash monitor`

=> issue solved for me.

lwshowl, 2022-04-08:
facing the same error on windows , i have previously installed esp-idf lower version on the desktop folder ,after deleting it and removing all the path variables ,i reinstalled esp-idf v5.0 on another directory, after running install.ps1 and export.ps1 the output of the export shows that the right path viariable is added , but after checking the PATH ,i found out that the former path to the lower idf is added even if the IDF_PATH is correct, i manually added the right path , the error still exist
i dont know if this have anything to do with windows

lwshowl, 2022-04-08:
Traceback (most recent call last):
  File "H:\esp-idf\tools\ldgen\ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "H:\esp-idf\tools\ldgen\ldgen\entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (C:\Users\xy\AppData\Local\Programs\Python\Python39\lib\site-packages\pyparsing.py)
[2/8] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D H:\esp32\esp32_ble_sensor\build\bootloader\esp-idf\esptool_py && C:\Users\xy\AppData\Local\Programs\Python\Python39\python.exe H:/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 H:/esp32/esp32_ble_sensor/build/bootloader/bootloader.bin"
Bootloader binary size 0x4e30 bytes. 0x31d0 bytes (39%) free.
ninja: build stopped: subcommand failed.

lwshowl, 2022-04-08:
[image]
[image]

dobairoland, 2022-04-08:
@lwshowl  The issue is cause propably by the existence of the last item in your list (with `idf4.4_py3.8`).

jonathanfoster, 2022-10-24:
Just ran into this issue myself. I upgraded from v4.4 to v5.0, deleted the entire `.espressif` directory, and installed and exported. I don't even have the previous IDF version installed. Would it make more sense to re-open this issue or submit another one?

jonathanfoster, 2022-10-24:
Looks like this issue was resolved just a few days ago (https://github.com/espressif/esp-idf/issues/9993). Temporary workaround is `idf.py fullclean` before you build.

cyberman54, 2023-02-06:
Windows / VS-Code / Platformio users, look here:
https://esp32.com/viewtopic.php?f=13&t=31744&p=109394#p108741

davec25, 2023-05-29:
File "/home/dave/esp/esp-idf/tools/ldgen/ldgen.py", line 16, in <module>

    from ldgen.entity import EntityDB

  File "/home/dave/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 12, in <module>

    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,

ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/dave/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pyparsing.py)

ninja: build stopped: subcommand failed.

I did a idf.py fullclean before I ran this build.  Getting this while trying to build generic_switch.

Josh-TapNoa, 2023-07-20:
Traceback (most recent call last):
  File "/opt/esp/idf/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/opt/esp/idf/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/opt/esp/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/pyparsing.py)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /workspaces/Xarkle-hub/test/build/log/idf_py_stderr_output_41069 and /workspaces/Xarkle-hub/test/build/log/idf_py_stdout_output_41069

Same error here, this pc has never had any other version of ESP IDF on it other than 5.02. Building on Docker, problem started seemingly without reason.

swdcom, 2023-12-06:
> …
> ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/dave/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pyparsing.py)
> …
Take a close look at your python environment path: It is pointing to the virtual environment Matter/ConnectedHomeIP generated. Try to figure out where you espressif setup put its own environment and try to activate that, as [jonathanforster](https://github.com/espressif/esp-idf/issues/8489#issuecomment-1289618318) suggested.

It would be much much easier if espressif an matter could use just one and the same virtual environment for their python stuff!

swg