espressif/esp-idf #8490: 请为TWAI驱动添加接收消息中断

flair0626, 2022-03-03:
TWAI的接收中断服务还是比较常用的，希望开发人员能够给驱动添加相关函数功能。

Dazza0, 2022-03-21:
@flair0626 请尝试调用 `twai_read_alerts()`. 如果此函数不满足要求，请随时重新打开这个 issue。

closed 2022-03-21 by Dazza0

chuhangRadarDev, 2022-06-09:
@Dazza0 CAN数据发送一条给ESP32-C3，板子会收到数百条相同的重复数据，我在`twai_read_alerts()`接收到TWAI_ALERT_RX_DATA消息的时候开始读数据  
我使用如下代码：  
```c
#include "headers.h"

static void twai_receive_task(void *arg)
{
    xSemaphoreTake(rx_sem, portMAX_DELAY);
    Serial.println("can receive:");
    while (true) {
        uint32_t alerts;
        twai_read_alerts(&alerts, portMAX_DELAY);
        if (alerts & TWAI_ALERT_RX_DATA) {
            Serial.println(to_string(alerts).c_str());
            twai_message_t rx_msg;
            twai_receive(&rx_msg, portMAX_DELAY);
            Serial.printf("CAN Receive ID:0x%x,Msg:0x%x 0x%x 0x%x 0x%x 0x%x 0x%x 0x%x 0x%x\r\n", rx_msg.identifier,
            rx_msg.data[0], rx_msg.data[1], rx_msg.data[2], rx_msg.data[3], rx_msg.data[4], rx_msg.data[5], rx_msg.data[6], rx_msg.data[7]);
            // twai_clear_receive_queue();
        }
    }

    xSemaphoreGive(rx_sem);
    vTaskDelete(NULL);
}

void can_receive(void)
{
    rx_sem = xSemaphoreCreateBinary();
    xTaskCreatePinnedToCore(twai_receive_task, "TWAI_rx", 4096, NULL, RX_TASK_PRIO, NULL, tskNO_AFFINITY);

    //Install and start TWAI driver
    ESP_ERROR_CHECK(twai_driver_install(&g_config, &t_config, &f_config));
    ESP_LOGI(EXAMPLE_TAG, "Driver installed");
    ESP_ERROR_CHECK(twai_start());
    ESP_LOGI(EXAMPLE_TAG, "Driver started");
    twai_reconfigure_alerts(TWAI_ALERT_RX_DATA, NULL);
    xSemaphoreGive(rx_sem);                     //Start RX task
}
```
我收到的数据如下：
[image]  

请问是不是有什么其他方式避免一条数据被重复接收呢？

chuhangRadarDev, 2022-06-09:
> @Dazza0 CAN数据发送一条给ESP32-C3，板子会收到数百条相同的重复数据，我在`twai_read_alerts()`接收到TWAI_ALERT_RX_DATA消息的时候开始读数据 我使用如下代码：
>
> ```c
> #include "headers.h"
>
> static void twai_receive_task(void *arg)
> { [...]

### 我如果再并上一个CAN卡就只接收到一条，是不是我用`TWAI_MODE_NORMAL`模式驱动也没有回复ACK导致发送方的重发？