espressif/esp-idf #8491: uart_isr_register crashes when a nullptr is passed for handle parameter

MartinMueller2003, 2022-03-03:
A check to test for nullptr has been removed from the uart_isr_register  function. This results in a crash when the uart is initialized with a nullptr as the handle pointer. 

The function used to check to see if the handle was valid prior to using it. Now it checks to see if the lower level function was happy and then uses the nullptr. The lower level function WAS happy with the nullptr. It is NOT acceptable programming practice to not check for a nullptr when the documentation shows that a nullptr is allowed.

Offending code
`esp_err_t uart_isr_register(uart_port_t uart_num, void (*fn)(void *), void *arg, int intr_alloc_flags,  uart_isr_handle_t *handle)
{
    int ret;
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");
    UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
    ret = esp_intr_alloc(uart_periph_signal[uart_num].irq, intr_alloc_flags, fn, arg, handle);
**_This test should be for a nullptr, NOT an ok from a sub function._**
    if (ret == ESP_OK) { 
        p_uart_obj[uart_num]->intr_handle = *handle;
    }
    UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
    return ret;
}`

o-marshmallow, 2022-03-04:
Hi @MartinMueller2003 ,

Thanks for your message, this is indeed a mistake that appeared in a recent commit, I am going to manage this

AxelLin, 2022-04-06:
@o-marshmallow 
v4.3 and v4.4 still have this bug.
I don't expect such long time to get this trivial fix.

bearsh, 2022-06-02:
@o-marshmallow any news on this?

AxelLin, 2022-06-18:
@o-marshmallow 
I don't expect such long time to fix a **mistake**.

o-marshmallow, 2022-06-20:
Hi @bearsh, 

The fix has been merged into master internally, it will be synchronized to Github as soon as CI is passing.

AxelLin, 2022-07-14:
Looks like v4.3 also needs fix? @o-marshmallow

o-marshmallow, 2022-08-08:
Merged into 4.3 too. Closing this issue.

closed 2022-08-08 by o-marshmallow

bearsh, 2022-08-10:
it still crashes for me when I use `uart_isr_register(UART, HANDLER_FUNC, NULL, INT_ALLOC_FLAGS, NULL);`

```
uru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400fbfbc  PS      : 0x00060530  A0      : 0x800da7ea  A1      : 0x3ffbbf50  
0x400fbfbc: uart_isr_register at /home/bearsh/esp/esp-idf-4.4/components/driver/uart.c:631
```

o-marshmallow, 2022-08-10:
Hi @bearsh ,

Have you initialized your UART bus with `uart_driver_install` ?

bearsh, 2022-08-10:
@o-marshmallow wow we are fast :smile:. No after looking through the code I realized that this function requires the driver to be installed and I don't do that as I have my customized driver. 

Now I use `esp_intr_alloc()` directly and it works as expected.

so sorry to bother you! stupid of me not using `esp_intr_alloc()` in the first place...

o-marshmallow, 2022-08-10:
@bearsh No problem, it happens, thanks for coming back with the solution :grinning: