espressif/esp-idf #8492: How to uset the NimBLE spp example?

brahmajit, 2022-03-03:
I'm trying to use the [NimBLE spp example (server)](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_spp/spp_server). I can connect to my ESP from my android and send data, but can't see what is being sent. I just want read the data that is being sent to my ESP from my android.

I'm using an ESP32 DevKit and ESP-IDF version at `v5.0-dev-1766-g84f9aa13c9`. I can't see that the NimBLE spp example are fairly new, so is it not possible to read the incoming data?

Log from terminal:

```
I (26622) NimBLE: mtu update event; conn_handle=0 cid=4 mtu=256

I (30132) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 15
I (36382) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 15
I (36832) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 15
I (37012) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 15
I (37152) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 15
I (37282) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 15
```

rahult-github, 2022-03-04:
Hi @brahmajit 
  
  SPP is implemented as a custom service in BLE. User needs to do read / write on gatt characteristics to achieve the same functionality as SPP.  In above, the application indicates data is received.  Please refer to the fields of [ble_gatt_access_ctxt](https://github.com/espressif/esp-nimble/blob/51f4b84aca5ae5a4083fa31b0cb5f34bfabef2ee/nimble/host/include/host/ble_gatt.h#L621) , which you can refer in your application to get needed information. The buffer [om](https://github.com/espressif/esp-nimble/blob/51f4b84aca5ae5a4083fa31b0cb5f34bfabef2ee/nimble/host/include/host/ble_gatt.h#L641) present inside this structure can be accessed and used based on the operation being attempted.

Thanks,
 Rahul

brahmajit, 2022-03-04:
@rahult-github thanks. I'm able to read the data being sent using the `ctxt->om->om_data`. But I get some unwanted character at the end of the message. Any idea what's happening?

```
I (9322) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 11
I (9322) NimBLE_SPP_BLE_PRPH: test
��q
```

rahult-github, 2022-03-04:
@brahmajit  , Can you please confirm that you are printing data of correct length ?  This is the data that remote has sent. You can even enable DEBUG log level and see what is the actual data received to host over HCI.

brahmajit, 2022-03-04:
@rahult-github I can confirm that the data I'm getting is of correct size.

Code snippet

```c
case BLE_GATT_ACCESS_OP_WRITE_CHR:
	ESP_LOGI(
		tag,
		"Data received in write event,conn_handle = %x,attr_handle = %x",
		conn_handle, attr_handle);
	ESP_LOGI(tag, "%s %d", ctxt->om->om_data, ctxt->om->om_len);
	break;
```

```
I (7846) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 11
I (7846) NimBLE_SPP_BLE_PRPH: test
��q 6
I (8066) NimBLE_SPP_BLE_PRPH: Data received in write event,conn_handle = 0,attr_handle = 11
I (8066) NimBLE_SPP_BLE_PRPH: test
��q 6
```

rahult-github, 2022-03-04:
Hi @brahmajit ,
 
  I think you are doing this the wrong way. om_data is a pointer of type uint8_t. I would rather have a for loop running for the length and printing each byte using %x to get the contents . I think this is now more of a C coding issue rather than BLE SPP server functionality issue. 

Thanks,
 Rahul

brahmajit, 2022-03-04:
@rahult-github Thanks for the heads up. I'll do that.

closed 2022-03-04 by brahmajit

DurvalMenezes, 2025-01-26:
Hi @brahmajit (or anyone, actually, now that it seems you're no longer around). I'm also trying to get Android (or a Windows or Linux client, or actually anything  :) ) to interact with my ESP32-S3 using the nimble SPP example (esp-idf/examples/bluetooth/nimble/ble_spp/spp_server). If you can, I would like to ask a couple of questions:

1) What Android app did you use to make the connection? 
2) Do you know of any other way to connect to the ESP32-S3 using that example (eg, some linux or windows program)?
3) Can you please give us an update on your progress on this? 

TIA!