espressif/esp-idf #8493: ESP-IDF 16 bit register address

domko99, 2022-03-03:
I need to implement I2C functions for reading and writing to device with 16 bit address length. But I2C functions which are declared 
in i2c.h are supporting only uint8_t register address and I need support for int16_t register address. Is there any solution with i2c.h library? I need to implement this functions. Thanks for help and suggestions.

   Read the register_address value from the mlx90632
 
 * i2c read is processor specific and this function expects to have address of mlx90632 known, as it operates purely on
 * register addresses.
 *
 * @note Needs to be implemented externally
 * @param[in] register_address Address of the register to be read from
 * @param[out] *value pointer to where read data can be written
 * @retval 0 for success
 * @retval <0 for failure

extern int32_t mlx90632_i2c_read(int16_t register_address, uint16_t *value);
/************************************************************************************************/
   Write value to register_address of the mlx90632
 
 * i2c write is processor specific and this function expects to have address of mlx90632 known, as it operates purely
 * on register address and value to be written to.
 *
 * @note Needs to be implemented externally
 * @param[in] register_address Address of the register to be read from
 * @param[in] value value to be written to register address of mlx90632
 * @retval 0 for success
 * @retval <0 for failure

extern int32_t mlx90632_i2c_write(int16_t register_address, uint16_t value);

negativekelvin, 2022-03-03:
Those are devices addresses not register addresses.  mlx90632_i2c_read and write can be implemented using read and write functions in i2c.h.

o-marshmallow, 2022-03-04:
Hi @domko99 ,

I think you are mixing up between device address and register address. The I2C specification defines a device address as either 7-bit or 10-bit, every peripheral shall have an address that let the master communicate with it.

The register address, on the other hand, is not standard in the sense that it can be 8bits, 16bits, 32bits, and specific to the device you are using.

In the case of your mlx90632, after taking a look at the datasheet, the device address is by default `0x3a`:
```
9.1. I2C address

By default, the device responds to the 7-bit slave address 0x3A.
(Page 22)
```

However, you register addresses are 16-bit, you can take a look at `Mapping` section of the datasheet.

Thus, as @negativekelvin stated, you can use the functions defined in `i2c.h`. Let's have a try:
```
#define MX90632_ADDR 0x3A

int32_t mlx90632_i2c_read(int16_t register_address, uint16_t value) {
    /* I assume that your I2C driver has already been initialized */
    esp_err_t err = ESP_OK;
    /* In order to retrieve a register from a device, we have to write its address on the bus,
       then directly initiate a new read transaction, with a repeated start signal.
       Fortunately, we have a function in the API for that. Prepare the write buffer,
       which contains the register address. According to the datasheet (9.2. Addressed read),
       the address shall be passed as a big endian value, so MSB first. */
    uint8_t write_buf[] = { (register_address >> 8) & 0xff, register_address & 0xff };
    /* Prepare the read buffer, which contain the 32-bits you need.
       From what I understand from the datasheet, it is possible to read as many bytes as we want
       from the device, from any address, until the master sends a STOP.
       So let's try reading 32-bits directly. If that doesn't work, split the following transaction
       into 2, with an incremented address */
    uint8_t read_buf[4] = { 0 };
    
    /* Initiate the transaction, give it a 1 second delay before timeout */
    err = i2c_master_write_read_device(YOUR_I2C_BUS, MLX90632_ADDR, write_buf, sizeof(write_buf), read_buf, sizeof(read_buf), 1000 / portTICK_PERIOD_MS);
    if (err != ESP_OK) {
      return 0; /* do what you want here */
    }

    /* Finally, convert the data received into an int32_t.
       The datasheet informs us that the first 16-bit word is the least significant one,
       but it's first byte is the most significant byte.
       So if our 32-bit value should be: B3 B2 B1 B0,
       read_buf in fact contains: B1 B0 B3 B2 */
    return (read_buf[2] << 24) | (read_buf[3] << 16) | (read_buf[0] << 8) | read_buf[1];
}
```

I haven't tested that code but this gives you an idea on how to do this. Please check carefully the endianness of the data too. For the write, it's even simpler as you only need to provide a single write buffer to the function `i2c_master_write_to_device`. More info on these functions [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv426i2c_master_write_to_device10i2c_port_t7uint8_tPK7uint8_t6size_t10TickType_t)

domko99, 2022-03-05:
Thanks for help. I was able to implement write and read functions and now my sensor is working. Thank you once again.

o-marshmallow, 2022-03-05:
@domko99 Great, thanks for your feedback!

closed 2022-03-05 by o-marshmallow