espressif/esp-idf #8495: Component dependency issue

svcguy, 2022-03-04:
## Environment

- Development Kit:      **ESP32-S2 Kaluga**
- Kit version (for WroverKit/PicoKit/DevKitC):  **v1.3**
- Module or chip used:  **ESP32-S2 WROVER**
- IDF version (run ``git describe --tags`` to find it): **v4.4**
- Build System:         **idf.py**
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): **2021r2-patch2-8.4.0**
- Operating System:     **Windows**
- (Windows only) environment type: **ESP Command Prompt**
- Using an IDE?: **Yes VSCode (ESP-IDF VSCode extension 1.4.0)**
- Power Supply:        **N/A**

## Problem Description

Component Manager seems to have trouble with ^ and ~ notation

### Expected Behavior

Resolve dependencies

### Actual Behavior

Fails to resolve dependencies

### Steps to reproduce

```
PS C:\Users\svcgu\OneDrive\Documents\ESP32\ltc2488> idf.py add-dependency espressif/esp32_s2_kaluga_kit==0.0.4
Executing action: add-dependency
Created "c:\users\svcgu\onedrive\documents\esp32\ltc2488\main\idf_component.yml"
Successfully added dependency "espressif/esp32_s2_kaluga_kit==0.0.4" for component "main"
PS C:\Users\svcgu\OneDrive\Documents\ESP32\ltc2488> idf.py add-dependency lvgl/lvgl==8.2.0
Executing action: add-dependency
Successfully added dependency "lvgl/lvgl==8.2.0" for component "main"
PS C:\Users\svcgu\OneDrive\Documents\ESP32\ltc2488> idf.py reconfigure
Executing action: reconfigure
Running cmake in directory c:\users\svcgu\onedrive\documents\esp32\ltc2488\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\users\svcgu\onedrive\documents\esp32\ltc2488"...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
Solving dependencies requirements
CMake Error at C:/Users/svcgu/esp/esp-idf/tools/cmake/build.cmake:455 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 3, in <module>
      main()
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 110, in main
      args.func(args)
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 37, in prepare_dep_dirs
      ComponentManager(args.project_dir).prepare_dep_dirs(
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 325, in prepare_dep_dirs
      downloaded_component_paths = download_project_dependencies(
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 36, in download_project_dependencies
      solution = solver.solve()
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 30, in solve
      result = self._solver.solve()
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\mixology\version_solver.py", line 63, in solve
      if not self._run():
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\mixology\version_solver.py", line 76, in _run
      self._propagate(next_package)
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\mixology\version_solver.py", line 108, in _propagate
      root_cause = self._resolve_conflict(incompatibility)
    File "C:\Users\svcgu\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\mixology\version_solver.py", line 282, in _resolve_conflict
      raise SolverFailure(incompatibility)

  idf_component_manager.version_solver.mixology.failure.SolverFailure:
[... 151 lines cut ...]
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------
```

svcguy, 2022-03-04:
Just noticed that the above set of dependencies will never resolve due to LVGL.  But the issue with the ```lvgl_esp32_drivers==0.0.4``` is the actual issue.  Revert that to ```lvgl_esp32_drivers==0.0.3``` and it will resolve
```Processing 7 dependencies:
[1/7] espressif/button (2.0.0)
[2/7] espressif/es8311 (1.0.0)
[3/7] espressif/esp32_s2_kaluga_kit (0.0.3)
[4/7] espressif/led_strip (1.1.0-alpha)
[5/7] idf (4.4.0)
[6/7] lvgl/lvgl (7.11.0)
[7/7] lvgl/lvgl_esp32_drivers (0.0.3)```

kumekay, 2022-03-04:
Hello @svcguy
Thank you 
Could you please also share the full content of your `idf_component.yml` file for `main` where you have the problem (and for other components in the `components` directory if you have any)?

tore-espressif, 2022-03-04:
@svcguy thank you for reporting.

Board Support Package (BSP) for Kaluga kit (both version 0.0.3 and 0.0.4) was one of our first trials with BSPs, so it uses already obsolete https://components.espressif.com/component/lvgl/lvgl_esp32_drivers, this package supports only LVGL v7. So when you define dependency on LVGL v8.2.0 it can't be resolved.

I'll update espressif/esp32_s2_kaluga_kit for LVGL v8 soon

igrr, 2022-03-04:
Just my 2c, it would be excellent if the error message reported by the component manager pointed at the root cause of this issue (`lvgl` incompatibility). Although I understand that version solving is a complex problem, so maybe it is too optimistic to require this...

svcguy, 2022-03-04:
Correct.  I realized that after posting.  The version of LVGL isn’t important to me right now.  Even removing LVGL as a dependency, it still fails (LVGL as a dependency isn’t necessary anyway since lvgl_esp32_drivers will pull LVGL in).  The issue is actually the es8311 dependency and I think the root issue is with the use of the ^ and ~ version specifiers.  It’s the only difference between the 0.0.3 and 0.0.4 versions.

And an update to LVGL v8 would be wonderful, thank you so much.  I’ll wait patiently :)

From: Tomas Rezucha
Sent: Friday, March 4, 2022 07:56
To: espressif/esp-idf
Cc: Andy; Mention
Subject: Re: [espressif/esp-idf] Component dependency issue (IDFGH-6873) (Issue #8495)

@svcguy thank you for reporting.
Board Support Package (BSP) for Kaluga kit (both version 0.0.3 and 0.0.4) was one of our first trials with BSPs, so it uses already obsolete https://components.espressif.com/component/lvgl/lvgl_esp32_drivers, this package supports only LVGL v7. So when you define dependency on LVGL v8.2.0 it can't be resolved.
I'll update espressif/esp32_s2_kaluga_kit for LVGL v8 soon

kumekay, 2022-03-07:
@svcguy Thank you for the issue, yes there is an with the solver and pre-release dependencies like `0.0.2-alpha`, we will check it

kumekay, 2022-03-23:
@svcguy The issue with caret was fixed in this commit https://github.com/espressif/idf-component-manager/commit/bede7f34e5cb329a2b7513252bba68173b0d36a0 however the package is not published on PyPI yet. 

If you want to try it now, you can install it into your ESP-IDF virtual env by running this command after export.ps1:

```
pip install git+https://github.com/espressif/idf-component-manager.git --upgrade

```

closed 2022-04-19 by Alvin1Zhang