espressif/esp-idf #8499: Since latest esp-idf rebase there are compiler errors in crypto_mbedtls-ec.c

0xFEEDC0DE64, 2022-03-04:
How did the struct member `grp` go missing in the first place?

```
In file included from <project-root>/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:19,                                                                                                       [15/617]
                 from <project-root>/esp-idf/components/wpa_supplicant/src/utils/common.h:432,                                                                                   
                 from <project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:12:
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c: In function 'crypto_debug_print_ec_key':
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:804:43: error: 'mbedtls_ecp_keypair' {aka 'struct mbedtls_ecp_keypair'} has no member named 'grp'
    mbedtls_ecp_curve_info_from_grp_id( ecp->grp.id )->name );                                                                                                                                                                                
                                           ^~                                                                                                                                                                                                 
<project-root>/esp-idf/components/log/include/esp_log.h:413:116: note: in definition of macro 'ESP_LOG_LEVEL'                      
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format, tag), ##__VA_ARGS__); } \
                                                                                                                    ^~~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:64:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~ 
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:803:2: note: in expansion of macro 'wpa_printf'
  wpa_printf(MSG_ERROR, "curve: %s\n",  
  ^~~~~~~~~~                 
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:804:43: error: 'mbedtls_ecp_keypair' {aka 'struct mbedtls_ecp_keypair'} has no member named 'grp'
    mbedtls_ecp_curve_info_from_grp_id( ecp->grp.id )->name );
                                           ^~
<project-root>/esp-idf/components/log/include/esp_log.h:414:116: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_WARN )      { esp_log_write(ESP_LOG_WARN,       tag, LOG_FORMAT(W, format, tag), ##__VA_ARGS__); } \
                                                                                                                    ^~~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:64:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~ 
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:803:2: note: in expansion of macro 'wpa_printf'
  wpa_printf(MSG_ERROR, "curve: %s\n",
  ^~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:804:43: error: 'mbedtls_ecp_keypair' {aka 'struct mbedtls_ecp_keypair'} has no member named 'grp'
    mbedtls_ecp_curve_info_from_grp_id( ecp->grp.id )->name );
                                           ^~
<project-root>/esp-idf/components/log/include/esp_log.h:415:116: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format, tag), ##__VA_ARGS__); } \
                                                                                                                    ^~~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:64:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~ 
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:803:2: note: in expansion of macro 'wpa_printf'
  wpa_printf(MSG_ERROR, "curve: %s\n",
  ^~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:804:43: error: 'mbedtls_ecp_keypair' {aka 'struct mbedtls_ecp_keypair'} has no member named 'grp'
    mbedtls_ecp_curve_info_from_grp_id( ecp->grp.id )->name );
                                           ^~
<project-root>/esp-idf/components/log/include/esp_log.h:416:116: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_VERBOSE )   { esp_log_write(ESP_LOG_VERBOSE,    tag, LOG_FORMAT(V, format, tag), ##__VA_ARGS__); } \
                                                                                                                    ^~~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:64:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~ 
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:803:2: note: in expansion of macro 'wpa_printf'
  wpa_printf(MSG_ERROR, "curve: %s\n",
  ^~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:804:43: error: 'mbedtls_ecp_keypair' {aka 'struct mbedtls_ecp_keypair'} has no member named 'grp'
    mbedtls_ecp_curve_info_from_grp_id( ecp->grp.id )->name );
                                           ^~
<project-root>/esp-idf/components/log/include/esp_log.h:417:116: note: in definition of macro 'ESP_LOG_LEVEL'
         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format, tag), ##__VA_ARGS__); } \
                                                                                                                    ^~~~~~~~~~~
<project-root>/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:64:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~ 
<project-root>/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:803:2: note: in expansion of macro 'wpa_printf'
  wpa_printf(MSG_ERROR, "curve: %s\n",
  ^~~~~~~~~~
[20/641] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/tcp_in.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

0xFEEDC0DE64, 2022-03-04:
```
    wpa_printf(MSG_ERROR, "curve: %s\n",
               mbedtls_ecp_curve_info_from_grp_id( ecp->MBEDTLS_PRIVATE(grp).id )->name );
```
seems to fix the issue :)
who decided to port private/protected/public to outdated C?

negativekelvin, 2022-03-04:
Guess it was missed in https://github.com/espressif/esp-idf/commit/3b71bd73265de7af7a091838d77ca527cba9aa36#diff-00f66c8d0a1d85eaa2cca3e48beb1703ed613c0586848e5a2fb394165eccc6a4R188

No CI test with wpa debugging on?

0xFEEDC0DE64, 2022-03-07:
Merging my esp log improvements would also have shown the issue here

closed 2022-03-07 by 0xFEEDC0DE64