espressif/esp-idf #8503: seems that if `Minimum Supported ESP32-C3 Revision` lower than `Rev3`,`idf.py` can't work!

wonderxue, 2022-03-05:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32 C3]
- IDF version : [v4.4]
- Build System: [idf.py]
- Compiler version :[8.4.0]
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
if `Minimum Supported ESP32-C3 Revision` lower than `Rev3`,`idf.py` can't work!
here is console print:
```
Executing action: flash
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Running ninja in directory /home/cxk/Desktop/Project/esp_hid_device/build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /home/cxk/Desktop/Project/esp_hid_device/sdkconfig
Loading defaults file /home/cxk/Desktop/Project/esp_hid_device/sdkconfig.defaults...
Loading defaults file /home/cxk/Desktop/Project/esp_hid_device/sdkconfig.defaults.esp32c3...
...
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cxk/Desktop/Project/esp_hid_device/build
[16/1259] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[274/1259] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/cxk/Desktop/ESP/ESP32/tools/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/cxk/Desktop/ESP/ESP32/tools/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/cxk/Desktop/ESP/ESP32/tools/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /home/cxk/Desktop/Project/esp_hid_device/sdkconfig
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bootloader/subproject/main/ld/esp32c3/bootloader.ld
-- Adding linker script /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bootloader/subproject/main/ld/esp32c3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32c3 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bootloader /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bootloader_support /home/cxk/Desktop/ESP/ESP32/esp-idf/components/efuse /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp32c3 /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_common /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_hw_support /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_rom /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_system /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esptool_py /home/cxk/Desktop/ESP/ESP32/esp-idf/components/freertos /home/cxk/Desktop/ESP/ESP32/esp-idf/components/hal /home/cxk/Desktop/ESP/ESP32/esp-idf/components/log /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bootloader/subproject/main /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bootloader/subproject/components/micro-ecc /home/cxk/Desktop/ESP/ESP32/esp-idf/components/newlib /home/cxk/Desktop/ESP/ESP32/esp-idf/components/partition_table /home/cxk/Desktop/ESP/ESP32/esp-idf/components/riscv /home/cxk/Desktop/ESP/ESP32/esp-idf/components/soc /home/cxk/Desktop/ESP/ESP32/esp-idf/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cxk/Desktop/Project/esp_hid_device/build/bootloader
[361/1259] Performing build step for 'bootloader'

...

[87/88] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated /home/cxk/Desktop/Project/esp_hid_device/build/bootloader/bootloader.bin
[88/88] cd /home/cxk/Desktop/Project/esp_hid_device/build/bootloader/esp-idf/esptool_py && /home/cxk/Desktop/ESP/ESP32/tools/python_env/idf4.4_py3.9_env/bin/python /home/cxk/Desktop/ESP/ESP32/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/cxk/Desktop/Project/esp_hid_device/build/bootloader/bootloader.bin
Bootloader binary size 0x4d80 bytes. 0x3280 bytes (39%) free.
[1256/1259] Linking CXX executable esp_hid_device.elf
FAILED: esp_hid_device.elf 
: && /home/cxk/Desktop/ESP/ESP32/tools/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imc -nostartfiles -march=rv32imc --specs=nosys.specs CMakeFiles/esp_hid_device.elf.dir/project_elf_src_esp32c3.c.obj -o 
...
wrap=__gxx_personality_v0  -u __cxa_guard_dummy  -lstdc++  esp-idf/pthread/libpthread.a  -lgcc  esp-idf/cxx/libcxx.a  -u __cxx_fatal_exception  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/app_trace/libapp_trace.a  -lgcov  -lc  -L/home/cxk/Desktop/ESP/ESP32/esp-idf/components/bt/controller/lib_esp32c3_family/esp32c3  -lbtdm_app && :
/home/cxk/Desktop/ESP/ESP32/tools/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/cxk/Desktop/ESP/ESP32/esp-idf/components/esp_phy/lib/esp32c3/libphy.a(phy_chip_v7.o): in function `phy_get_romfunc_addr':
(.text.phy_get_romfunc_addr+0x1a): undefined reference to `rom_temp_to_power'
/home/cxk/Desktop/ESP/ESP32/tools/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/cxk/Desktop/ESP/ESP32/esp-idf/components/bt/controller/lib_esp32c3_family/esp32c3/libbtdm_app.a(lld_adv.o): in function `r_lld_ext_adv_dynamic_aux_pti_process_hack':
(.iram1.0+0x22): undefined reference to `r_lld_ext_adv_dynamic_aux_pti_process'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
following are `ESP32C3-Specific` in sdkconfig file
```
#
# ESP32C3-Specific
#
# CONFIG_ESP32C3_DEFAULT_CPU_FREQ_80 is not set
CONFIG_ESP32C3_DEFAULT_CPU_FREQ_160=y
CONFIG_ESP32C3_DEFAULT_CPU_FREQ_MHZ=160
# CONFIG_ESP32C3_REV_MIN_0 is not set
# CONFIG_ESP32C3_REV_MIN_1 is not set
CONFIG_ESP32C3_REV_MIN_2=y
# CONFIG_ESP32C3_REV_MIN_3 is not set
CONFIG_ESP32C3_REV_MIN=2
CONFIG_ESP32C3_DEBUG_OCDAWARE=y
# CONFIG_ESP32C3_DEBUG_STUBS_ENABLE is not set
CONFIG_ESP32C3_BROWNOUT_DET=y
CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_7=y
# CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_6 is not set
# CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_5 is not set
# CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_4 is not set
# CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_3 is not set
# CONFIG_ESP32C3_BROWNOUT_DET_LVL_SEL_2 is not set
CONFIG_ESP32C3_BROWNOUT_DET_LVL=7
CONFIG_ESP32C3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
# CONFIG_ESP32C3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32C3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32C3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32C3_RTC_CLK_SRC_INT_RC=y
# CONFIG_ESP32C3_RTC_CLK_SRC_EXT_CRYS is not set
# CONFIG_ESP32C3_RTC_CLK_SRC_EXT_OSC is not set
# CONFIG_ESP32C3_RTC_CLK_SRC_INT_8MD256 is not set
CONFIG_ESP32C3_RTC_CLK_CAL_CYCLES=1024
# end of ESP32C3-Specific
```

AxelLin, 2022-03-05:
https://github.com/espressif/esp-idf/issues/7946

wonderxue, 2022-03-05:
@AxelLin thx let me know, i bought 3 rev2, sad

closed 2022-03-05 by wonderxue