espressif/esp-idf #8505: spiffsgen example mount failed

rcmz, 2022-03-05:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- Module or chip used: Esp32-s3-wroom-1 N32R8V
- IDF version: v5.0-dev-1829-ge29e96101d
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- IDE: VScode
- Power Supply: USB

## Problem Description

I get an "SPIFFS: mount failed" when running the spiffsgen example

## Debug Logs

```
rcmz/esp/esp-idf/tools/idf.py -p /dev/ttyUSB0 monitor
Executing action: monitor
Running idf_monitor in directory /home/rcmz/esp/spiffsgen
Executing "/home/rcmz/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/rcmz/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /home/rcmz/esp/spiffsgen/build/spiffsgen.elf -m '/home/rcmz/.espressif/python_env/idf5.0_py3.10_env/bin/python' '/home/rcmz/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x16e0
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f9c
entry 0x403b6254
I (32) boot: ESP-IDF v5.0-dev-1829-ge29e96101d 2nd stage bootloader
I (33) boot: compile time 11:48:58
I (33) boot: chip revision: 0
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : SLOW READ
I (46) boot.esp32s3: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08208h ( 33288) map
I (117) esp_image: segment 1: paddr=00018230 vaddr=3fc90b70 size=02770h ( 10096) load
I (120) esp_image: segment 2: paddr=0001a9a8 vaddr=40374000 size=05670h ( 22128) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1cad4h (117460) map
I (162) esp_image: segment 4: paddr=0003cafc vaddr=40379670 size=074fch ( 29948) load
I (171) esp_image: segment 5: paddr=00044000 vaddr=50000000 size=00010h (    16) load
I (176) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (191) cpu_start: Pro cpu up.
I (191) cpu_start: Starting app cpu, entry point is 0x403750ec
0x403750ec: call_start_cpu1 at /home/rcmz/esp/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (205) cpu_start: Pro cpu start user code
I (205) cpu_start: cpu freq: 160000000 Hz
I (205) cpu_start: Application information:
I (208) cpu_start: Project name:     spiffsgen
I (213) cpu_start: App version:      1
I (218) cpu_start: Compile time:     Mar  5 2022 11:48:54
I (224) cpu_start: ELF file SHA256:  bfc1a2f132653bc0...
I (230) cpu_start: ESP-IDF:          v5.0-dev-1829-ge29e96101d
I (237) heap_init: Initializing. RAM available for dynamic allocation:
I (244) heap_init: At 3FC93CD8 len 0004C328 (304 KiB): D/IRAM
I (250) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (257) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (263) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (270) spi_flash: detected chip: mxic
W (274) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (285) spi_flash: flash io: dio
W (289) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (303) sleep: Configure to isolate all GPIO pins in sleep state
I (309) sleep: Enable automatic switching of GPIO sleep configuration
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (336) example: Initializing SPIFFS
E (336) SPIFFS: mount failed, -10025
E (336) example: Failed to mount or format filesystem
```

igrr, 2022-03-07:
@alive2 It looks like you are using a module with Octal SPI flash:
```
Octal Flash Mode Enabled
```
but the corresponding option is not enabled, according to this:
```
I (285) spi_flash: flash io: dio
```

Could you please try enabling `CONFIG_ESPTOOLPY_OCT_FLASH` in menuconfig and then run the example again?

igrr, 2022-03-07:
(sorry for tagging the wrong person at first — the message was directed at @rcmz)

rcmz, 2022-03-07:
It works ! (at least for spiffsgen) thanks a lot 😃

rcmz, 2022-03-07:
So it works but it's muuuuuuch (>~10x) slower than when using an SD card
My SPI flash is running in octal mode DDR at 80MHz
And my SD card is only running in sdmmc 1 lane 20MHz
All I'm doing is accessing a big (4MB) file
Isn't this weird ?

closed 2022-03-12 by rcmz