espressif/esp-idf #8506: windows11上构建hello_world项目，ninja编译失败，报找不到目录的错误

arashrun, 2022-03-05:
环境：win11
idf版本：4.4
按照https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/get-started/linux-macos-setup.html网址进行操作的。idf.py build 调用cmake编译完成之后，调用ninja进行编译。报系统找不到指定的路径的错误
[image]

尝试：我手动执行报错的命令：cmd.exe /C "cd /D C:\Users\ming\hello_world\build && E:\Espressif\tools\cmake\3.20.3\bin\cmake.exe -E touch C:/Users/ming/hello_world/build/project_elf_src_esp32.c"。发现还是报错，通过排查，发现不支持cmd.exe /C "cd /D xxx”这种用法
[image]

dobairoland, 2022-03-07:
Hi @arashrun. 

You are using Windows so please use the Windows Tool Installer and follow the recommendations from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html. You have linked the guide for Linux and MacOs. The Windows Tool Installer will install an environment where everything will be set up.

If you still encountering an error then please copy the error output here instead of sharing it as an image. I cannot use translator if image is provided and some of the command lines are cut in half and not shown. Thanks!

arashrun, 2022-03-07:
Hi @dobairoland .
I am sure i have install the Windows Tool on my Windows11. Here is the complete output.

[hello-world-log.txt](https://github.com/espressif/esp-idf/files/8198313/hello-world-log.txt)
By the way. I had modify the **CMakeTestCCompiler.cmake** for solving cmake  test c/c++ compiler failed. At line of 61
` 58 message("${CMAKE_BINARY_DIR}+++")`
` 59 message("test program+++++++${CMAKE_C_COMPILER_WORKS}")`
` 60   #set(CMAKE_C_COMPILER_WORKS ${CMAKE_C_COMPILER_WORKS})`
` 61   set(CMAKE_C_COMPILER_WORKS TRUE)`

dobairoland, 2022-03-07:
Thanks @arashrun for the log.

Just to confirm, you are running the following command from PowerShell and it returns that the path doesn't exist. Right?
```
cmd.exe /C "cd /D xxx”
```

Do you use instead of `xxx` a valid path, for example `build` if you are in the project directory?

dobairoland, 2022-03-07:
Does the following command work for you?
```sh
E:\Espressif\tools\cmake\3.20.3\bin\cmake.exe --version
```

georgik, 2022-03-07:
Please could you try build of example bundled in ESP-IDF?
```
cd E:\Espressif\frameworks\esp-idf-v4.4\examples\get-started\blink
idf.py build
```

arashrun, 2022-03-07:
> Thanks @arashrun for the log.
> 
> Just to confirm, you are running the following command from PowerShell and it returns that the path doesn't exist. Right?
> 
> ```
> cmd.exe /C "cd /D xxx”
> ```
> 
> Do you use instead of `xxx` a valid path, for example `build` if you are in the project directory?

Yes, The second image is the result of insteaded. And `cmake --version` is work for me
```
PS E:\Espressif\frameworks\esp-idf-v4.4> E:\Espressif\tools\cmake\3.20.3\bin\cmake.exe --version
cmake version 3.20.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

arashrun, 2022-03-07:
>

arashrun, 2022-03-07:
> Please could you try build of example bundled in ESP-IDF?
> 
> ```
> cd E:\Espressif\frameworks\esp-idf-v4.4\examples\get-started\blink
> idf.py build
> ```

Hi, @georgik . That still not work for me. The log is under
[blink.txt](https://github.com/espressif/esp-idf/files/8201458/blink.txt)

georgik, 2022-03-08:
Thank you for the test @arashrun . 
I re-run the test in CMD and  PowerShell with ESP-IDF 4.4 installed on the same path on E: drive and it works on my Windows 11.

Is the E: drive regular HDD mounted to the system or is in network drive?

Please, could you also run `chcp` command and provide here output, just to make sure that code page is not causing a problem. It might not be connected to the problem, but better to double-check.
The test was performed on Windows with chcp = 437

georgik, 2022-03-08:
@arashrun Thank you for additional information.
I re-run test case on Windows 11 with code page 936 on Azure. The build was without problem.
Also the command with cmd.exe mentioned above works without problem.

Can you check what is the filesystem on external drive E:? Is it NTFS?
My I ask you to test the installation on C:\ drive?

georgik, 2022-03-08:
Thank you for confirmation @arashrun . I'm glad that ESP-IDF now works for you.

Could you share with us which registry change lead to the problem? We might be able to add integrity check for modified cmd value to idf-installer.

closed 2022-03-08 by dobairoland

arashrun, 2022-03-09:
Sure. The registry is `HKEY_LOCAL_MACHINE\Software\Microsoft\Command Processor\AutoRun\REG_SZ` which i found on this link [cmd microsoft](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/cmd). Just as i said, that's my fault. I set an not exists path as the registry value that cause the error...
Thank you for your patience @georgik

georgik, 2022-03-09:
Thank you for providing details.