espressif/esp-idf #8507: esp32c3: USB stopped working when light sleep is enabled.

HiFiPhile, 2022-03-05:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- Kit version : [v1]
- IDF version (run ``git describe --tags`` to find it):
```
v5.0-dev-1730-g229ed08484
 f31694c9f1746ba189a4bcae2e34db15135ddb22 components/asio/asio (asio-1-12-0-201-gf31694c9)
 d037ec89546fad14b5c4d5456c2e23a71e554966 components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.0)
 26afd7bb31c9bf1c7072b47d06293321ddcd5775 components/bt/controller/lib_esp32 (heads/master)
 a3f7d7ffa2b261b1415042d8b7cd457cc2b4b1de components/bt/controller/lib_esp32c3_family (heads/master)
 1dc1ec6e76b0ab3bf93cc9f1ff7a2a09141e7c61 components/bt/host/nimble/nimble (heads/nimble-1.3.0-idf-1-g1dc1ec6e)
 eeecc49ce8af123cf8ad40efdb9673e37b56230f components/cmock/CMock (v2.5.2-2-geeecc49)
 fced1a554f1b870f5a4e336431ab9915c4d90326 components/esp_phy/lib (heads/master)
 d464572fa5be0b6508cdb6a93124f9ea72280225 components/esp_wifi/lib (heads/master)
 ded13e025628d4c5ca350378b391e9b0d38bcb57 components/esptool_py/esptool (v3.2-47-gded13e0)
 5545eb14d1117a7865a05a6b4a0f19c62760f8c7 components/ieee802154/lib (heads/master)
 d348621ca93571343a56862df7de4ff3bc9b5667 components/json/cJSON (v1.7.15)
 2749568fe15df2003f6c3f37f0dfd44f8f01fcd6 components/lwip/lwip (remotes/origin/master-224-g2749568f)
 8b0e35f2ad477fcc2a267cf434528024b8499085 components/mbedtls/mbedtls (mbedtls-2.13.1-8771-g8b0e35f2a)
 985078affa8a2d2b56b87c8e6455252850f895c6 components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_3.1-293-g985078a)
 901353a8c17d5c3040b3850d571550b0472b921b components/openthread/lib (heads/master)
 e736d2488d9d4f787650d81f2b84f59a194f2b41 components/openthread/openthread (thread-reference-20200818-1424-ge736d2488)
 dac1a65feac4ad72f612aab99f487056fbcf5c1a components/protobuf-c/protobuf-c (v1.3.0)
 f5e26c4e933189593a71c6b82cda381a7b21e41c components/spiffs/spiffs (0.2-221-gf5e26c4)
 c1986586098643edfa6b75a776299c576ad0ce8b components/tinyusb/tinyusb (remotes/origin/esp-based_on_9e992698)
 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 components/unity/unity (v2.4.3-51-g7d2bf62)
 7e8e249990ec491ec15990cf95b6d871a66cf64a examples/build_system/cmake/import_lib/main/lib/tinyxml2 (2.0.2-659-g7e8e249)
 077758db50d4b55addab56e723a5a379f09077d0 examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib (077758d)
```
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
(crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

USB stops working if `light_sleep_enable` is true in `esp_pm_configure`.

According to the doc:
```
 if you don’t need USB CDC in sleep and want to keep low power consumption, please disable the menuconfig CONFIG_RTC_CLOCK_BBPLL_POWER_ON_WITH_USB
```
So I set `CONFIG_RTC_CLOCK_BBPLL_POWER_ON_WITH_USB=y` alongside with `CONFIG_ESP_SLEEP_GPIO_RESET_WORKAROUND is not set` & `CONFIG_PM_SLP_DISABLE_GPIO is not set`

### Expected Behavior

USB keep working under light sleep.

### Actual Behavior

USB stops working under light sleep, `Unknown USB device (descriptor request failed)`

### Steps to reproduce

1. Connect USB pins of dev kit to PC.
2. Compile & run following code
3. Remove `.light_sleep_enable = true`
4. Compile & run again

### Code to reproduce this issue

```cpp
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_pm.h"
#include "esp_log.h"

static const char *TAG = "example";

static void example_task(void * parm)
{
    while(1)
    {
        vTaskDelay(100);
        ESP_LOGI(TAG, "Loop"); 
    }
}

void app_main(void)
{
    esp_pm_config_esp32c3_t pm_config = {
        .max_freq_mhz = 80,
        .min_freq_mhz = 80,
    };
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config));
    xTaskCreate(example_task, "example_task", 4096, NULL, 3, NULL);
}
```

## Debug Logs

### From UART console, USB console no output

```
I (30) boot: ESP-IDF v5.0-dev-1730-g229ed08484-dirty 2nd stage bootloader
I (30) boot: compile time 14:32:46
I (30) boot: chip revision: 3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 coredump         Unknown data     01 03 00010000 00010000
I (87) boot:  3 factory          factory app      00 00 00020000 00140000
I (94) boot:  4 ota_0            OTA app          00 10 00160000 00140000
I (102) boot:  5 ota_1            OTA app          00 11 002a0000 00140000
I (109) boot: End of partition table
I (114) esp_image: segment 0: paddr=00020020 vaddr=3c020020 size=07e60h ( 32352) map
I (127) esp_image: segment 1: paddr=00027e88 vaddr=3fc8d600 size=02920h ( 10528) load
I (132) esp_image: segment 2: paddr=0002a7b0 vaddr=40380000 size=05868h ( 22632) load
I (143) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=18448h ( 99400) map
I (163) esp_image: segment 4: paddr=00048470 vaddr=40385868 size=07c5ch ( 31836) load
I (176) esp_image: segment 5: paddr=000500d4 vaddr=50000010 size=00010h (    16) load
I (180) boot: Loaded app from partition at offset 0x20000
I (180) boot: Disabling RNG early entropy source...
I (196) cpu_start: Pro cpu up.
I (204) cpu_start: Pro cpu start user code
I (204) cpu_start: cpu freq: 80000000 Hz
I (205) cpu_start: Application information:
I (207) cpu_start: Project name:     template
I (212) cpu_start: App version:      1
I (217) cpu_start: Compile time:     Mar  5 2022 14:32:09
I (223) cpu_start: ELF file SHA256:  0384805c96134d09...
I (229) cpu_start: ESP-IDF:          v5.0-dev-1730-g229ed08484-dirty
I (236) heap_init: Initializing. RAM available for dynamic allocation:
I (243) heap_init: At 3FC91690 len 0002E970 (186 KiB): DRAM
I (249) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (256) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (263) spi_flash: detected chip: generic
I (267) spi_flash: flash io: dio
I (282) sleep: Configure to isolate all GPIO pins in sleep state
I (283) esp_core_dump_uart: Init core dump to UART
I (283) coexist: coexist rom version 9387209
I (289) cpu_start: Starting scheduler.
I (293) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (293) sleep: Enable automatic switching of GPIO sleep configuration
I (1303) example: Loop
I (2313) example: Loop
I (3323) example: Loop
I (4333) example: Loop
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[template.zip](https://github.com/espressif/esp-idf/files/8190779/template.zip)

HiFiPhile, 2022-04-22:
Hello, any investigation on this ?

AxelLin, 2022-05-04:
The same issue on v4.3 branch (v4.3.2-555-gb4ef1b80a2).

AxelLin, 2022-05-23:
Hi @igrr 
Is someone working on this issue now? https://github.com/espressif/esp-idf/issues/8884#issuecomment-1121153725

MMucka, 2022-05-24:
I have got the same problem, but with manual light sleep:

```
esp_sleep_enable_timer_wakeup(ms * 1000);
esp_light_sleep_start();
```

after this, the USB serial connection stops communicating. 
Device is still connected, present in Device Manager, but not sending data.

Is there any progress?

MMucka, 2022-06-13:
Hi, is there anything new on this issue? 

ESP32 C3 should be power efficient MCU, but Power Management is not working, if I want to watch logs on USB output.
I tried to disable "Power Down CPU in light sleep", but it didn't make any difference.
I always have to set esp_pm_config_esp32c3_t.light_sleep_enable = false to be able watching serial output, or JTAG debug.

AxelLin, 2022-06-13:
+1, this issue makes it difficult to debug with light_sleep_enable.

AxelLin, 2022-06-14:
When CONFIG_PM_DFS_INIT_AUTO is set, the .light_sleep_enable setting is always false.
Any reason to not provide config option for enabling .light_sleep_enable with CONFIG_PM_DFS_INIT_AUTO?

igrr, 2022-06-14:
@AxelLin on the last question, the CONFIG_PM_DFS_INIT_AUTO option isn't intended to cover all the use cases.
The code required to enabled DFS and light sleep at run time is very simple (just 5 lines), so I doubt it is necessary to add yet another Kconfig option for this.

ginkgm, 2022-07-12:
Hi @HiFiPhile , @AxelLin , @MMucka ,

The USB peripherals (USB OTG, USB serial JTAG) on all ESP chips till now are not designed to be used in the sleep modes. In the sleep modes, the clock of CPU is turned off and the APB clock which is associated with the CPU clock is also turned off.

The USB peripherals depends on the APB. When the clock is turned off, the peripheral will fail to return even a NAK to the host. The host will detect this, and have different abnormal actions.

I think the documentation of CONFIG_RTC_CLOCK_BBPLL_POWER_ON_WITH_USB is misleading, we will correct this.

AxelLin, 2022-07-12:
@ginkgm 
This issue makes it very difficult to debug program with light-sleep enabled.
The document clearly mentioned that: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html
If the application enters light-sleep (including automatic light-sleep) or software reset, etc. The USB CDC device will still work on the system. 

So it takes several months to confirm a documentation issue?

ginkgm, 2022-07-13:
@alisitsyn 
sorry, it was our mistake to write the documents carelessly. As it's a mix area issue (sleep & usb), it really takes some time to know how's the hardware designed and its behavior in sleep modes..

AxelLin, 2022-10-18:
@ginkgm 
The status is changed to "Selected for Development", does that mean someone is working on this support?

AxelLin, 2022-11-16:
> I have got the same problem with light sleep. But also with manual light sleep, the problem is the same :
> 
> ```
> esp_sleep_enable_timer_wakeup(ms * 1000);
> esp_light_sleep_start();
> ```
> 
> after this, the USB serial connection stops communicating. Device is still connected, present in Device Manager, but not sending data.
> 
> Is there any progress?

@ginkgm 

Hi @ginkgm,
I'm corious what was fixed?
I test with both master and v5.0 branche, It seems working now with idf.py monitor.
After calling esp_light_sleep_start(); and wakeup, I can still see the debug log.

ginkgm, 2022-11-28:
I only updated the document, and I believe that this feature can't be reliable due to the clock limitation.

Did you use another USB host? What we suspect that happen is:

1. The chip goes into sleep mode, clock source of USB get switched off
2. USB can't reply with even a NAK to ALL host packets
3. The host send some packets to the device regularly, and find the device is not responding to these packets in a valid way. (NAK is a valid response)
4. The host found this error, and choose one approach to handle the error. One possible way is to halt the endpoint that doesn't respond.

It's possible that your device wake up fast enough before the host found this issue, or the host just ignore the error and continue working. But I think these condition are not reliable.

closed 2023-02-28 by espressif-bot

AxelLin, 2023-02-28:
The fix seems not helpful at all.
The original report is about "esp32c3: USB stopped working when light sleep is enabled." and
it expected USB keep working under light sleep.
But the fix is avoid automatic light-sleep when USB CDC port is used, this does not fix this issue.

igrr, 2023-02-28:
@AxelLin As mentioned in https://github.com/espressif/esp-idf/issues/8507#issuecomment-1181383637, this is a hardware limitation — there is no way for USB_SERIAL_JTAG to [reliably](https://github.com/espressif/esp-idf/issues/8507#issuecomment-1328709695) work when light sleep mode is used. So the commit offers the only viable compromise, given what the hardware can do — offers an option to disable light sleep when USB is used.

Of course, we can keep the Github issue option since the original problem is not fixed. But it would be a bit pointless, since there is nothing ESP-IDF project can do in the future to make the ESP32-C3 work as expected by the issue reporter.

HudsonReynolds2, 2023-12-05:
I am using arduino IDE and esp32c3 SEEED STUDIO. I have been struggling with the same issue :(