espressif/esp-idf #8508: To exit from IDF monitor please use "Ctrl+]" not working

diplfranzhoepfinger, 2022-03-06:
how to exit IDF-Monitor ? 

To exit from IDF monitor please use "Ctrl+]"
is quite misleading. 

what does the Customer have to press ? 

the character "]" is on the key 9 as the 3rd Function. 

so pressing 9 makes 9. 
pressing Shift+9 makes )
pressing AltGr and 9 makes ]

so one must press Ctrl+AltGr+9
why 3 keys ? 

moreover it does not work ... any compbination of this 3keys does never quit monitor. 

before we were able to exit it with Ctrl+C. 
this is also no more possible now.

negativekelvin, 2022-03-06:
I guess this problem is specific to certain language keyboards like German and Nordic languages?

igrr, 2022-03-06:
@diplfranzhoepfinger you can use Ctrl-T Ctrl-X to exit, if Ctrl-] doesn't work: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#keyboard-shortcuts
(ref. https://github.com/espressif/esp-idf/issues/5233)

diplfranzhoepfinger, 2022-03-06:
Ctrl-T Ctrl-X work !

please: add this hint to the Message if one type Ctrl-C, 

add a hint to the page above. for me it was not clear that you must type Ctrl-T Ctrl-X.

diplfranzhoepfinger, 2022-03-06:
> I guess this problem is specific to certain language keyboards like German and Nordic languages?

YES! 

English Keyboard no Problem. 

but where to get ? if not steal ?

ddomnik, 2025-02-17:
On Windows 11 using a German keyboard layout, only Ctrl+T+X all together worked.