espressif/esp-idf #8510: Guru Meditation Error. I2C driver with commands of only one byte

armandr, 2022-03-06:
## Environment
- Development Kit:      ESP32-DevKitC
- Kit version [v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-IESP32]
- IDF version: v4.4 - issue may be related to changes in 4.3
 - Build System:         [CMake|idf.py]
- Compiler version: latest
 - Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB]

## Problem Description
After upgrading the SDK from 4.0 to 4.4 receiving Guru panics with our I2C driver.

//Detailed problem description goes here.
With our I2C peripheral device we issue a command to start performing work (not shown).

While the I2C peripheral is still working, it responds NACK to its address (+ read) and ACK when the work is complete and ready to be read. 

To check the completion of the work, we issue a single byte command as shown below: 

```cpp

cmd = i2c_cmd_link_create();
 i2c_master_start(cmd);
 i2c_master_write_byte(cmd, (DEVICE_ADDRESS << 1) | I2C_MASTER_READ, ACK_EN);
 ret = i2c_master_cmd_begin(I2C_NUM, cmd, 2000 / portTICK_PERIOD_MS);
 i2c_cmd_link_delete(cmd);

if (ret == ESP_OK)
    work_is_complete();
```

### Expected Behavior
To get either ESP_OK or NACK.

### Actual Behavior
```cpp
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008ab5f  PS      : 0x00060b30  A0      : 0x801322d5  A1      : 0x3ffbc730  
0x4008ab5f: i2c_master_cmd_begin_static at /Users/user/esp/esp-idf/components/driver/i2c.c:1345

A2      : 0x00000000  A3      : 0x3ffdf098  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0xffffff00  A7      : 0x6001301c  A8      : 0x3ffdf1fc  A9      : 0x3ff53000  
A10     : 0x00000901  A11     : 0x0000008b  A12     : 0x3ffbf22c  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000800  SAR     : 0x00000009  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x4008ab5c:0x3ffbc7300x401322d2:0x3ffbc770 0x400da1ee:0x3ffbc7b0 0x400da357:0x3ffbc7f0 0x400d83f3:0x3ffbc810 0x401b73fb:0x3ffbc930 
0x4008ab5c: i2c_master_cmd_begin_static at /Users/user/esp/esp-idf/components/driver/i2c.c:1345

0x401322d2: i2c_master_cmd_begin at /Users/user/esp/esp-idf/components/driver/i2c.c:1446
```

i2.c snippet:

```cpp
1344                p_i2c->cmd_link.head = p_i2c->cmd_link.head->next;
 1345 ->               p_i2c->cmd_link.head->cmd.bytes_used = 0;
```

I have not scrutinized the i2c.c file, but it's possible head->next is NULL given that we only have one command.

### Steps to reproduce

Craft a similar cmd.

## Debug Logs

negativekelvin, 2022-03-06:
https://github.com/espressif/esp-idf/issues/8348

o-marshmallow, 2022-03-07:
Hi @armandr ,

As @negativekelvin showed, this is due to the absence of a `stop` at the end of the transaction, this has been fixed in the other thread, the commit is number for v4.4 is f60a914bae19caf201da5188d391641bc29c1164