espressif/esp-idf #8511: customizable abort() message and restart delay

moefear85, 2022-03-06:
**Is your feature request related to a problem? Please describe.**

It is a bit confusing when my own calls to abort() trigger a backtrace automatically, since it pushes away my custom error messages using ESP_LOGE(). Moreover, If the chip restarts immediately, then I have no time to read anything, since often new output is printed to the screen very fast. Configuring the panic handler to halt instead is not always a solution, since one would like the chip then to save my custom error messages as well as optionally the backtrace to non-volatile storage, and then reboot, but only after sufficient time has passed to absorb the errors visually during debugging.

**Describe the solution you'd like**

There should be a configurable timer option as to how long the chip waits before resetting after abort() or other errors, and abort() should take an optional parameter to configure what is displayed and how verbose, including the backtrace as well as a custom message to the debugger. Ideally the delay time should be configurable in the abort call itself, so the programmer can decide based on abort() reason, whether to wait at all or how long. Also, a parameter to decide if/when to save the current abort info to disk or not, since it might not be needed for some cases, but very important for others.

**Describe alternatives you've considered**

There aren't any alternatives to consider, as far as I know.

**Additional context**

None necessary.

o-marshmallow, 2022-03-07:
Hi @moefear85 ,

In the `menuconfig`, you can already set the behaviour on panic:
* Print registers and halt
* Print registers and reboot
* Silent reboot

These can be found in `Component config -> ESP System Settings -> Panic handler behaviour`. The first one may suit your needs the best as you will be able to see the message you are logging.

If you are still willing to replace or override the panic handler, this is totally possible, I already wrote a guide about it here: https://github.com/espressif/esp-idf/issues/7681#issuecomment-941957783

You can pretty much override any non-static/non-inlined function, in particular, `panic_abort` one.

moefear85, 2022-03-07:
ok thx

closed 2022-03-07 by moefear85