espressif/esp-idf #8512: Notifications/Flags in case any IO ringbuffers overflow (specifically wifi)

moefear85, 2022-03-06:
**Is your feature request related to a problem? Please describe.**

When using promiscuous mode directly, I assume my callback is getting called directly whenever a packet is placed by the wifi chip/driver into a ringbuffer using DMA. IE I assume it behaves analogously to the UART FIFOs. However, I am not aware of any way of knowing if, due to excessive processing elsewhere, any packets are dropped/lost due to a full buffer.

**Describe the solution you'd like**

It would be nice if there were events or a flag or callbacks that would indicate to the user if any packet loss in the buffer/driver has occured during promiscuous mode.

**Describe alternatives you've considered**

The UART API is very mature and feature-rich. It would be nice if there was a direct analog to it for wifi, both rx & tx. IE it would be nice to have a uniformely named API for read and write, as opposed to esp_wifi_80211_tx and esp_wifi_set_promiscuous_rx_cb. Rather something like:

wifi_read_packet()
wifi_write_packet()
wifi_enable_tx_intr()
wifi_enable_rx_intr()
wifi_clear_intr_status()
wifi_enable_intr_mask()
wifi_wait_tx_done()
**wifi_set_rx_full_threshold()**
**wifi_set_tx_empty_threshold()**
wifi_set_rx_timeout()
**wifi_get_rx_overflow_flag()**
**wifi_reset_rx_overflow_flag()**

The wifi driver can in the background continue to verify integrity of the packet & enforce necessary security aspects, while still being as flexible as esp_wifi_80211_tx currently is. Ideally it would also allow the user to encrypt/decrypt any packets for which it already knows the keys, and use hardware acceleration for this.

This gives the developer great freedom & flexibility in choosing which communications subsys/api to use. I personally greatly disdain tcp/ip in itself, thus anything built on it, such as lwIP, due to bloat & inefficiency, and excessive iterative compile times. If the user tries to use any other conflicting API, the library can simply present an error message indicating the user must choose one mode of operation or the other.

**Additional context**

N/A

jack0c, 2022-05-20:
@moefear85 promiscuous mode is actually not a normal mode, it cost lots of CPU performance. It is not suitable to have a callback when buffer full happens, since it will cause more buffer full. How about provide an API to get the buffer overflow counter?

moefear85, 2022-05-21:
@jack0c:

I've been doing alot of further coding & testing since first posting this. I can confirm using promiscuous mode & esp_wifi_80211_tx to send and recieve works perfectly and without any burden on the CPU. I'm getting much better flexibility & throughput than using wifi/lwIP.

As for the API question I posted, it's actually not so important, since I've realized how to efficiently use the existing mentioned functions. promiscuous takes a callback that runs as an ISR, and I just forward the packets to the default event queue to be processed at lower priority. It works amazing. I see transfer speeds of around 6MB/s. Ofcourse packet drops happen. Even if I catch these and resend, I still see average throughput of around 1.5MB/s, even if I have all stack protections & heap debugging options set to the most resource-intensive settings.

closed 2022-05-21 by moefear85