espressif/esp-idf #8514: Configuring incomplete, errors occurred!

zhy102, 2022-03-07:
OS: [Ubuntu]
ESP-IDF Version: [v4.3]

I have called ''git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git'' after ran "git submodule update --remote". Errors:
zhy123@zhy123-Lenovo-XiaoXin-700-15ISK:~/esp$ git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git
正克隆到 'esp-idf'...
fatal: 无法访问 '[https://github.com/espressif/esp-idf.git/'：GnuTLS](https://github.com/espressif/esp-idf.git/'%EF%BC%9AGnuTLS) recv error (-110): The TLS connection was non-properly terminated.
So i called ''git clone -b release/v4.3 https://github.com/espressif/esp-idf.git'' instead of the ''git clone -b release/v4.3 --recursive https://github.com/espressif/esp-idf.git''. Is it okay?
However, the errors as follows when i run "idf.py menuconfig":
Executing action: menuconfig
Running cmake in directory /home/zhy123/esp/ESP32-CSI-Tool-master/active_ap/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/zhy123/esp/ESP32-CSI-Tool-master/active_ap"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Initialising new submodule components/asio/asio...
子模组 'components/asio/asio'（[https://github.com/espressif/asio.git）已对路径](https://github.com/espressif/asio.git%EF%BC%89%E5%B7%B2%E5%AF%B9%E8%B7%AF%E5%BE%84) 'components/asio/asio' 注册
正克隆到 '/home/zhy123/esp/esp-idf/components/asio/asio'...
error: RPC 失败。curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated.
fatal: 远端意外挂断了
fatal: 协议错误：坏的包头
fatal: 无法克隆 'https://github.com/espressif/asio.git' 到子模组路径 '/home/zhy123/esp/esp-idf/components/asio/asio'
克隆 'components/asio/asio' 失败。按计划重试
正克隆到 '/home/zhy123/esp/esp-idf/components/asio/asio'...
子模组路径 'components/asio/asio'：检出 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
子模组路径 'components/asio/asio'：检出 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
-- Initialising new submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
子模组 'components/bootloader/subproject/components/micro-ecc/micro-ecc'（[https://github.com/kmackay/micro-ecc.git）已对路径](https://github.com/kmackay/micro-ecc.git%EF%BC%89%E5%B7%B2%E5%AF%B9%E8%B7%AF%E5%BE%84) 'components/bootloader/subproject/components/micro-ecc/micro-ecc' 注册
正克隆到 '/home/zhy123/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
子模组路径 'components/bootloader/subproject/components/micro-ecc/micro-ecc'：检出 'd037ec89546fad14b5c4d5456c2e23a71e554966'
-- Initialising new submodule components/bt/controller/lib_esp32...
子模组 'components/bt/controller/lib_esp32'（[https://github.com/espressif/esp32-bt-lib.git）已对路径](https://github.com/espressif/esp32-bt-lib.git%EF%BC%89%E5%B7%B2%E5%AF%B9%E8%B7%AF%E5%BE%84) 'components/bt/controller/lib_esp32' 注册
正克隆到 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'...
fatal: 无法访问 '[https://github.com/espressif/esp32-bt-lib.git/'：GnuTLS](https://github.com/espressif/esp32-bt-lib.git/'%EF%BC%9AGnuTLS) recv error (-110): The TLS connection was non-properly terminated.
fatal: 无法克隆 'https://github.com/espressif/esp32-bt-lib.git' 到子模组路径 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'
克隆 'components/bt/controller/lib_esp32' 失败。按计划重试
正克隆到 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'...
fatal: 无法访问 '[https://github.com/espressif/esp32-bt-lib.git/'：GnuTLS](https://github.com/espressif/esp32-bt-lib.git/'%EF%BC%9AGnuTLS) recv error (-110): The TLS connection was non-properly terminated.
fatal: 无法克隆 'https://github.com/espressif/esp32-bt-lib.git' 到子模组路径 '/home/zhy123/esp/esp-idf/components/bt/controller/lib_esp32'
第二次尝试克隆 'components/bt/controller/lib_esp32' 失败，退出
CMake Error at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:48 (message):
Git submodule init failed for components/bt/controller/lib_esp32
Call Stack (most recent call first):
/home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
/home/zhy123/esp/esp-idf/tools/cmake/build.cmake:166 (__build_get_idf_git_revision)
/home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
/home/zhy123/esp/esp-idf/tools/cmake/project.cmake:7 (include)
CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1

And my version of cmake is version 3.23.0-rc2

wujiangang, 2022-03-07:
hi @zhy102, you can try the gitee mirror and tools, https://gitee.com/EspressifSystems/esp-gitee-tools, please have a try.

zhy102, 2022-03-07:
> hi @zhy102, you can try the gitee mirror and tools, https://gitee.com/EspressifSystems/esp-gitee-tools, please have a try.
I tried the gitee mirror and tools, then ran the code: ```
cd esp-gitee-tools
./submodule-update.sh ~/esp/esp-idf
```
However, the errors as follows:
zhy123@zhy123-Lenovo-XiaoXin-700-15ISK:~/esp/hello_world$ idf.py set-target esp32
```
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/zhy123/esp/hello_world/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/zhy123/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/zhy123/esp/hello_world"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Warning at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule
  components/bootloader/subproject/components/micro-ecc/micro-ecc is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:171 (__build_get_idf_git_revision)
  /home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:51 (__build_init)
  /home/zhy123/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c3_family is out of date.
  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:171 (__build_get_idf_git_revision)
  /home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:51 (__build_init)
  /home/zhy123/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/host/nimble/nimble is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:171 (__build_get_idf_git_revision)
  /home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:51 (__build_init)
  /home/zhy123/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/cmock/CMock is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:171 (__build_get_idf_git_revision)
  /home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:51 (__build_init)
  /home/zhy123/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_phy/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/zhy123/esp/esp-idf/tools/cmake/build.cmake:171 (__build_get_idf_git_revision)
  /home/zhy123/esp/esp-idf/tools/cmake/idf.cmake:51 (__build_init)
  /home/zhy123/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/zhy123/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esptool_py/esptool is out of date.  Run 'git
  submodule update --init --recursive' to fix.
[... 126 lines cut ...]
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
CMake Error at /home/zhy123/esp/esp-idf/tools/cmake/component.cmake:304 (message):
  Include directory
  '/home/zhy123/esp/esp-idf/components/mbedtls/mbedtls/include' is not a
  directory.
Call Stack (most recent call first):
  /home/zhy123/esp/esp-idf/tools/cmake/component.cmake:477 (__component_add_include_dirs)
  /home/zhy123/esp/esp-idf/components/mbedtls/CMakeLists.txt:10 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/zhy123/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code `1`
```
I tried the gitee mirror and tools, then ran the code: ```
cd esp-gitee-tools
./submodule-update.sh ~/esp/esp-idf
```
However, the errors as follows:

wujiangang, 2022-04-14:
@zhy102 Please check all submodules are updated. You can try `git submodule foreach --recursive git remote -v` to check whether all submodules are linkd to gitee. If already link to gitee, you can use `git submodule update --init --recursive` to update, if link to GitHub, then you need to run `./submodule-update.sh ~/esp/esp-idf` again.

closed 2022-12-08 by Alvin1Zhang