espressif/esp-idf #8515: Sending POST to https server panic if CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y

AxelLin, 2022-03-07:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v5.0-dev-1829-ge29e96101d
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I test sending POST to https://hookbin.com, then it always hits below panic if CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
Note, if CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=n it works.

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008e520  PS      : 0x00060b33  A0      : 0x8008ece5  A1      : 0x3ffdf2c0  
0x4008e520: remove_free_block at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:213
 (inlined by) block_locate_free at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:757

A2      : 0x3ffb93ac  A3      : 0x3ffdffa4  A4      : 0x00000010  A5      : 0x3ffdf410  
A6      : 0x871682ac  A7      : 0x00000001  A8      : 0x00000003  A9      : 0x00000000  
A10     : 0x3ffb93b8  A11     : 0x0000000c  A12     : 0x91b3e144  A13     : 0x53496bda  
A14     : 0x2086b79b  A15     : 0x53496bd8  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x91b3e150  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x4008e51d:0x3ffdf2c00x4008ece2:0x3ffdf2e0 0x40081f3f:0x3ffdf300 0x40081f51:0x3ffdf330 0x4008215d:0x3ffdf350 0x400856a0:0x3ffdf370 0x40144e2d:0x3ffdf390 0x4013cfd7:0x3ffdf3b0 0x4013d141:0x3ffdf3d0 0x40142831:0x3ffdf3f0 0x4013f94f:0x3ffdf410 0x4013fb39:0x3ffdf430 0x40139dbe:0x3ffdf450 0x401390de:0x3ffdf510 0x40139117:0x3ffdf530 0x401329d5:0x3ffdf550 0x401326bd:0x3ffdf570 0x40132797:0x3ffdf5a0 0x401338c3:0x3ffdf5d0 0x40192f07:0x3ffdf5f0 0x40134af8:0x3ffdf610 0x40134d8d:0x3ffdf630 0x400dac71:0x3ffdf650 0x400db219:0x3ffdf6c0 0x401951c3:0x3ffdf6e0 0x400db245:0x3ffdf730 
0x4008e51d: remove_free_block at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:212
 (inlined by) block_locate_free at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:757

0x4008ece2: multi_heap_malloc_impl at /home/axel/esp/esp-idf/components/heap/multi_heap.c:189

0x40081f3f: heap_caps_malloc_base at /home/axel/esp/esp-idf/components/heap/heap_caps.c:139

0x40081f51: heap_caps_malloc at /home/axel/esp/esp-idf/components/heap/heap_caps.c:159

0x4008215d: heap_caps_calloc at /home/axel/esp/esp-idf/components/heap/heap_caps.c:433

0x400856a0: esp_mbedtls_mem_calloc at /home/axel/esp/esp-idf/components/mbedtls/port/esp_mem.c:17

0x40144e2d: mbedtls_calloc at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/platform.c:61

0x4013cfd7: mbedtls_mpi_grow at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:129
 (inlined by) mbedtls_mpi_grow at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:119

0x4013d141: mbedtls_mpi_lset at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:283

0x40142831: ecp_use_curve25519 at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecp_curves.c:4646
 (inlined by) mbedtls_ecp_group_load at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecp_curves.c:4809

0x4013f94f: ecdh_setup_internal at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c:201
 (inlined by) mbedtls_ecdh_setup at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c:234

0x4013fb39: mbedtls_ecdh_setup_no_everest at ??:?

0x40139dbe: ssl_tls13_generate_and_write_ecdh_key_exchange at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:295
 (inlined by) ssl_tls13_write_key_share_ext at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:424
 (inlined by) ssl_tls13_write_client_hello_body at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:796
 (inlined by) ssl_tls13_write_client_hello at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:896
 (inlined by) mbedtls_ssl_tls13_handshake_client_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:1733

0x401390de: mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5250
 (inlined by) mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5225

0x40139117: mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5317
 (inlined by) mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5293

0x401329d5: esp_mbedtls_handshake at /home/axel/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:180

0x401326bd: esp_tls_handshake at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:79 (discriminator 15)
 (inlined by) esp_tls_low_level_conn at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:421 (discriminator 15)

0x40132797: esp_tls_conn_new_sync at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:483

0x401338c3: ssl_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport_ssl.c:111

0x40192f07: esp_transport_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport.c:173

0x40134af8: esp_http_client_connect at /home/axel/esp/esp-idf/components/esp_http_client/esp_http_client.c:1247
 (inlined by) esp_http_client_connect at /home/axel/esp/esp-idf/components/esp_http_client/esp_http_client.c:1219

0x40134d8d: esp_http_client_open at /home/axel/esp/esp-idf/components/esp_http_client/esp_http_client.c:1402

AxelLin, 2022-03-07:
assert failed: block_trim_free heap_tlsf.c:377 (block_is_free(block) && "block must be free")

Backtrace:0x40081de2:0x3ffdf2200x400896ed:0x3ffdf240 0x40090469:0x3ffdf260 0x4008e4c1:0x3ffdf380 0x4008ece2:0x3ffdf3a0 0x40081f3f:0x3ffdf3c0 0x40081f51:0x3ffdf3f0 0x4008215d:0x3ffdf410 0x400856a0:0x3ffdf430 0x40144e2d:0x3ffdf450 0x4013cfd7:0x3ffdf470 0x4013d141:0x3ffdf490 0x4014283f:0x3ffdf4b0 0x4013f94f:0x3ffdf4d0 0x4013fb39:0x3ffdf4f0 0x40139dbe:0x3ffdf510 0x401390de:0x3ffdf5d0 0x40139117:0x3ffdf5f0 0x401329d5:0x3ffdf610 0x401326bd:0x3ffdf630 0x40132797:0x3ffdf660 0x401338c3:0x3ffdf690 0x40192f07:0x3ffdf6b0 0x40134af8:0x3ffdf6d0 0x40134d8d:0x3ffdf6f0 0x400dac71:0x3ffdf710 0x400db219:0x3ffdf780 0x401951c3:0x3ffdf7a0 0x400db245:0x3ffdf7f0 
0x40081de2: panic_abort at /home/axel/esp/esp-idf/components/esp_system/panic.c:403

0x400896ed: esp_system_abort at /home/axel/esp/esp-idf/components/esp_system/esp_system.c:130

0x40090469: __assert_func at /home/axel/esp/esp-idf/components/newlib/assert.c:85

0x4008e4c1: block_locate_free at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:447
 (inlined by) tlsf_malloc at /home/axel/esp/esp-idf/components/heap/heap_tlsf.c:757

0x4008ece2: multi_heap_malloc_impl at /home/axel/esp/esp-idf/components/heap/multi_heap.c:189

0x40081f3f: heap_caps_malloc_base at /home/axel/esp/esp-idf/components/heap/heap_caps.c:139

0x40081f51: heap_caps_malloc at /home/axel/esp/esp-idf/components/heap/heap_caps.c:159

0x4008215d: heap_caps_calloc at /home/axel/esp/esp-idf/components/heap/heap_caps.c:433

0x400856a0: esp_mbedtls_mem_calloc at /home/axel/esp/esp-idf/components/mbedtls/port/esp_mem.c:17

0x40144e2d: mbedtls_calloc at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/platform.c:61

0x4013cfd7: mbedtls_mpi_grow at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:129
 (inlined by) mbedtls_mpi_grow at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:119

0x4013d141: mbedtls_mpi_lset at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:283

0x4014283f: ecp_use_curve25519 at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecp_curves.c:4649
 (inlined by) mbedtls_ecp_group_load at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecp_curves.c:4809

0x4013f94f: ecdh_setup_internal at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c:201
 (inlined by) mbedtls_ecdh_setup at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ecdh.c:234

0x4013fb39: mbedtls_ecdh_setup_no_everest at ??:?

0x40139dbe: ssl_tls13_generate_and_write_ecdh_key_exchange at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:295
 (inlined by) ssl_tls13_write_key_share_ext at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:424
 (inlined by) ssl_tls13_write_client_hello_body at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:796
 (inlined by) ssl_tls13_write_client_hello at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:896
 (inlined by) mbedtls_ssl_tls13_handshake_client_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_client.c:1733

0x401390de: mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5250
 (inlined by) mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5225

0x40139117: mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5317
 (inlined by) mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5293

0x401329d5: esp_mbedtls_handshake at /home/axel/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:180

0x401326bd: esp_tls_handshake at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:79 (discriminator 15)
 (inlined by) esp_tls_low_level_conn at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:421 (discriminator 15)

0x40132797: esp_tls_conn_new_sync at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:483

0x401338c3: ssl_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport_ssl.c:111

0x40192f07: esp_transport_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport.c:173

0x40134af8: esp_http_client_connect at /home/axel/esp/esp-idf/components/esp_http_client/esp_http_client.c:1247
 (inlined by) esp_http_client_connect at /home/axel/esp/esp-idf/components/esp_http_client/esp_http_client.c:1219

0x40134d8d: esp_http_client_open at /home/axel/esp/esp-idf/components/esp_http_client/esp_http_client.c:1402

laukik-hase, 2022-03-07:
Hello, @AxelLin!

Thank you for sending the report. I was investigating this issue but unfortunately could not reproduce it with the `esp_http_client` example. Rather than a `system abort`, I got the following logs (with `hookbin.com` and `api.github.com`) indicating a TLS extension mismatch. (The `mbedtls` support for TLS 1.3 is still developing - see [here](https://github.com/ARMmbed/mbedtls/blob/development/docs/architecture/tls13-support.md)).

```log
I (12022) http_client: Connected to AP, begin http example
E (16982) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7500
I (16992) esp-tls-mbedtls: (FFFF8B00): SSL - Client received an extended server hello containing an unsupported extension
I (16992) esp-tls-mbedtls: Certificate verified.
E (17002) esp-tls: Failed to open new connection
E (17002) TRANSPORT_BASE: Failed to open a new connection
E (17012) HTTP_CLIENT: Connection failed, sock < 0
E (17022) http_client: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
I (17032) http_client: HTTP_EVENT_DISCONNECTED
I (17032) http_client: Last esp error code: 0x801a
I (17042) http_client: Last mbedtls failure: 0x7500
I (17042) http_client: Finish http example
```

Could you please attach -
1. Minimal code snippet that reproduces the bug
2. The `sdkconfig` file
3. Debug logs (with `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`)

AxelLin, 2022-03-08:
Hi @laukik-hase 
You can use esp_http_client example code to reproduce the issue (without any code modification), just need to set below settings:
(I think the key point is to set CONFIG_MBEDTLS_DYNAMIC_BUFFER=y)

```
#
# mbedTLS
#
CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=y
# CONFIG_MBEDTLS_DEFAULT_MEM_ALLOC is not set
# CONFIG_MBEDTLS_CUSTOM_MEM_ALLOC is not set
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=16384
CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN=4096
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
# CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA is not set
# CONFIG_MBEDTLS_DEBUG is not set

#
# mbedTLS v3.x related
#
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
CONFIG_MBEDTLS_SSL_TLS1_3_COMPATIBILITY_MODE=y
# CONFIG_MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH is not set
# CONFIG_MBEDTLS_X509_TRUSTED_CERT_CALLBACK is not set
# CONFIG_MBEDTLS_SSL_CONTEXT_SERIALIZATION is not set
CONFIG_MBEDTLS_SSL_KEEP_PEER_CERTIFICATE=y
# end of mbedTLS v3.x related

#
# Certificate Bundle
#
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=y
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN is not set
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE is not set
# CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE is not set
# end of Certificate Bundle

# CONFIG_MBEDTLS_ECP_RESTARTABLE is not set
# CONFIG_MBEDTLS_CMAC_C is not set
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_HARDWARE_MPI=y
CONFIG_MBEDTLS_HARDWARE_SHA=y
CONFIG_MBEDTLS_ROM_MD5=y
# CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN is not set
# CONFIG_MBEDTLS_ATCA_HW_ECDSA_VERIFY is not set
CONFIG_MBEDTLS_HAVE_TIME=y
# CONFIG_MBEDTLS_HAVE_TIME_DATE is not set
CONFIG_MBEDTLS_ECDSA_DETERMINISTIC=y
CONFIG_MBEDTLS_SHA512_C=y
# CONFIG_MBEDTLS_TLS_SERVER_AND_CLIENT is not set
# CONFIG_MBEDTLS_TLS_SERVER_ONLY is not set
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
# CONFIG_MBEDTLS_TLS_DISABLED is not set
CONFIG_MBEDTLS_TLS_CLIENT=y
CONFIG_MBEDTLS_TLS_ENABLED=y

#
# TLS Key Exchange Methods
#
# CONFIG_MBEDTLS_PSK_MODES is not set
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_RSA=y
# end of TLS Key Exchange Methods

CONFIG_MBEDTLS_SSL_RENEGOTIATION=y
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=y
# CONFIG_MBEDTLS_SSL_PROTO_GMTSSL1_1 is not set
# CONFIG_MBEDTLS_SSL_PROTO_DTLS is not set
CONFIG_MBEDTLS_SSL_ALPN=y
CONFIG_MBEDTLS_CLIENT_SSL_SESSION_TICKETS=y
CONFIG_MBEDTLS_SERVER_SSL_SESSION_TICKETS=y

```

AxelLin, 2022-03-08:
BTW,  I always got compile error with bellow config:

```
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
# CONFIG_MBEDTLS_HARDWARE_SHA is not set
```

Must set CONFIG_MBEDTLS_HARDWARE_SHA=y.

AxelLin, 2022-03-08:
> Hello, @AxelLin!
> 
> (The `mbedtls` support for TLS 1.3 is still developing - see [here](https://github.com/ARMmbed/mbedtls/blob/development/docs/architecture/tls13-support.md)).
> ```

@laukik-hase 

I understand the TLS 1.3 is in developing state.
It's difficult to know which part is already supported (should be working), and cannot differentiate the non-working part is becasue of bug or just not yet supported.
Do you have test cases to verify the working part of TLS1.3?

laukik-hase, 2022-03-14:
Hello, @AxelLin

The commit https://github.com/espressif/esp-idf/commit/87d3296c37a35caad790cb172d4441fcfb689d22 disables the usage of `MBEDTLS_DYNAMIC_BUFFER` when TLS 1.3 is enabled. Support for this is in progress and will be merged to `ESP-IDF` soon. 

> It's difficult to know which part is already supported (should be working), and cannot differentiate the non-working part is becasue of bug or just not yet supported.
Do you have test cases to verify the working part of TLS1.3?

We too are exploring more on TLS 1.3 support, and it has been difficult due to very few test endpoints being able to connect to `mbedtls` TLS 1.3 clients. The only endpoint we found was `google.com` (gives the `HTTP/1.1 301 Moved Permanently` response) but it is a functional example and can be used to test some of the features.

Sorry for the inconvenience.

AxelLin, 2022-04-07:
@laukik-hase 

I try to compile a third party libary but got build error as below in master tree.
(It was fine to compile with esp-idf stable branches)

/home/axel/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/private_access.h:27:33: error: 'mbedtls_x509_crt' {aka 'const struct mbedtls_x509_crt'} has no member named 'private_version'; did you mean 'private_sig'?
 #define MBEDTLS_PRIVATE(member) private_##member
                                 ^~~~~~~~

I check the code in esp-idf/components/mbedtls, only some file defines MBEDTLS_ALLOW_PRIVATE_ACCESS.
So now all the source file that includes private_access.h must define MBEDTLS_ALLOW_PRIVATE_ACCESS explictly?

laukik-hase, 2022-04-08:
Hello, @AxelLin!

> So now all the source file that includes private_access.h must define MBEDTLS_ALLOW_PRIVATE_ACCESS explicitly?

Yes, you will have to define `MBEDTLS_ALLOW_PRIVATE_ACCESS` at the top of every file explicitly.

AxelLin, 2022-08-30:
> Hello, @AxelLin
> 
> The commit [87d3296](https://github.com/espressif/esp-idf/commit/87d3296c37a35caad790cb172d4441fcfb689d22) disables the usage of `MBEDTLS_DYNAMIC_BUFFER` when TLS 1.3 is enabled. Support for this is in progress and will be merged to `ESP-IDF` soon.

@laukik-hase 
What will be merged **soon**? It has been 5 months since you replied but I still don't find the fix.

laukik-hase, 2022-09-08:
Hello, @AxelLin!

Sorry for the late reply and the inconvenience.

1. Mbed TLS has now been upgraded to `v3.2.1`. We have tried the `https_request` example with `CONFIG_MBEDTLS_SSL_PROTO_TLS1_3` enabled with several test endpoints ([example list](https://github.com/tlswg/tlswg-wiki/blob/master/IMPLEMENTATIONS.md#test-servers)), but none of them was able to connect to Mbed TLS-based TLS 1.3 clients. Could you suggest any endpoints with which you have established a connection with TLS 1.3? If not, you can raise an issue [upstream](https://github.com/Mbed-TLS/mbedtls/issues) about this.
3. As Mbed TLS is being upgraded, the dynamic buffers feature is now subject to some refactoring. Mbed TLS is splitting up TLS 1.2 and 1.3 implementations and adding their own variable length buffers feature. Even the update from `v3.1.0` to `v3.2.1` broke the feature (as fixed in https://github.com/espressif/esp-idf/commit/9b290e3668a562ed279445131f4e523a370e05e5).

Thus, this request has been now pushed to the backlog queue. Sadly, we will have to wait until Mbed TLS's TLS 1.3 implementation gets stable.

AxelLin, 2023-04-08:
> Hello, @AxelLin!
> 
> Sorry for the late reply and the inconvenience.
> 
> 1. Mbed TLS has now been upgraded to `v3.2.1`. We have tried the `https_request` example with `CONFIG_MBEDTLS_SSL_PROTO_TLS1_3` enabled with several test endpoints ([example list](https://github.com/tlswg/tlswg-wiki/blob/master/IMPLEMENTATIONS.md#test-servers)), but none of them was able to connect to Mbed TLS-based TLS 1.3 clients. Could you suggest any endpoints with which you have established a connection with TLS 1.3? If not, you can raise an issue [upstream](https://github.com/Mbed-TLS/mbedtls/issues) about this.
> 2. As Mbed TLS is being upgraded, the dynamic buffers feature is now subject to some refactoring. Mbed TLS is splitting up TLS 1.2 and 1.3 implementations and adding their own variable length buffers feature. Even the update from `v3.1.0` to `v3.2.1` broke the feature (as fixed in [9b290e3](https://github.com/espressif/esp-idf/commit/9b290e3668a562ed279445131f4e523a370e05e5)).
> 
> Thus, this request has been now pushed to the backlog queue. Sadly, we will have to wait until Mbed TLS's TLS 1.3 implementation gets stable.

@laukik-hase 

The latest mbedTLS release is v3.4.0 now.
Do you have further update regarding TLS 1.3 support?

BTW, wolfssl also supports TLS 1.3, you can take a look how they test TLS 1.3.

Harshal5, 2023-04-17:
Hello @AxelLin,

As mentioned in the mbedtls documentation and it's workflow TLS1.3 uses PSA crypto APIs. Adding support for mbedtls' PSA crypto in ESP-IDF would be a major task and we are looking into it. Once the support is completely added, I think we could also support TLS1.3.

mahavirj, 2023-08-18:
@AxelLin 

We have now enabled and verified the TLS 1.3 with `https_request` and `simple_ota_example`. Relevant CI tests also have been added. Please find details in following commits:

cba923788ec2d4721ad7996db5c956a8e3305d23
200cf10373f3ab8cf1429a34d96cd98a249fc458

Please note that we are not planning to support dynamic buffer feature with TLS 1.3, as the upstream API is not yet stable. Easiest way to verify the use-case could be, host the OTA update file on `github.com`, enable TLS 1.3 and then verify OTA update example. This feature will be supported in IDF 5.2 release, as we are still evaluating memory profile and few other pending items.

Please do let us know if you run into any issues, otherwise we can close this ticket.

AxelLin, 2024-03-29:
> @AxelLin
> 
> We have now enabled and verified the TLS 1.3 with `https_request` and `simple_ota_example`. Relevant CI tests also have been added. Please find details in following commits:
> 
> [cba9237](https://github.com/espressif/esp-idf/commit/cba923788ec2d4721ad7996db5c956a8e3305d23) [200cf10](https://github.com/espressif/esp-idf/commit/200cf10373f3ab8cf1429a34d96cd98a249fc458)
> 
> Please note that we are not planning to support dynamic buffer feature with TLS 1.3, as the upstream API is not yet stable. 

Hi @mahavirj 

The Mbed TLS 3.6.0 LTS is released.
I'm wondering if you can check again for the dynamic buffer feature with TLS 1.3.
Since it's a LTS version, I think the API is stable for LTS branch.

And maybe you can also check if https://github.com/espressif/esp-idf/issues/8325 can be addressed by @LJYSP 's patch (https://github.com/espressif/esp-idf/issues/8325#issuecomment-1118134508).