espressif/esp-idf #8518: esp_wifi_set_bandwidth() function not working

Wasabinary, 2022-03-07:
Hi, 

I've been trying to change the ESP32 soft AP bandwidth using the function esp_wifi_set_bandwidth() (ESP-IDF v4.4). Unfortunately, it seems like the function does not work in my case, as it always return ESP_ERR_INVALID_ARG code everytime. I basically reused the [provisioning manager example ](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/wifi_prov_mgr/main/app_main.c), where i added the function after initializing the wifi: 

```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));   
esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
```

What am I doing wrong ?

Wasabinary, 2022-03-14:
Does anyone noticed the problem too? I'm kind of stuck at the moment

MaxwellAlan, 2022-03-24:
hi @Wasabinary 
The https://github.com/espressif/esp-idf/blob/master/examples/provisioning/wifi_prov_mgr/main/app_main.c actually does init softap, so before you call esp_wifi_set_bandwidth to config SoftAP, you need to init SoftAP like below code or you can try with SoftAP example: examples/wifi/getting_started/softAP/main/softap_example_main.c
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "EXAMPLE_ESP_WIFI_SSID",
            .ssid_len = strlen("EXAMPLE_ESP_WIFI_SSID"),
            .channel = 1,
            .password = "EXAMPLE_ESP_WIFI_PASS",
            .max_connection = 5,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK,
            .pmf_cfg = {
                    .required = false,
            },
        },
    };
    if (strlen("EXAMPLE_ESP_WIFI_PASS") == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20));
Thanks.

Alvin1Zhang, 2022-03-24:
Thanks for reporting, feel free to reopen.

closed 2022-03-24 by Alvin1Zhang

Wasabinary, 2022-03-25:
Thanks for your answer @MaxwellAlan. I was able to "override" provisioning manager bandwidth by adding the piece of code your mentioned above before calling _wifi_prov_mgr_start_provisioning_ function. Unfortunately, the wifi channel remains unchanged, even after changing _wifi_config.ap.channel_ field.

I don't want to tune ESP-IDF wifi provisioning manager by myself too much, so is there an API to actually choose the provisioning manager soft ap config (channel + bandwidth)? That would be very useful

MaxwellAlan, 2022-04-07:
Hi @Wasabinary 
If you are using only station mode or only softap mode, then you can config channel in the init or you can using esp_wifi_set_bandwidth and esp_wifi_set_channel to config bandwidth and primary channel or secondary channel.
If you are using sta+softap mode, then the softap bandwidth and channel will follow the station, which means if your station connecting to channel 1 bw20 ap, then you can't setting your softap's channel and bandwidth exception channel 1 and bw20.