espressif/esp-idf #8522: More info about the LCD i80_controller example

ZaneL, 2022-03-07:
Hello,

I'm looking for more info about this example:
[LVGL porting example (based on i80 interfaced LCD controller)](https://github.com/espressif/esp-idf/tree/5faf116d26d1f171b6fc422a3a8c9c0b184bc65b/examples/peripherals/lcd/i80_controller)

What is the FPS you were able to achieve when using the 8080 interface? It looks like only 10 FPS? And what size screen were you using? If you increase the PSRAM frequency, are you able to get anywhere near 60 FPS? What about if you just use the internal SRAM?

suda-morris, 2022-03-08:
Hi @ZaneL As you didn't mention which branch you're working on, let's take the master branch as an example.
1. ESP32 can generate the 8080 timing by I2S peripheral, the `esp_lcd` component will take care of it, and provide a general user API, naming i80 bus and i80 io device driver. You can find the driver in `esp_lcd_panel_io_i2s.c`
2. ESP32's I2S DMA can't use PSRAM, so you can see `EXAMPLE_LCD_I80_COLOR_IN_PSRAM` is only visible for ESP32S3, where the LCD DMA can fetch data from PSRAM
3. In the example, we locked down the FPS by a little for ESP32S3, by this Kconfig `CONFIG_LV_DISP_DEF_REFR_PERIOD`. This is because the example is default to enable the PSRAM if the target is select to ESP32S3. (you can check the `sdkconfig.defaults.esp32s3` file). And 10FPS should work on most of PSRAM frequency. 
4. For higher FPS, you can either i) adjust the PSRAM mode (like frequency, mode) ii) Using internal SRAM
5. This example uses a 240*280 LCD, because the author only has that LCD at hand, no saying the resolution limitation is that.

closed 2022-04-11 by suda-morris

chewhs00, 2022-09-18:
I tried to run it on an ESP32_DevkitC_V4 with 8M Flash ESP32-Wrover-E and 8M PSRAM. The module was not connected to any thing. With the example from espidf release/5.0 example from i80_example, the target was set to esp32, flash/ram were set accordingly based on the module used, GPIO pins were set based what are available from the module. The build was success, but the code crashed during heap allocation occurred in esp_lcd_new_panel_io_i80 from file components/esp_lcd/src/esp_lcd_panel_io_i2s.c. See below