espressif/esp-idf #8524: esp32-s3 cannot run as usb host mode with msc demokit! it crash when wait for USB stick connect and restart continous

hancunchou, 2022-03-08:
I want to use my ESP32-S3-USB-OTG dev board for usb master ,but it cannot work. my idf version is esp-idf v5.0.
 Please help me. this also problem happen with the Ai-Thinker dev board esp-s3-12k-kit.
  

    -----------------------------
ELF file SHA256: 94a2b630fffa4e21
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT) Saved PC:0x403753a4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1650
load:0x403b6000,len:0xb1c
load:0x403ba000,len:0x2f48
entry 0x403b622c
[0;32mI (29) boot: ESP-IDF v4.4-dirty 2nd stage bootloader[0m [0;32mI (29) boot: compile time 22:15:20[0m
[0;32mI (29) boot: chip revision: 0[0m
[0;32mI (31) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (36) boot.esp32s3: SPI Mode : DIO[0m
[0;32mI (41) boot.esp32s3: SPI Flash Size : 8MB[0m
[0;32mI (45) boot: Enabling RNG early entropy source...[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label
[0;32mI (62) boot: 0 nvs
[0;32mI (69) boot: 1 phy_init
[0;32mI (77) boot: 2 factory
[0;32mI (84) boot: End of partition table[0m
[0;32mI (88) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=358a8h (219304) map[0m
[0;32mI (136) esp_image: segment 1: paddr=000458d0 vaddr=3fc90430 size=01f2ch ( 7980) load[0m
[0;32mI (138) esp_image: segment 2: paddr=00047804 vaddr=40374000 size=08814h ( 34836) load[0m
[0;32mI (149) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=2d994h (186772) map[0m
[0;32mI (184) esp_image: segment 4: paddr=0007d9bc vaddr=4037c814 size=03c18h ( 15384) load[0m
[0;32mI (187) esp_image: segment 5: paddr=000815dc vaddr=50000000 size=00010h ( 16) load[0m
[0;32mI (194) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (196) boot: Disabling RNG early entropy source...[0m
[0;32mI (212) cpu_start: Pro cpu up.[0m
[0;32mI (212) cpu_start: Single core mode[0m [0;32mI (220) cpu_start: Pro cpu start user code[0m [0;32mI (221) cpu_start: cpu freq: 240000000[0m [0;32mI (221) cpu_start: Application information:[0m
Usage WiFi data RF data
Type ST Offset
01 02 00009000 00006000[0m
factory app
01 01 0000f000 00001000[0m 00 00 00010000 00100000[0m
[0;32mI (223) cpu_start: Project name:
[0;32mI (229) cpu_start: App version:
[0;32mI (234) cpu_start: Compile time:
[0;32mI (240) cpu_start: ELF file SHA256: 94a2b630fffa4e21...[0m [0;32mI (246) cpu_start: ESP-IDF: v4.4-dirty[0m
[0;32mI (251) heap_init: Initializing. RAM available for dynamic allocation:[0m [0;32mI (259) heap_init: At 3FC92CD0 len 0004D330 (308 KiB): D/IRAM[0m [0;32mI (265) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM[0m [0;32mI (272) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (278) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM[0m
msc_esp32_s3[0m 05c4977-dirty[0m
Mar 7 2022 16:30:24[0m
Length[0m
 [0;32mI (285) spi_flash: detected chip: gd[0m
[0;32mI (289) spi_flash: flash io: dio[0m
[0;32mI (293) cpu_start: Starting scheduler on PRO CPU.[0m [0;32mI (358) example: Waiting for USB stick to be connected[0m [0;32mI (608) HUB: HCD PORT EVENT : 1
[0m
[0;31mE (668) HUB: Short device descriptor transfer failed[0m
assert failed: enum_stage_cleanup_failed hub.c:459 (dequeued_enum_urb == p_hub_driver_obj- >single_thread.enum_urb)
Backtrace:0x403756c2:0x3fce0f700x40378919:0x3fce0f90 0x4037e955:0x3fce0fb0 0x420138f8:0x3fce10d0 0x42013a2f:0x3fce10f0 0x42013d76:0x3fce1110 0x4200ecc8:0x3fce1130 0x42005ad1:0x3fce1150 0x4037b32d:0x3fce1180
ELF file SHA256: 94a2b630fffa4e21
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT) Saved PC:0x403753a4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1650
load:0x403b6000,len:0xb1c
load:0x403ba000,len:0x2f48
entry 0x403b622c
[0;32mI (29) boot: ESP-IDF v4.4-dirty 2nd stage bootloader[0m [0;32mI (29) boot: compile time 22:15:20[0m
[0;32mI (29) boot: chip revision: 0[0m
[0;32mI (31) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (36) boot.esp32s3: SPI Mode : DIO[0m
[0;32mI (41) boot.esp32s3: SPI Flash Size : 8MB[0m
[0;32mI (45) boot: Enabling RNG early entropy source...[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label
[0;32mI (62) boot: 0 nvs
[0;32mI (69) boot: 1 phy_init
[0;32mI (77) boot: 2 factory
[0;32mI (84) boot: End of partition table[0m
[0;32mI (88) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=358a8h (219304) map[0m
[0;32mI (136) esp_image: segment 1: paddr=000458d0 vaddr=3fc90430 size=01f2ch ( 7980) load[0m
[0;32mI (138) esp_image: segment 2: paddr=00047804 vaddr=40374000 size=08814h ( 34836) load[0m
[0;32mI (149) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=2d994h (186772) map[0m
[0;32mI (184) esp_image: segment 4: paddr=0007d9bc vaddr=4037c814 size=03c18h ( 15384) load[0m
Usage WiFi data RF data
Type ST Offset
01 02 00009000 00006000[0m
factory app
01 01 0000f000 00001000[0m 00 00 00010000 00100000[0m
Length[0m

[0;32mI (187) esp_image: segment 5: paddr=000815dc vaddr=50000000 size=00010h ( 16) load[0m
[0;32mI (194) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (196) boot: Disabling RNG early entropy source...[0m
[0;32mI (212) cpu_start: Pro cpu up.[0m
[0;32mI (212) cpu_start: Single core mode[0m [0;32mI (220) cpu_start: Pro cpu start user code[0m [0;32mI (221) cpu_start: cpu freq: 240000000[0m [0;32mI (221) cpu_start: Application information:[0m
[0;32mI (223) cpu_start: Project name:
[0;32mI (229) cpu_start: App version:
[0;32mI (234) cpu_start: Compile time:
[0;32mI (240) cpu_start: ELF file SHA256: 94a2b630fffa4e21...[0m [0;32mI (246) cpu_start: ESP-IDF: v4.4-dirty[0m
[0;32mI (251) heap_init: Initializing. RAM available for dynamic allocation:[0m [0;32mI (259) heap_init: At 3FC92CD0 len 0004D330 (308 KiB): D/IRAM[0m [0;32mI (265) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM[0m [0;32mI (272) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (278) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM[0m [0;32mI (285) spi_flash: detected chip: gd[0m

hancunchou, 2022-03-10:
该问题升级到master版本后解决了

closed 2022-03-10 by hancunchou

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.