espressif/esp-idf #8527: C++ nested try/catch: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. ESP-IDF v4.4 / v5.0

ferbar, 2022-03-08:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32 (tested with different modules)
- IDF version: v5.0-dev-1318-g34c21cb693
- Build System:         idf.py
- Compiler version: (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

Nested exceptions + rethrow don't work if the outer try/catch has more than 2 catch blocks, so if `} catch(exceptiontype1) {` is removed the test code doesn't throw an excption/abort

```
try {
   try {
     throw some_exception();
   } catch(...) {
     throw;
   }
} catch(exceptiontype1) {
} catch(exceptiontype1) {
} catch(...) {
}
printf("try/catch test done");
```

### Expected Behavior

Normal C++ exception handling; Example code seems to work with only 2 catch() blocks!

### Actual Behavior

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

### Steps to reproduce

slightly modified hello-world example, hello_world_main.c renamed to hello_world_main.cpp; as well as in CMakeLists.txt; idf.py menuconfig -> Compiler options -> Enable C++ exceptions

see C++ code

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

#include <stdexcept>

extern "C" void app_main(void)
{
printf("Hello world!\n");
printf("ESP_IDF Version: %s\n", esp_get_idf_version());
try {
    try {
        printf("try / try\n");
        throw std::runtime_error("test runtime error");
    } catch(...) {
        printf("catch1\n");
        throw;
    }
} catch(const char *e) {
    printf("~~~~~ cought outer exception char * %s ~~~~\n", e);
} catch(const std::exception &e ) {
    printf("~~~~~ cought outer exception %s ~~~~\n", e.what());
} catch(...) {
    printf("~~~~~ all caught ~~~~\n");
}
printf("try/catch test done\n");

sleep(10);
abort();
// rest of hello_world.c

```
[hello_world_main.cpp.txt](https://github.com/espressif/esp-idf/files/8209392/hello_world_main.cpp.txt)

## Debug Logs

```
Hello world!
ESP_IDF Version: v5.0-dev-1318-g34c21cb693-dirty
try / try
catch1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000127a  PS      : 0x00060030  A0      : 0x800d7270  A1      : 0x3ffb5840  
A2      : 0x3f406b74  A3      : 0x00000000  A4      : 0x00060c20  A5      : 0x00060c23  
A6      : 0x00000031  A7      : 0x3ffb59f0  A8      : 0x00000063  A9      : 0x3ffb5820  
A10     : 0x3ffb2728  A11     : 0x3ffb2728  A12     : 0x00060c20  A13     : 0x00060023  
A14     : 0x432b2b00  A15     : 0x3ffb5a00  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x40001277:0x3ffb58400x400d726d:0x3ffb5850 0x400e6e37:0x3ffb5870 0x400d7f51:0x3ffb5890 0x400e6e03:0x3ffb58b0 0x400d7629:0x3ffb58e0 0x400d8589:0x3ffb5950 0x400d8747:0x3ffb5a00 0x400d6f3d:0x3ffb5aa0 0x400d4ecc:0x3ffb5ac0 0x400e72dc:0x3ffb5ae0 0x40087ac9:0x3ffb5b00 
0x400d726d: std::type_info::operator==(std::type_info const&) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/typeinfo:124
 (inlined by) std::type_info::__do_catch(std::type_info const*, void**, unsigned int) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/tinfo.cc:72   

0x400e6e37: __cxxabiv1::__pbase_type_info::__pointer_catch(__cxxabiv1::__pbase_type_info const*, void**, unsigned int) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/cxxabi.h:307                                                                                                                                                              
 (inlined by) __cxxabiv1::__pointer_type_info::__pointer_catch(__cxxabiv1::__pbase_type_info const*, void**, unsigned int) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/pointer_type_info.cc:51                                                                                                                                               

0x400d7f51: __cxxabiv1::__pbase_type_info::__do_catch(std::type_info const*, void**, unsigned int) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/pbase_type_info.cc:100
 (inlined by) __cxxabiv1::__pbase_type_info::__do_catch(std::type_info const*, void**, unsigned int) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/pbase_type_info.cc:32

0x400e6e03: get_adjusted_ptr(std::type_info const*, std::type_info const*, void**) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:228

0x400d7629: __gxx_personality_v0 at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:595

0x400d8589: _Unwind_RaiseException at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/unwind.inc:118

0x400d8747: _Unwind_Resume_or_Rethrow at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/unwind.inc:264

0x400d6f3d: __cxa_rethrow at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:126

0x400d4ecc: app_main at /home/chris/bin/esp/esp-idf/examples/get-started/hello_world/build/../main/hello_world_main.cpp:30

0x400e72dc: main_task at /home/chris/bin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x40087ac9: vPortTaskWrapper at /home/chris/bin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder) [sdkconfig.txt](https://github.com/espressif/esp-idf/files/8209388/sdkconfig.txt)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**) - can be provided if necessary 
- [ ] coredump (This provides stacks of tasks.)

Related to #5360 possible regression caused in fixes there.

igrr, 2022-03-08:
Hi @ferbar, given that the backtrace points to `std::type_info`, could you please try enabling `CONFIG_COMPILER_CXX_RTTI` as well?

ferbar, 2022-03-08:
@igrr, you are right, this doesn't crash:

```
Hello world!
ESP_IDF Version: v5.0-dev-1318-g34c21cb693-dirty
try / try
catch1
~~~~~ cought outer exception test runtime error ~~~~
try/catch test done

```

However the arduino libraries are compiled without RTTI, so changing the `CONFIG_COMPILER_CXX_RTTI` isn't that easy.

atanisoft, 2022-03-09:
@ferbar log an issue and/or submit a PR to https://github.com/espressif/esp32-arduino-lib-builder so that RTTI can be evaluated for enabling in Arduino.

ferbar, 2022-03-09:
(off topic) @atanisoft it's for this: https://github.com/ferbar/btcontrol/tree/master/control-esp32
I think RTTI is disabled intentionally in the arduino libs, I would not change it :-) Since this bug can be easily reproduced I would prefer a bugfix in unwind.inc os somewhere around.

0xjakob, 2022-03-10:
@ferbar @atanisoft C++ exception handling should generally work without RTTI enabled. Hence, enabling RTTI is more a workaround. I'll take a look at this by the end of this week or the beginning of next week.

In GCC, C++ exceptions and RTTI both use the same system to recognize C++ types, hence there might be some bug with correctly excluding the RTTI-based parts while still retaining the parts used by C++ exceptions.

0xjakob, 2022-03-21:
@ferbar I've taken a look last week but couldn't find out the root cause so far. We are suspecting some kind of compiler bug or a bug in our linker scripts or the linker itself. In particular, if you remove the second `printf` line `printf("ESP_IDF Version: %s\n", esp_get_idf_version());`, then the program runs fine again, even if you just print a simple string. 

We need further investigation for this.

0xjakob, 2022-03-21:
The issue does occur on all *Xtensa architectures*. On ESP32-C3 (*RISC-V architecture*), this problem does not occur with aforementioned code.

ferbar, 2022-03-21:
@0xjakob I didn't see the printf thing when I was testing. Looks as if there is something with the stack, maybe I found an easier way to reproduce the issue:

```
printf("string");
printf("string");
nested_try_catch_block
```
=> works

```
void *n=NULL
printf("string");
printf("string");
nested_try_catch_block
printf("test done");
printf("string %p", n);  // <- some usage of n that it doesn't get optimized out. 
```
=> does NOT work

0xjakob, 2022-04-26:
@ferbar Sorry for the long turnaround, but this issue took a while to investigate. I [opened a bug on GCC for this](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=105387) since it looks like an issue in `libsupc++` which is part of `libstdc++`.

BTW, a minimal reproducer would be:
```
#include <cstdio>
#include <stdexcept>

int main(int argc, char** argv) {
    try {
        throw std::runtime_error("test runtime error");
    } catch (const char *e) {
        printf("~~~~~ cought outer exception char * %s ~~~~\n", e);
    } catch (const std::exception &e) {
        printf("~~~~~ cought outer exception %s ~~~~\n", e.what());
    }
}
```

and then shuffle some `printf()`s around until you get the error. I.e., you don't need a nested try-catch, the important bit is that you have a catch statement with a pointer type as argument (`const char*` in the example) and a non-pointer type exception passing by (`std::runtime_error` in the example).

ferbar, 2022-04-26:
OMG thanks a lot!

If you still need to analyze SEGVs on x86 maybe valgrind can help you.

0xjakob, 2022-06-13:
@ferbar We merged the new toolchain with the fix for this bug (among other things) into master. You shouldn't be able to observe this particular bug anymore with the newest toolchain (2022r1). There's a slight caveat, though: catching an exception of pointer-to-member type will not work anymore if the standard library without RTTI is used. It's a consequence or rather tradeoff of the (trlsyibrly easy) fix. If catching this type of exception is unavoidable, then a standard library with RTTI has to be used. However, I don't think pointer-to-member types are used that often. Please let us know if there are any further problems related to this issue, otherwise we'd like to close it.

0xjakob, 2022-06-16:
@ferbar I'm closing this issue for now. We'll backport the fix to the older toolchains in the near future. If you still observe the bug on current master, please let us know!

closed 2022-06-16 by 0xjakob

ferbar, 2022-06-28:
Thanks, my test code is working as expected