espressif/esp-idf #8529: esp_vfs_l2tap_intf_register() not implemented.

djmordor, 2022-03-08:
I am trying to implement PTP using Netif as per documentation regarding L2 TAP interface.
google esp_vfs_l2tap_intf_register() returns hits only refering to the documentation. Am I supposed to implement esp_vfs_l2tap_intf_register() myself? Any clues?

kostaond, 2022-03-10:
Hi @djmordor we are currently unfortunately missing an example related to L2 TAP interface. I have opened tasks to create it but I am busy with other issues. However, you could get better picture of how to use it in L2 TAP [unit test](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/test/test_vfs_l2tap.c#L418).

Anyway, implementing PTP on ESP32 sounds interesting, it would be nice if you shared if you were successful. Are you planning to port PTPd?

djmordor, 2022-03-11:
I am but a novice really kostaond. I have done a little programming in D and python and swerved C and C ++. I have a couple of Olimex POE boards which I have been tinkering with for a few months. Most of that has been just learning the build system using WSL under Visual Studio Code on Windows11. Its taken me two days just to get the master branch working (cmake and intellisence problems).
I have PTPd for inspiration but I am hopefully going to build something more lightweight and direct than going through the entire lwip TCP/IP stack. Reading the L2 TAP documentation indicates that this is possible but I don't know if I am on the right track. I still haven't solved the mystery of the hardware timestamp yet but I have spent days looking at the ESP32 Technical Reference manual. Am I right in thinking that the timestamp is automatically inserted on frame reception/transmission? How do I access the outgoing timestamp? I have more questions than answers, most of what I have been looking at is completely new and alien, so don't expect any timely or professional results ; ) End goal is AES67 cos I'm an audio nut. The I2S concept makes more sense to me coming from an audio background. Ethernet is a new topic for me, but I like the challenge.

kostaond, 2022-03-11:
@djmordor 1588 standard defines various transport layers to be used. Ethernet (accessed via L2 TAP on ESP) is among them and PTPd has support for it but the implementation is platform specific. However, that's just a part of the problem. To be able to provide precise time stamping, the HW support is needed. ESP32 EMAC should have support for 1588 time stamping but I am not exactly sure about its capabilities. I haven't had time to fully investigate (test) yet since PTP does not have high priority for us right now. In general, time stamps should be available in Tx/Rx Ethernet DMA descriptors when time stamping enabled. It is straight forward for Rx. For Tx, you firstly copy data and configure the Tx descriptor, then command to transmit and read back the same descriptor which should be updated with the status of previous Tx operation. However, as I said, I've never tried it so I don't know how exactly it behaves. Therefore, the first thing to do should be ensuring ESP32 time stamping capabilities, then update Ethernet driver to be able to provide time stamps to SW layer and then port PTPd. In any case, it is quite complex task...