espressif/esp-idf #8530: The USB slave device of esp32-s3 chip cannot be recognized normally in windows11 and deepin OS

maplerian, 2022-03-09:
Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue. 

## Environment
- Development Kit:      [ESP32-S3-DevKitM-1]
- Module or chip used:  [ESP32-S3-MINI-1]
- IDF version : v5.0-dev-1881-g6ddd6e6d29
- Build System:         [idf.py]
- Compiler version: xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows11]
- (Windows only) environment type: VSCode
- Using an IDE:  master
- Power Supply:         [USB]

## Problem Description
Reset to download mode, install the driver according to the official tutorial USB download normal, USB ID identification normal, use examples\peripherals\ USB \device\tusb_console, set to ESP32S3 chip. After compiling, download the program through USB without error. After the update is successful, power off and re-insert USB, and the system cannot identify USB device. Vendor ID: 0x0000, Product ID:0x0002, even if zadig-2.7.exe is used to update the driver for it, it cannot be used normally.

### Expected Behavior
After the USB is inserted, it can be identified normally
### Actual Behavior
After the USB is inserted, it cannot be identified properly
### Steps to reproduce
1. cd examples\peripherals\ USB \device\tusb_console
2. idf.py set-target esp32s3
3. idf.py build
4. idf.py dfu
5. idf.py dfu-flash
6. Unplug the USB cable and plug in the USB again
## Debug Logs

```
...
I (270) sleep: Configure to isolate all GPIO pins in sleep state
I (277) sleep: Enable automatic switching of GPIO sleep configuration
I (284) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) example: USB initialization
I (304) tusb_desc:
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 0           │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 0           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 0           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x3000      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x101       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (474) TinyUSB: TinyUSB Driver installed
I (474) example: USB initialization DONE
I (484) example: log -> UART
example: print -> stdout
example: print -> stderr
I (6484) example: log -> UART
example: print -> stdout
example: print -> stderr
...
```

maplerian, 2022-03-09:
Today, I just bought the mass production version of S3 chip development board from Anxin Ke. After the same operation, it can be used normally. Therefore, do the S3 chips before mass production have such problems?

FanhuaCloud, 2022-04-04:
Hello, when did you buy your S3 mod, I have the same problem.

frankcohen, 2022-05-12:
I uploaded a sketch and now the board (Adafruit Feather ESP32-S3) does not register as a USB device on my MacBook Pro. Pressing the Boot and Reset buttons on the board have no change. The Neopixel LED turns-on (that's not my sketch doing that) on boot. I may have encountered this issue.

frankcohen, 2022-05-12:
> 

Solved this by powering the board (connecting the USB C port to my laptop) and pressing both Boot and Reset simultaneously. Now the board registers as a USB device on my laptop and in Arduino IDE.

tore-espressif, 2022-06-02:
Sorry no one got back to you earlier. However, I couldn't reproduce the issue. Could you please describe what behavior you see when the device 'cannot be identified properly'?

Could you also share your chip revision? You should see it during boot in the serial console., something like this:
```
I (25) boot: compile time 13:55:29
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
```

maplerian, 2022-06-06:
> 抱歉没有人早点回复你。但是，我无法重现该问题。您能否描述一下当设备“无法正确识别”时您看到的行为？
> 
> 你也可以分享你的芯片版本吗？您应该在引导期间在串行控制台中看到它，如下所示：
> 
> ```
> I (25) boot: compile time 13:55:29
> I (25) boot: chip revision: 0
> I (28) boot.esp32s3: Boot SPI Speed : 80MHz
> ```

I'm very sorry that the old version of the chip no longer exists, and in the new version of the chip, there is no such problem again

Alvin1Zhang, 2022-07-20:
Thanks for reporting, feel free to reopen. 感谢反馈，如有问题请重新打开此问题，谢谢。

closed 2022-07-20 by Alvin1Zhang

Adancurusul, 2023-03-06:
I also have the same problem, can the new version of the chip solve this problem？

FanhuaCloud, 2023-03-06:
> I also have the same problem, can the new version of the chip solve this problem？

我的问题是无法使用USB DFU，貌似是因为旧版芯片efuse的问题，根据文档，新版已经修复这个错误

Adancurusul, 2023-03-06:
邮件已收到，谢谢