espressif/esp-idf #8533: ESP32s3: high power consumption in deep sleep and with 1 running core

tbultel, 2022-03-09:
## Environment

- Development Kit:      [ESP32S3-Wroom-1]
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         [idf.py]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

When using one core, measured consumption is 79.56mA  (was 31mA on my former setup on ESP32)
(see example code below)
In deep sleep, it is 0.8mA 

### Expected Behavior

According to the reference manual, it should be between 28.4 mA and 42.6 mA at 80 Mhz
Notice that in deep sleep, in should be 7 to 8 uA

### Actual Behavior
as described in problem description

### Steps to reproduce

use code below

### Code to reproduce this issue

1 Core 100% busy: 

```cpp
...
#include "esp_log.h"

void app_main()
{
  setCpuFrequencyMhz(80);
  while(1) { }
}

```

jack0c, 2022-04-27:
Sorry, since the lock down in ShangHai, we can't do the test. Will update after we are back to the office.

lennyJL, 2022-07-05:
Hi @tbultel. I am looking into this issue. But I can not reproduce:
1. I use `deep_sleep` example with CONFIG_FREERTOS_UNICORE set, and I get 5-6 uA.
2. I run `while (1)` with 80M, and I get 35-36 mA.

I do above test at commit `v4.4 tag`. Can you provide more information:
1. Do you have more esp32s3 development boards? Do all boards have this issue?
2. How did you measure the current? Onboard peripherals may also consume power, and we need to ensure that the measured power consumption comes only from the chip.
3. Your Kconfig file?

Alvin1Zhang, 2022-11-07:
Thanks for reporting, would close due to short of feedback, feel free to reopen with more updates.

closed 2022-11-07 by Alvin1Zhang