espressif/esp-idf #8534: SPI driver ST95 missing/shift 1 bit on reception line

ElieSKickMaker, 2022-03-09:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32S2 DevKitC1-1

- Module or chip used:  ESP32S2FH4
- IDF version (run ``git describe --tags`` to find it):
    v4.3.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)] VS Code
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
I'm coding a driver ESP32 for the nfc reader ST95HF by spi. It's SPI mode 3 (CPOL=1, CPHA =1). 

The problem is when I received data, it's shifted by 1 like it's missing one clock edge. 
For exemple, the echo command is sending 0x55 and I should received 0x55 by the slave. But the esp32 is reading 0xAA.  
Here is the logic analyzer view and the code for initialize spi bus, add the device and do the spi transmission. 

** Initialize spi bus **

```
spi_bus_config_t buscfg={
            .mosi_io_num=35,
            .miso_io_num = 37,
            .sclk_io_num=CONFIG_EINK_SPI_CLK,
            .quadwp_io_num=-1,
            .quadhd_io_num=-1,
            .max_transfer_sz= 4094
        };

        //Initialize the SPI bus
       //ret=spi_bus_initialize(SPI_HOST, &buscfg, SPI_HOST);
        ret=spi_bus_initialize(SPI_HOST, &buscfg, 0); /* try to disable dma */
        ESP_ERROR_CHECK(ret);
```

** Configure the spi device **

```
//Config Frequency and SS GPIO
[... 153 lines cut ...]
I (4604) SPI_Recv:  data send : 2 data recv[0]:1c  data shift:e
I (4614) SPI_Recv:  data send : ff data recv[0]:aa  data shift:55
I (4614) SPI_Recv:  data send : ff data recv[0]:0  data shift:0
I (4624) SPI_Recv:  data send : 1 data recv[0]:c  data shift:6
I (4654) SPI_Recv:  data send : 0 data recv[0]:c  data shift:6
I (4654) SPI_Recv:  data send : 55 data recv[0]:c  data shift:6
I (4654) drv95HF_ReceiveSPI: cs low
I (4654) SPI_Recv:  data send : 2 data recv[0]:1c  data shift:e
I (4664) SPI_Recv:  data send : ff data recv[0]:aa  data shift:55
I (4664) SPI_Recv:  data send : ff data recv[0]:0  data shift:0
I (4674) SPI_Recv:  data send : 1 data recv[0]:c  data shift:6
I (4704) SPI_Recv:  data send : 0 data recv[0]:c  data shift:6
I (4704) SPI_Recv:  data send : 55 data recv[0]:c  data shift:6
I (4704) drv95HF_ReceiveSPI: cs low
I (4704) SPI_Recv:  data send : 2 data recv[0]:1c  data shift:e
I (4714) SPI_Recv:  data send : ff data recv[0]:aa  data shift:55
I (4714) SPI_Recv:  data send : ff data recv[0]:0  data shift:0
I (4724) SPI_Recv:  data send : 1 data recv[0]:c  data shift:6
E (4754) ConfigManager_PORsequence: PORERROR
E (4754) ConfigManager_HWInit: error PORsequence
E (4754) NFC_APP: Start NFC ERROR
```

[image]
[image]

KaeLL, 2022-03-09:
Any particular reason to use mode 3 over 0? Legitimately curious.
Try setting `spi_device_interface_config_t::cs_ena_pretrans` to 1 and see if it works.

ElieSKickMaker, 2022-03-09:
@KaeLL  I'm using mode 3 because I have a stm32 example working on mode 3 but it's also working on mode 0 due to the datasheet. I've also tried mode 0 (and 1 and 2 ..) ,same issue. 
No it's not working because I'm not in half duplex 

 " 
cs_ena_pretrans;        ///< Amount of SPI bit-cycles the cs should be activated before the transmission (0-16). This only works on half-duplex transactions.
"

KaeLL, 2022-03-09:
> No it's not working because I'm not in half duplex

What I said verbatim when I ported the TRF7970A driver, yet testing on the logic analyzer showed it worked.

KaeLL, 2022-03-09:
If by "falsed" you mean invalid, then yes. Mode 1, btw. The behavior seems to make sense if CPOL were to be 1, but that's not the case.
[image]
Making the cs line active 1 cycle earlier generates the correct result
[image]

KaeLL, 2022-03-09:
I don't know why setting `cs_ena_pretrans` works in full-duplex;
I'm not sure why the MISO line is idle high;
I'm don't know why `cs_ena_pretrans=0` causes all the oddities at transaction start in the first image.
The SPI code in IDF is still very confusing to me, which put me off of trying to look for answers. At this point, I'm not leaning towards my mental model of SPI being significantly wrong.

ginkgm, 2022-03-10:
Hi @ElieSKickMaker , 
I guess it's possible that @KaeLL 's opinion is quite close to the reason - ESP32's CS active edge is at the same moment as the first clock edge when cs_setup (cs_ena_pretrans) is kept 0. I guess ST95 detect mode 0 or 3 by the timing sequence of CS and clock edge. 

[image]

If the CS is pulled low when clock is high, it will give bit 6 at the second falling edge ; however if CS is pulled low when clock is low, it will give bit 6 at the first falling edge. 

Solutions: 
1. If you still want to use mode 3, we should give ST95 a determined clock state when CS is pulled low, so could you please also provide the waveform of CS (in total 4 wires) so that we can check it?

And btw, cs_ena_pretrans is the easiest way to solve this problem, and the limitation is a bit looser than 'only in half duplex mode'. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#known-issues. If the driver doesn't check it strictly, you may still set it to non-0 value and see what happen.

2. Or you may use mode 0. This may also directly solve the problem.

ElieSKickMaker, 2022-03-10:
Thanks for your response. 

So, I've tried to play  on cs_ena_pretrans (even on mode 0 also). I've still the same issue, it's receiving 0x6 but it's reading 0xC  
[image]

KaeLL, 2022-03-10:
Could that be due to the [MISO lag](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#timing-considerations)?

ginkgm, 2022-11-01:
Hi @ElieSKickMaker @KaeLL ,

1. the `0x06` (`0x0c`) is sent by ST95HF, right? But according to its spec, it always launch the data at the falling edge, no matter mode 0 or 3. Do you know what cause the difference between its spec and what we see? Actually it's half a clock ahead of spec.

2. According to SPI spec, in mode 3 both sides should latch (read) data at the rising edge. But we see there is transition on MISO at that edge, so the read data is undetermined. Both 0x06 or 0x0c are understandable. 

   To tolerate higher input delay, ESP32 hardware and driver latch the data a bit later (at the falling edge). This is usually OK if the slave launch the data according to the spec. But ST95HF also sends data half a clock ahead of spec, together they make one bit shift. You may try to force enable this block to delay the input by half an SPI clock to workaround it: https://github.com/espressif/esp-idf/blob/master/components/hal/spi_hal_iram.c#L76

ginkgm, 2022-11-28:
Issue closed. If any more found, feel free to re-open

closed 2022-11-28 by ginkgm

ManuelHuf, 2023-09-14:
I have the same problem using an esp32c3 with the ST CR95HF. Are there any updates regarding this issue? I tried SPI mode 0 as well as mode 3. Also the `.cs_ena_pretrans = 1` didn't help.