espressif/esp-idf #8537: real error code is lost in gpio.c gpio_isr_register()

peturdainn, 2022-03-09:
## Environment

- Module or chip used:  ESP32c3 mini
- IDF version 4.4

## Problem Description

in components/driver/gpio.c function gpio_isr_register() valuable error info is lost by not returning it to the caller:

on line 924 and following we find this

```cpp
#if CONFIG_FREERTOS_UNICORE
    gpio_isr_register_on_core_static(&p);
    ret = ESP_OK;
#else /* CONFIG_FREERTOS_UNICORE */
    ret = esp_ipc_call_blocking(gpio_context.isr_core_id, gpio_isr_register_on_core_static, (void *)&p);
#endif /* !CONFIG_FREERTOS_UNICORE */
    if(ret != ESP_OK || p.ret != ESP_OK) {
        return ESP_ERR_NOT_FOUND;
    }
```

now observe the last 3 lines: if the registering fails, we return a fixed error code, not the actual reason for failing, which could be in ret or p.ret

### Proposed handling:

```cpp
#if CONFIG_FREERTOS_UNICORE
    gpio_isr_register_on_core_static(&p);
    ret = ESP_OK;
#else /* CONFIG_FREERTOS_UNICORE */
    ret = esp_ipc_call_blocking(gpio_context.isr_core_id, gpio_isr_register_on_core_static, (void *)&p);
#endif /* !CONFIG_FREERTOS_UNICORE */
    if(ret != ESP_OK)
        return ret;
    if(p.ret != ESP_OK)
        return p.ret;
```

ginkgm, 2022-03-10:
Hi @peturdainn ,
due to back-compatibility reasons, we can't randomly change the return values.

Do you have any use cases that strongly needed to get the original return error code?

Michael

peturdainn, 2022-03-10:
What's strong? It was annoying when debugging an issue, to the point that I briefly changed the code like above to discover what's going on.
Do you think there's code that relies on getting back an inappropriate error code?

Anyway, I'm not emotionally attached to this issue, I just (as a rule) try to contribute/report to things I use or encounter ;)

peturdainn, 2022-03-10:
I understand that it currently matches the documentation too:
ESP_ERR_NOT_FOUND No free interrupt found with the specified flags 

So nevermind then

closed 2022-03-10 by peturdainn