espressif/esp-idf #8539: ESP-IDF 4.4 Eclipse Application Size Analysis Failed to create the part's controls.

Ceyuan, 2022-03-10:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP-32-DevkitM-1]
- Kit version V1.0
- Module or chip used:  [ESP32-Mini-1]
- IDF version (run ``git describe --tags`` to find it): [v3.1-41-gb082b0e]
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: 
- Using an IDE?: ECLIPSE Version: 2021-06 (4.20.0)   Build id: 20210612-2011
- Installed tools:
 Executing C:\ESP32\esp-tools\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\ESP32\ESP-IDF\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

## Problem Description
Was using ESP-IDF V4.3 and can use the Application Size Analysis but it had indexing problems and was solved when upgrading to V4.4. But with V4.4 the Application Size Analysis shows "Failed to create the part's controls"
Download Blink example, build successful, right click on project and click "ESP-IDF: Application Size Analysis"
[... 67 lines cut ...]
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3629)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1157)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:78)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:567)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:654)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:591)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1462)

### Expected Behavior
Show an analysis like in V4.3

igrr, 2022-03-10:
@kolipakakondal please advise if this issue should be moved to idf-eclipse-plugin repository.

(I suppose the underlying issue may be due to the difference in JSONs returned by idf_size.py in v4.3 and v4.4 — let us know if you think something should be fixed on IDF side.)

kolipakakondal, 2022-03-11:
Hi @ivmarkov Yes, please move to the idf-eclipse-plugin repository, let me verify this on the latest build v2.4.2. 
This was fixed in v2.4.0 earlier with https://github.com/espressif/idf-eclipse-plugin/pull/450

Hi @Ceyuan Could you mention what is the IDF Eclipse plugin version which you are using, [this](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-view-product-information) would help you to find the plugin version.  In case if you are not the latest(v2.4.2), please update to the [latest](https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url) and see if you are able to reproduce the problem.

Thanks