espressif/esp-idf #8540: undefined reference to `chdir'

Rajssss, 2022-03-10:
## Environment

- Development Kit:     none (custom)
- Module or chip used:  ESP32-WROOM-32
- IDF version : v5.0-dev
- Build System:    CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:    Linux
- Using an IDE?: Eclipse Version: 4.22.0.v20211124-1800 | Eclipse CDT Version: 10.5.0.202109270023 | IDF Eclipse Plugin Version: 2.4.1.202202161532
- Power Supply:  USB

## Problem Description

I am using fatfs with sdcard through spi, doing chdir with correct args, linker gives undefined reference to `chdir' error. It seems chdir is enabled in the vfs but not implemented.

### Expected Behavior
Compile successfully
### Actual Behavior
Error: undefined reference to `chdir' 
### Steps to reproduce

1. setup sd_spi as per available example
2. try chdir

igrr, 2022-03-10:
Hi @Rajssss, `chdir` was implemented (as a no-op stub) in https://github.com/espressif/esp-idf/commit/d83ce227aa987d648e1a0dddba32b88846374815.

Which IDF commit are you working with? You can find it by running `git describe --tags --dirty` or by running `idf.py --version`.

Rajssss, 2022-03-10:
> Hi @Rajssss, `chdir` was implemented (as a no-op stub) in [d83ce22](https://github.com/espressif/esp-idf/commit/d83ce227aa987d648e1a0dddba32b88846374815).
> 
> Which IDF commit are you working with? You can find it by running `git describe --tags --dirty` or by running `idf.py --version`.

`git describe --tags --dirty` : v5.0-dev-dirty
`idf.py --version` : ESP-IDF v5.0-dev-dirty

igrr, 2022-03-10:
I see, so it means the feature was added in a more recent version than the commit you are using (about 710 commits later). If you upgrade to the latest `master` branch the feature should become available.

> What does no-op stub means?

It means that the function is defined, but it doesn't do anything, returns an error code (-1) and sets `errno` to `ENOSYS`, indicating that the feature is not supported. At the same time, `getcwd` always returns "/", i.e. root directory.
ESP-IDF doesn't have concept of a "current working directory" so we provide these functions only for compatibility with existing libraries which may in some cases call them. If you are writing an application from scratch, you should not depend on these functions doing anything useful.

Rajssss, 2022-03-10:
> It means that the function is defined, but it doesn't do anything, returns an error code (-1) and sets `errno` to `ENOSYS`, indicating that the feature is not supported. ([Link to the code](https://github.com/espressif/esp-idf/commit/d83ce227aa987d648e1a0dddba32b88846374815#diff-91b1abbc42db113c5dcf9c11c594f0b3e75cae5f6dfba9ef8b68e26bbd422f4aR120-R125).) At the same time, `getcwd` always returns "/", i.e. root directory.
> ESP-IDF doesn't have concept of a "current working directory" so we provide these functions only for compatibility with existing libraries which may in some cases call them. If you are writing an application from scratch, you should not depend on these functions doing anything useful.

I see, thanks for the explanation.

Rajssss, 2022-03-10:
But still, this will not solve my problem, as my application requires chdir, is it possible to support it in the future? @igrr

igrr, 2022-03-10:
@Rajssss could you explain a bit what the application is doing, and why you think it's not possible to avoid using `chdir`? This would help us understand how to solve this problem.

Rajssss, 2022-03-10:
> @Rajssss could you explain a bit what the application is doing, and why you think it's not possible to avoid using `chdir`? This would help us understand how to solve this problem.

I am writing a function to get the oldest file in a dir, the issue is without doing chdir to that dir, I am always getting the same date for all the files. But I have tested the same code in my PC with chdir it works as expected.

```
void directoryManager(char *dir)
{
    DIR *dp;
    struct dirent *entry, *oldestFile;
    struct stat statbuf;
    int numberOfEntries=0;
    time_t t_oldest;
    double sec;

    time(&t_oldest);
    printf("now:%s\n", ctime(&t_oldest));
    if((dp = opendir(dir)) != NULL) {
    // chdir(dir);
    while((entry = readdir(dp)) != NULL) {
        stat(entry->d_name, &statbuf);       
        if(strcmp(".",entry->d_name) == 0 || strcmp("..",entry->d_name) == 0)
            continue;

        numberOfEntries++;

        printf("no of files: %d\n", numberOfEntries);
        printf("%s\t%s", entry->d_name, ctime(&statbuf.st_ctime));
        if(difftime(statbuf.st_ctime, t_oldest) < 0){
                t_oldest = statbuf.st_mtime;
            oldestFile = entry;
        }
    } 
    }       
        printf("no of files: %d", numberOfEntries);

    printf("\n\noldest file: %s\n", oldestFile->d_name);
    //    remove(oldestFile->d_name);

    printf("total no of files: %d\n", numberOfEntries);
    printf("\noldest time:%s", ctime(&t_oldest));
    closedir(dp);
}
```

Rajssss, 2022-03-10:
> In this case, you were passing `entry->d_name` to `stat`, which expected to see absolute path.

That was a mistake I did :'). Thanks for pointing that out for me, my issue is fixed.

closed 2022-03-10 by Rajssss