espressif/esp-idf #8541: USB MIDI example

ultrapalace, 2022-03-10:
**Is your feature request related to a problem? Please describe.**

I have been struggling to get esp32s2 to receive midi data over USB. There are Arduino libraries that can do this, but I want to use esp-idf :)

**Describe the solution you'd like**

I would love to have a midi usb device example to study.

**Describe alternatives you've considered**

I have tried to port the Arduino examples to esp-idf but it is too complicated, the API is very confusing.

**Additional context**

chegewara, 2022-03-10:
https://github.com/hathach/tinyusb/tree/master/examples/device/midi_test

ultrapalace, 2022-03-13:
@chegewara I definitely looked at this example, but it is not ported to esp32. Would you expect this to compile and run if I just copy it into my program? There are so many defines and unusual import patterns that I am unable to really see how these things work.

ultrapalace, 2022-03-13:
for example `#include "bsp/board.h"` seems doomed to fail.

ultrapalace, 2022-03-13:
@chegewara you have done a lot of work on USB on ESP32, including a midi usb driver for Arduino. Would you be willing to share a bare-bones working example of usb midi device for esp-idf? Just getting access to midi data, via usb on esp32s2, and printing it to the console (not parsing or anything) would be so very helpful, and I imagine MANY users would be overjoyed to have access to this feature.

chegewara, 2022-03-14:
All arduino libraries ported to work on esp32 S2, which are my library, espressif and adafruit, are based on that tinyusb.
Adafruit and my library is also supporting MIDI, so it is possible to do it with esp-idf too.
If you have problems with something which does not look like USB code, just like `bsp/board.h`, then remove it from code (this header is to use LEDs and buttons).
Currently i am working on full time with esp32 and rest of my time i am just resting (i am lazy). In other hand there are few skilled, having knowledge about USB and helpful people at espressif, so i believe they may help you.

marchingband, 2022-03-14:
@chegewara  Yes I noticed all the Arduino USB libraries work from tinyUSB, I have looked at them all, but I find it too complicated, and under documented, to replicate. I have a feeling that there are some functions and macros that make the existing esp-idf HID examples deceptively simple looking, but those same helpers do not exist for MIDI.
It's by far the most complex and confusing protocol I've attempted to work with. A working sdk example file is so vital in moments like this, and I agree with your feeling that surely someone at Espressif would be able to make such an example quite easily. Is there anyone that comes to mind that you would be willing to tag in this issue?

ultrapalace, 2022-03-14:
@chegewara Yes I noticed all the Arduino USB libraries work from tinyUSB, I have looked at them all, but I find it too complicated, and under documented, to replicate. I have a feeling that there are some functions and macros that make the existing esp-idf HID examples deceptively simple looking, but those same helpers do not exist for MIDI.
It's by far the most complex and confusing protocol I've attempted to work with. A working sdk example file is so vital in moments like this, and I agree with your feeling that surely someone at Espressif would be able to make such an example quite easily. Is there anyone that comes to mind that you would be willing to tag in this issue?

spacenate, 2022-03-15:
@ultrapalace take a look here: https://github.com/touchgadget/esp32-usb-host-demos/blob/main/examples/usbhmidi/usbhmidi.ino

I tried to run this on ESP32-S2, and had some trouble: the `midi_transfer_cb()` was not being called (though I was not running that example code directly, but using it as a reference for my own implementation. Perhaps the issue is with my own code, which I will get back to debugging soon). Good luck.

ultrapalace, 2022-03-15:
thanks @spacenate but I am looking for USB device, not host, and looking for esp-idf not Arduino :)

adokitkat, 2022-04-12:
> thanks @spacenate but I am looking for USB device, not host, and looking for esp-idf not Arduino :)

I will take a look at it. For now here's a `git patch` file containing a fixes to ESP-IDF's TinyUSB wrapper and a ported MIDI example.
[TinyUSB-MIDI.zip](https://github.com/espressif/esp-idf/files/8476019/TinyUSB-MIDI.zip)

You can apply the patch to IDF's `master` branch and try it out. The example will appear in `examples/peripherals/usb/device/tusb_midi`.

I've tested it on ESP32-S2.

ultrapalace, 2022-04-16:
> I've tested it on ESP32-S2.

I have tested it as well and the examples works :)
Thank you so so so much @adokitkat ! I appreciate you!

Is it likely this will make its way into master at some point?
Is there any way I can contribute? Testing / example code / docs / etc. Just let me know <3

ultrapalace, 2022-04-16:
This works great as well, which is so exciting 
```
#define MIDI_RX_BUFFER_SIZE 128
uint8_t midi_in[MIDI_RX_BUFFER_SIZE];

void tud_midi_rx_cb(uint8_t itf)
{
    ESP_LOGI(TAG, "new midi message");
    uint32_t bytes_read = tud_midi_n_stream_read(itf, 0, midi_in, MIDI_RX_BUFFER_SIZE);
    for(int i=0;  i<bytes_read; i++)
    {
        ESP_LOGI(TAG, "byte %d = %d", i, midi_in[i]);
    }
}
```

adokitkat, 2022-04-17:
You're welcome :)

> Is it likely this will make its way into master at some point?

It is likely. Hopefully soon enough. 

> Is there any way I can contribute? Testing / example code / docs / etc. Just let me know <3

It's up to you ;) IDF is open-source.

I personally don't have an extensive knowledge on MIDI or MIDI USB, I have only patched our wrapper and ported the MIDI example to IDF.

If you want to create e.g. a more advanced example code with documentation you are free to do so. 
You can always create a pull request. I would advise to look at another PRs about examples first.

marchingband, 2022-05-22:
@adokitkat this is amazing thank you! ... now ... is there ANY chance you would be willing to do the same, but for USB MIDI HOST?
You would be my absolute hero if so <3

adokitkat, 2022-05-22:
> @adokitkat this is amazing thank you! ... now ... is there ANY chance you would be willing to do the same, but for USB MIDI HOST?
> You would be my absolute hero if so <3

Hello. No problem :) 

Unfortunately I don't think I am able to do USB MIDI host since it seems like it is not a part of TinyUSB library which ESP-IDF is using.

marchingband, 2022-05-22:
oh shoot :( thanks for the reply!

marchingband, 2022-05-22:
it looks like it may be coming soon though!
https://github.com/hathach/tinyusb/pull/1219

bunalti, 2022-06-24:
> I will take a look at it. For now here's a `git patch` file containing a fixes to ESP-IDF's TinyUSB wrapper and a ported MIDI example. [...]

I am testing IDF releases. And come to realize it does not apply v4.2 and earlier ( no such directory). I am chronemically trying to apply the attached patch to the releases but I get similar/same error patterns on every other release starting from v4.3. 

Error is:

`error: patch failed: components/tinyusb/Kconfig:112
error: components/tinyusb/Kconfig: patch does not apply
error: patch failed: components/tinyusb/additions/include/tusb_config.h:85
error: components/tinyusb/additions/include/tusb_config.h: patch does not apply
error: patch failed: components/tinyusb/additions/src/usb_descriptors.c:125
error: components/tinyusb/additions/src/usb_descriptors.c: patch does not apply`

Which release is this patch indented for ?

adokitkat, 2022-06-24:
Hi @bunalti 

The quick patch I posted was intended for the master branch at the time of posting (as far as I remember). 

The real patch is waiting to be merged to master. Sorry fot the long wait but the patch is coming eventually.

bunalti, 2022-06-24:
> Hi @bunalti
> 
> The quick patch I posted was intended for the master branch at the time of posting (as far as I remember).
> 
> The real patch is waiting to be merged to master. Sorry fot the long wait but the patch is coming eventually.

By taking reference of the patch in the attachment I made a quick fix for the issue on ESP-IDF v4.4.1 release.I have tested the device with MIDI-OX software and it seems to be working correctly. 

[tinyusb_midi_device.zip](https://github.com/espressif/esp-idf/files/8976762/tinyusb_midi_device.zip)

peergum, 2022-06-27:
Unfortunately, patch was not integrated to release 5.0 😢 I got patch applied successfully on commit 01d014c42d
```
* d2fc90767d - (HEAD -> tusb-midi-fix) usb: Added MIDI example + fixed TinyUSB MIDI config (3 weeks ago) <Adam Múdry>
*   01d014c42d - (master) Merge branch 'contrib/github_pr_9062' into 'master' (4 weeks ago) <Mahavir Jain>
|\  
| * dd7aece4bf - Secure boot v2 image format: CRC size (4 weeks ago) <Stefan Wallentowitz>
```
but now that 5.0 was released I'll probably have to reapply... not funny...
Maybe I'll rebase the fix on 5.0 if it's not too much work... that patch is pretty useful indeed.

adokitkat, 2022-07-02:
@peergum @marchingband @bunalti @ultrapalace 

The patch was finally merged on internal servers. However I don't know when the commit propagates to Github's master branch.

So I created a patch file of what was merged against `release/v5.0` branch for you:

[midi_final.zip](https://github.com/espressif/esp-idf/files/9033102/midi_final.zip)

I'll try to notify you when the master branch will be updated on GH.

adokitkat, 2022-07-07:
@peergum @marchingband @bunalti @ultrapalace

USB MIDI device driver with an example code is now available in master branch :) Hope it works for you.

marchingband, 2022-07-07:
This really is a significant contribution to the esp-audio community, I can't thank you enough 🤍 happy to see this close!
Would it make sense for me to open a new issue to request midi USB Host? As development unfolds in the idf, I'd just like it to be absolutely clear that we are in need of host as well, and getting it working would be MASSIVE. I understand it may be a ways off, but id still like to keep it in the roadmap if at all possible. Thanks again!

adokitkat, 2022-07-08:
@marchingband Unfortunately since we depend on 3rd party TinyUSB software and they hasn't implemented MIDI host driver yet, there is only so much I can do. Once they implement it, then we can adopt it and make it work with ESP-IDF. You can open a issue but until the mentioned thing isn't resolved there won't be much of progress I am afraid. 

I guess the best thing you could do right now would be to ask nicely the developers and contributors of TinyUSB to get it working and merge it soon.

chegewara, 2022-07-08:
> @marchingband Unfortunately since we depend on 3rd party TinyUSB software and they hasn't implemented MIDI host driver yet, there is only so much I can do. Once they implement it, then we can adopt [...]

Since when USB host driver is dependent on tinyUSB?

adokitkat, 2022-07-08:
> Since when USB host driver is dependent on tinyUSB?

You are right, it is not. I am sorry for this misinformation. I got caught up in this one since I was dealing with my MR about this for a quite long time and the info from TinyUSB's pull request about MIDI host so I assumed we are using it as well. Sorry about this.

Well I guess then is alright to open an issue about MIDI host. @chegewara what do you think?