espressif/esp-idf #8542: ESP32-S - NRF52832 BLE notify and timeout problem

oguzhantokuc, 2022-03-10:
## Problem Description

I am coding in ESP32-S for BLE client. NRF52832 is coded as a BLE server which have 128bit UUID. I am using a Gatt_client_demo for 128bit. Link is given below. When i connect nrf, esp32-s can read and write data successfuly. When i try to notify it can't read any data from nrf. Also, when i connect to nrf for read/write/notify, a massage appear about timeout and after that it disconnects automatically. I couldn't solve these problems. If anyone help, i would be appriciate. By the way, when i try to conenct nrf with nrfconnect app on mobile phone, it works fine. Timeout probem can be seen in the Actual Behavior picture.

DEMO: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client 

### Expected Behavior

[image]

### Actual Behavior
// The red lines are not erros. I added them in the code. 
[image]

### Steps to reproduce

1. I download the demo code.
2. I tried it in UUD16 (works fine / timeout problem happened)
3. I changed it for UUD128
4. I tried it in Read/Write (works fine)
5. I tried it in notify. It didn't work. Timeout problem happened and i couldn't read data.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
//it it C not C++
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
https://gist.github.com/oguzhantokuc/302d0bf742f02f3e0cc21500a5e5dd58

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

esp-cjh, 2024-02-27:
Hi, @oguzhantokuc Could you please provide the service database or the source code of NRF52832 server? I need them to reproduce the problem.

Indastri, 2024-11-12:
Since there was no response from the user I am closing this issue. If you will encounter further problems please reopen this or create a new issue.

Cheers!

closed 2024-11-12 by Indastri