espressif/esp-idf #8545: WSS Server keep alive issue

EN20M, 2022-03-10:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version DevKitC: [v4]
- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.4
- Build System:  [idf.py]
- Compiler version: 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB and external 5V]

## Problem Description

I have frankensteined several of the example codes together to get my esp to read a html file from a sd card, be a wifi acces point, start a webserver that provieds the website from the sd card. the website has a button to open a websocket. when using that websocket to constantly send printouts to the website to be displayed everything is happy. when for a certain time nothing is to be send and the keep alive routine kicks in and sends a PING towards the client and the clients sends back the PONG everything starts to fall apart. On the clientside I was able to see via WireShark that PING and PONG where received and send as intended. On the ESP(serverside) I see from the log that keep_alives kicks in after 5 seconds as intended one PING gets send out (4 more try to) but before the received PONG gets handeled the keep alive times out and closes the connection. Only after that the the received PONG seems to be handled with errors that I'm not able to understand. Once the client gets disconnected the remaning 4 PINGs get send out but obviously don't reach the client anymore.

### Expected Behavior
I expect a PING to result in a PONG making its way into the aplication so it resetst the timeout for that connection.

### Actual Behavior
client sees PING and sends PONG but it does not reach the aplication layer on the server

## Debug Logs

```
I (101116) wss_echo_server: New client connected 59
I (101124) WebSocket: Handshake done, the new connection was opened
I (106117) wss_echo_server: Checking if client (fd=59) is alive
I (106119) Sending WS: PING
I (107119) wss_echo_server: Checking if client (fd=59) is alive
I (108121) wss_echo_server: Checking if client (fd=59) is alive
I (109122) wss_echo_server: Checking if client (fd=59) is alive
I (110123) wss_echo_server: Checking if client (fd=59) is alive
E (111124) wss_keep_alive: Client (fd=59) not alive!
E (111124) wss_echo_server: Client not alive, closing fd 59
W (111128) httpd_txrx: httpd_sock_err: error in recv : 11
W (111131) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
E (111139) WebSocket: httpd_ws_recv_frame failed with -1
I (111145) wss_echo_server: Client disconnected 59
I (111152) Sending WS: PING
I (111155) Sending WS: PING
I (111158) Sending WS: PING
I (111161) Sending WS: PING
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8223781/sdkconfig.txt)
[image]
[image]

EN20M, 2022-03-11:
Now that I managed to get Debug Log level running without constantly stackoverflowing. I have this log with a little more insight but I still cant wrap my head around where any wy everything is going south at some point

```
I (21514) WebSocket: Handshake done, the new connection was opened
D (21518) httpd_sess: httpd_sess_process: httpd_req_delete
I (21519) upm_send_bytes: start frame length bytes are: 8, 8 from 5
D (21525) httpd_sess: httpd_sess_process: success
D (21536) httpd: httpd_server: doing select maxfd+1 = 58
D (26024) wss_keep_alive: Haven't seen the client (fd=57) for a while
I (26024) wss_echo_server: Checking if client (fd=57) is alive
D (26028) httpd: httpd_server: processing ctrl message
D (26032) httpd: httpd_process_ctrl_msg: work
I (26036) Sending WS: PING
D (26041) httpd: httpd_server: doing select maxfd+1 = 58
I (26058) httpd: httpd_process_session: processing socket 57
D (26059) httpd_sess: httpd_sess_process: httpd_req_new
D (26059) httpd_parse: httpd_req_new: New request, has WS? Yes, sd->ws_handler valid? Yes, sd->ws_close? No
D (26069) httpd_txrx: httpd_recv_with_opt: requested length = 1
D (26076) httpd_txrx: httpd_recv_with_opt: received length = 1
D (26081) httpd_ws: httpd_ws_get_frame_type: First byte received: 0x8A
D (26088) httpd_parse: httpd_req_new: New WS request from existing socket, ws_type=10
D (26096) httpd_parse: httpd_req_new: Received PONG frame
D (26101) httpd_txrx: httpd_recv_with_opt: requested length = 1
D (26107) httpd_txrx: httpd_recv_with_opt: received length = 1
D (26113) httpd_txrx: httpd_recv_with_opt: requested length = 4
D (26119) httpd_txrx: httpd_recv_with_opt: received length = 4
D (26125) httpd_txrx: httpd_recv_with_opt: requested length = 1
D (27028) wss_keep_alive: Haven't seen the client (fd=57) for a while
I (27028) wss_echo_server: Checking if client (fd=57) is alive
D (28031) wss_keep_alive: Haven't seen the client (fd=57) for a while
I (28031) wss_echo_server: Checking if client (fd=57) is alive
D (29034) wss_keep_alive: Haven't seen the client (fd=57) for a while
I (29034) wss_echo_server: Checking if client (fd=57) is alive
D (30037) wss_keep_alive: Haven't seen the client (fd=57) for a while
I (30037) wss_echo_server: Checking if client (fd=57) is alive
D (31039) wss_keep_alive: Haven't seen the client (fd=57) for a while
E (31039) wss_keep_alive: Client (fd=57) not alive!
W (31131) httpd_txrx: httpd_sock_err: error in recv : 11
D (31131) httpd_txrx: httpd_recv_with_opt: error in recv_fn
W (31132) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
E (31139) WebSocket: httpd_ws_recv_frame failed with -1
I (31145) httpd_sess: httpd_sess_process: failed httpd_req_new
D (31152) httpd_sess: httpd_sess_delete: fd = 57
I (31156) wss_echo_server: Client disconnected 57
D (31162) httpd_sess: httpd_sess_delete: active sockets: 0
D (31167) httpd: httpd_server: doing select maxfd+1 = 56
D (31174) httpd: httpd_server: processing ctrl message
D (31178) httpd: httpd_process_ctrl_msg: work
I (31182) Sending WS: PING
D (31186) httpd: httpd_server: doing select maxfd+1 = 56
D (31192) httpd: httpd_server: processing ctrl message
D (31196) httpd: httpd_process_ctrl_msg: work
I (31201) Sending WS: PING
D (31204) httpd: httpd_server: doing select maxfd+1 = 56
D (31210) httpd: httpd_server: processing ctrl message
D (31215) httpd: httpd_process_ctrl_msg: work
I (31219) Sending WS: PING
D (31222) httpd: httpd_server: doing select maxfd+1 = 56
D (31228) httpd: httpd_server: processing ctrl message
D (31233) httpd: httpd_process_ctrl_msg: work
I (31237) Sending WS: PING
D (31241) httpd: httpd_server: doing select maxfd+1 = 56
```

closed 2022-03-21 by EN20M

claybailey2, 2024-01-24:
Did you ever find a fix for this?

EN20M, 2024-01-24:
I honestly can't remember and unfortunately can not test this because I don't work for the same company anymore. But the fact that I closed the issue without writing anything makes me believe it solved itself by an idf upgrade. You could double check if around the same time the issue was closed a new idf was released.