espressif/esp-idf #8547: ESP32-S3 Blink example does not work

eos1d3, 2022-03-10:
## Environment

- Development Kit:      [ESP32-S3-DevKitC-1 v1.0, N32R8V]
- Module or chip used:  [ESP32--S3-WROOM-2]
- IDF version: v4.4-351-g121ddb87e5
- Build System:         [idf.py]
- Compiler version:xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description
//Detailed problem description goes here.
The on board RGB LED almost does not response.

### Expected Behavior
The on board RGB LED should be on and off for every 1 second.

### Actual Behavior
The RGB LED will change to white color when power up for the first time. Then it mostly does not change even I see a lot of on/off messages from terminal. Wait a while (around one minute or more), it will change to other random color. And this patten will repeat.

This problem just occurs to ESP32-S3. ESP32-S2 does not have problem.

### Steps to reproduce
1. Create an IDF project of Blink example using ESP32-S3 
2. Build and flash it. 
3. The problem occurs immediately 

## Debug Logs

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x1658
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f74
entry 0x403b6248
I (32) boot: ESP-IDF v4.4-351-g121ddb87e5 2nd stage bootloader
I (32) boot: compile time 04:32:14
I (32) boot: chip revision: 0
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : SLOW READ
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08958h ( 35160) map
I (109) esp_image: segment 1: paddr=00018980 vaddr=3fc91a10 size=02718h ( 10008) load
I (112) esp_image: segment 2: paddr=0001b0a0 vaddr=40374000 size=04f78h ( 20344) load
I (124) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=189c0h (100800) map
I (150) esp_image: segment 4: paddr=000389e8 vaddr=40378f78 size=08a8ch ( 35468) load
I (160) esp_image: segment 5: paddr=0004147c vaddr=50000000 size=00010h (    16) load
I (166) boot: Loaded app from partition at offset 0x10000
I (166) boot: Disabling RNG early entropy source...
I (180) cpu_start: Pro cpu up.
I (181) cpu_start: Starting app cpu, entry point is 0x403751b8
0x403751b8: call_start_cpu1 at /Users/andy/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (195) cpu_start: Pro cpu start user code
I (195) cpu_start: cpu freq: 160000000
I (195) cpu_start: Application information:
I (198) cpu_start: Project name:     blink
I (202) cpu_start: App version:      1
I (207) cpu_start: Compile time:     Mar 11 2022 04:31:56
I (213) cpu_start: ELF file SHA256:  0665a6b0e5f5d903...
I (219) cpu_start: ESP-IDF:          v4.4-351-g121ddb87e5
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (232) heap_init: At 3FC94B50 len 0004B4B0 (301 KiB): D/IRAM
I (238) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (245) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (251) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (258) spi_flash: detected chip: mxic
W (262) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (273) spi_flash: flash io: dio
W (277) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (291) sleep: Configure to isolate all GPIO pins in sleep state
I (297) sleep: Enable automatic switching of GPIO sleep configuration
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (325) example: Example configured to blink addressable LED!
I (325) example: Turning the LED OFF!
I (1335) example: Turning the LED ON!
```

pedrominatel, 2022-03-11:
Hi @eos1d3,

Have you tried to flash a different ESP32-S3-DevKitC board? Does the same behavior happen?
I've tried to reproduce but I can't see the reported behavior.

pedrominatel, 2022-03-11:
Can you also test it using the [LED Strip Example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip)?

Be sure to configure the GPIO to 48 in case of the [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html).

eos1d3, 2022-03-11:
@pedrominatel I have only one ESP32-S3-DevKitC (N32R8V) which was ordered in Feb from Espressif China. But the factory pre-loaded firmware works perfectly to control the RGB LED with different color. And the LED works exactly like ESP32-S2 boards pre-loaded firmware.

I will try. to test LED Strip example.

GarMingLi, 2022-03-11:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在工作中，无法亲自回复您的邮件。我将在工作结束后，尽快给您回复。

eos1d3, 2022-03-11:
@pedrominatel Hi, I found something now.

Using LED Strip and Blink examples, the onboard LED does not work normally.

Using LED Strip and Blink examples with external WS2812B. I connect GPIO18 directly to the WS2812B led strip without resistor. Both work.

Using LED Strip and Blink examples with external WS2812B. I connect GPIO48 directly to the WS2812B led strip without resistor. The WS2812B LED script never turn on. But on board always work!

But the on board LED with pre-loaded firmware work without any error! So what is wrong?

eos1d3, 2022-03-11:
With both external WB2812B and on-board LED are connected to GPIO48. As I said before, both work.

Then I remove external one from GPIO48. The on-board LED is not normal immediately.

pedrominatel, 2022-03-11:
Hi,

Just to understand your test scenario.

Once you connect an external LED strip to the GPIO48 pin, the onboard LED start working? Is that correct?

eos1d3, 2022-03-11:
Right! The on-board LED never fails when 5V, control and GND from external LED strip are connected to the development board (5V, GPIO48 and GND)

eos1d3, 2022-03-11:
This is the video of Blink example without connecting external LED strip. 

https://www.youtube.com/watch?v=517dOgzqTms

pedrominatel, 2022-03-11:
I think we can say there is no issue with the example itself since it works with the ESP32-S2 and when the external LED strip is connected.
This seems to be some weird HW issue or a "defective" LED. It's hard to say without checking the HW.

Can you try to connect a 100nF ceramic capacitor to the 3V3 and GND close to the embedded LED?

pedrominatel, 2022-03-11:
If you have an oscilloscope, you can also connect it to the GPIO48 and compare it in both situations.

eos1d3, 2022-03-11:
It may not be caused by hardware. When I used first release of IDF 4.4, I remember it does not have problem until I applied current patches to it.

I can check the waveform later. But I want to reinstall original IDF 4.4 without patch. Please let me know how to do with git. Thanks!

pedrominatel, 2022-03-11:
You can use this guide: [Get ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-2-get-esp-idf)

eos1d3, 2022-03-11:
The guide will install IDF with new patches, not the original IDF 4.4 without patch.

I am seeing random square waveforms from the LED control signal. From the example code, I believe each update the waveform is identical. Is this right?

And I am going to compare this with ESP32-S2 soon..

eos1d3, 2022-03-11:
Finally got the reason why it failed. The board (ESP32-S3-DevKitC-1 v1.0, N32R8V) from Espressif may have problem.

The waveform from ESP32-S3, measured from the LED control signal pin solder pad (not from the pin header). PWM High is too low at around 1.88V
https://ibb.co/7Nry8F2

For ESP32-S2, it is around 3.1V with the same testing condition above.
https://ibb.co/ZSD2TTL

The two waveforms overlap image:
https://ibb.co/5Ywxf87

Do recent boards have similar problem? The board is also not recognized by RainMaker during claiming procedure.

eos1d3, 2022-03-14:
@pedrominatel Hi, I got some new findings today.

With ESP32-S2, it does not fully work with IDF master branch even with external LED strip connected. It will display random color with external LED strip while IDF 4.4 does not have this issue.

And surprisingly, I got ESP32-C3 today. The Blink example DOES NOT work as ESP32-S2 with IDF 4.4. And the strangest thing is the same ESP32-C3 works perfectly with IDF 4.4 + Rainmaker. I can control the LED on, off, and brightness and color change without any error.

So I think the control pin voltage is still within the limit. And the problem exists in Blink example. I do not know if Blink and led_light (Rainmaker) are using the same LED library.

phucng, 2022-03-16:
@eos1d3
1. There are a couple of warnings in your output:
```
W (262) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (273) spi_flash: flash io: dio
W (277) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (291) sleep: Configure to isolate all GPIO pins in sleep state
I (297) sleep: Enable automatic switching of GPIO sleep configuration
```
I once got nonworking examples while playing around with different settings for the flash and psram on an ESP32-S3-DevKitC-1-N8R8. Now they work. I'm not sure if it is the case in your situation, but I've attached my config file for reference.

2. Regarding your concern with the patches, I'm using v5.0-dev, which was cloned from the master branch a month ago. It may contain a different set of patches (compared to your v4.4) that fix some bugs and introduce new ones.

[sdkconfig.json.txt](https://github.com/espressif/esp-idf/files/8265189/sdkconfig.json.txt) (github doesn't allow .json file so I appended .txt to it)

eos1d3, 2022-03-17:
Just reinstall IDF master, and also replaced sdkconfig.json, the problem is still there. But there is new error in the log.
```
W (182) flash HPM: Enter HPM by reconfiguring dummy has not been fully tested
E (183) flash HPM: Flash high performance mode hasn't been supported

```
Is my board defective? As the onboard LED control signal (PWM) is just 1.8V while ESP32S2 is 3.1V. And this board does not work with Rainmaker too. https://github.com/espressif/esp-rainmaker/issues/112

The full log is here:

```
Executing action: monitor
Serial port /dev/cu.usbserial-14130
Connecting....
Detecting chip type... ESP32-S3
Running idf_monitor in directory /Users/andy/esp/esp-idf/examples/get-started/blink
Executing "/Users/andy/.espressif/python_env/idf5.0_py3.9_env/bin/python /Users/andy/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-14130 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /Users/andy/esp/esp-idf/examples/get-started/blink/build/blink.elf -m '/Users/andy/.espressif/python_env/idf5.0_py3.9_env/bin/python' '/Users/andy/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbserial-14130 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0110,len:0x18d0
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x319c
entry 0x403b6240
I (33) boot: ESP-IDF v5.0-dev-2046-g5963de1caf 2nd stage bootloader
I (33) boot: compile time 05:54:59
I (33) boot: chip revision: 0
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : SLOW READ
I (47) boot.esp32s3: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08a78h ( 35448) map
I (111) esp_image: segment 1: paddr=00018aa0 vaddr=3fc91c50 size=02b78h ( 11128) load
I (114) esp_image: segment 2: paddr=0001b620 vaddr=40374000 size=049f8h ( 18936) load
I (125) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18b54h (101204) map
I (152) esp_image: segment 4: paddr=00038b7c vaddr=403789f8 size=09254h ( 37460) load
I (162) esp_image: segment 5: paddr=00041dd8 vaddr=50000000 size=00010h (    16) load
I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
W (182) flash HPM: Enter HPM by reconfiguring dummy has not been fully tested
E (183) flash HPM: Flash high performance mode hasn't been supported
I (186) cpu_start: Pro cpu up.
I (190) cpu_start: Starting app cpu, entry point is 0x403751f4
0x403751f4: call_start_cpu1 at /Users/andy/esp/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (212) cpu_start: Pro cpu start user code
I (212) cpu_start: cpu freq: 160000000 Hz
I (212) cpu_start: Application information:
I (215) cpu_start: Project name:     blink
I (219) cpu_start: App version:      v5.0-dev-2046-g5963de1caf
I (226) cpu_start: Compile time:     Mar 18 2022 05:54:51
I (232) cpu_start: ELF file SHA256:  1dc37cb8e073e7ed...
I (238) cpu_start: ESP-IDF:          v5.0-dev-2046-g5963de1caf
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (252) heap_init: At 3FC95200 len 0004AE00 (299 KiB): D/IRAM
I (258) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (265) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (271) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (278) spi_flash: detected chip: mxic
W (282) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (293) spi_flash: flash io: dio
W (297) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (311) sleep: Configure to isolate all GPIO pins in sleep state
I (317) sleep: Enable automatic switching of GPIO sleep configuration
I (324) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (344) example: Example configured to blink addressable LED!
I (344) example: Turning the LED OFF!
I (1354) example: Turning the LED ON!

```

phucng, 2022-03-18:
@eos1d3
Accoring to https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf, it seems your SoC (ESP32-S3 N32R8V, with a V) uses 1.8V SPI flash, compared to 3.3V SPI in my N8R8 SoC. The settings for flash are unlikely interchangeable, so you may need to run
```
idf.py set-target esp32s3
idf.py menuconfig
```
yourself, instead of replacing the sdkconfig.json file. Besides the parameters for your specific examples (e.g., LED pin, number of LEDs), the remaining settings usually work fine without adjustment.

eos1d3, 2022-03-18:
@phucng With `menuconfig` or `sdkconfig.json`, none of these work.

So far, the voltage of on-board RGB LED control signal is just incorrect. Why is it just 1.8V? And all external WB2818B works, so I think there is nothing wrong with the library. it seems the board has placed incorrect resistor(s) for the LED control signal. The voltage is too low to control the LED.

It is not coincident with 1.8V SPI as both are 1.8V now. Is something wrong with the BOM or schematic of the board?

Would you please check with your design team for this? From what I tested, the only possibility is wrong values of resistors are used.

BTW, Rainmaker does not work with these board using: 

https://github.com/espressif/esp-idf/tree/master/examples/storage/nvs_rw_blob

https://github.com/espressif/esp-idf/tree/master/examples/storage/nvs_rw_value

https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr

pedrominatel, 2022-04-12:
Can you check the value of the R17?
[image]
Is the one close to the RGB LED:
[image]

It should be 0R. You can try to remove it and make a soldering jumper for testing.

zjanosy, 2022-04-12:
It reads 0 ohm "in circuit". I don't want to desolder it again, but I did desolder it before in order to check the voltage without load, and it was the same ~2V.

eos1d3, 2022-04-12:
Same here. Zero ohm on board.

eos1d3, 2022-04-12:
> I have noticed that while on the S3 board the LED is connected to GPIO48,
> on the similar S2 board it is connected to GPIO18. This may explain why the
> example works on the S2.
>
> According to the S3 datasheet GPIO48 is shared with SPICLK_N_DIFF and
> SUBSPICLK_N_DIFF, whatever they are (there is no mentioning of these
> anywhere in the available documentation). Could it be that this pin is
> powered from the VDD_SPI domain? But then, 2V is a bit too high.
>
> Another possibility is that for some reason the pull-down is active on
> this pin. Unfortunately I am just learning the ESP32 and ESP-IDF, so I
> wasn't able to check where this is set, or what the actual value is. Maybe
> it could be read from the corresponding IO_MUX register.

I have another problem with the board. All flash examples doesn't work too.
And this causes Rainmaker failed to provisioning.

pedrominatel, 2022-04-12:
@zjanosy, once you select the right target (`esp32s3`) the LED GPIO is automatically selected according to the [Kconfig.projbuild](https://github.com/espressif/esp-idf/blob/master/examples/get-started/blink/main/Kconfig.projbuild#L28-L37) file.

Can you share your `sdkconfig` file, @eos1d3?

zjanosy, 2022-04-12:
@pedrominatel Yes, the configuration is correct, GPIO48 is selected. And as I wrote, the signal on the pin is correct, too, just the level is too low.

This is my sdkconfig file (I'm using VSCode on Windows 7 x64):

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32s3-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
[... 1283 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_USB_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

eos1d3, 2022-04-12:
@pedrominatel It is not configuration issue. I can confirm pin assignment is correct. See my two waveforms above. The waveform from S3, measured directly on the LED solder pad, is perfectly matched with S2 but only at 1.8V. 

And I also tried another pin on S3 with external ws2812b strip, it works perfectly. So I am sure it is the voltage issue of control pin of the on board LED.

zjanosy, 2022-04-12:
@pedrominatel Yes, the configuration is correct, GPIO48 is selected. The signal on the pin is correct, just the level is too low. Please find attached the sdkconfig file.

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/8472726/sdkconfig.zip)

pedrominatel, 2022-04-12:
Thank you for the clarification @zjanosy and @eos1d3.
I'll check with our HW team if there is anything you can try to solve this issue.

Any news I'll let you know.

netlook01, 2022-04-16:
The issues is related with 1.8v VDD_SPI, the ESP32-S3-WROOM-2 is 1.8v SPI Flash and PSRAM, the GPIO 48 & 47 is powered from VDD_SPI and it can't be changed to other power domain, so the voltage of GPIO48 & 47 is limited on 1.8v .

zjanosy, 2022-04-16:
@netlook01 Thanks for the info! Is this stated somewhere in the documentation?

netlook01, 2022-04-16:
@zjanosy It is in "Table 2: Pin Description " of ESP32-S3 datasheet ，GPIO26~ 32 , GPIO48 & 47 is powered from the VDD_SPI domain，GPIO 33～ 37 is powered from VDD_SPI or VDD3P3_CPU domain.
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf

zjanosy, 2022-04-16:
@netlook01 Thanks, that explains the problem. I have overlooked that.

supcik, 2022-04-28:
Thank you for the information. I am having the same issue. Is there something that I can do to still use the onboard RGB LED, or is it useless?

zjanosy, 2022-04-28:
As I understand this is a hardware design issue. You could either remove 0 Ohm resistor R17 and connect the LED to another pin with a wire (and change the LED pin number in the software configuration), or you could add a transistor to switch the LED (this is more challenging given the small space, but you do not loose one GPIO -- if that matters to you).

supcik, 2022-04-28:
Thank you @zjanosy.

bhoffman17, 2022-05-26:
I also have the same problem turning on the RGB LED. Here's my two cents. I don't think it's a voltage problem as the LED was on and extremely bright and changing colors when first booted (before the first program is downloaded). I suspect it has something to do with how the IO MUX works. There's  an entire chapter on the subject in the reference manual:
https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf

This caught my attention:

[in section 5.6.2 on page 196 in the PDF version]
To bypass GPIO matrix for peripheral output signals, IO_MUX_MCU_SEL for the GPIO pin must be set to the required pin function.

[in table 5.3 on page 211]
Pin function 0 (default???) for GPIO_NUM_48 is SPICLK_DIFF, it appears that setting pin function 1 is required to get the pin into mode GPIO48.

Today us my first day using this dev environment, so I haven't figured this out yet. When I do, I will post here. If someone else gets to the bottom on this first, I hope they will also post the solution here, too.

zjanosy, 2022-05-26:
@bhoffman17 
Please note that while the LED power supply voltage is 3.3V (so the LED is bright), but the SPI communication line is only 1.8V. According to the datasheet this pin belongs to the VDD_SPI power domain, which is 1.8V on this particular board (ESP32-S3-DevKitC-1-N32R8V). Other boards with 3.3V SPI flash voltage are not affected.

jbaribeault, 2022-06-23:
I'm super new to all of this and was having, what I think, is the same issue (but using Circuitpython) ...but:
Silly question, but are you treating it like an addressable RGB LED? Mine works fine if I use the neopixel library to use it instead of just setting the pin high.

zjanosy, 2022-06-24:
@jbaribeault The blink example uses the NeoPixel driver. I'm pretty sure that the problem is really in the hardware, i.e., GPIO48 is connected to the 1.8V VDD_SPI power domain, and this is not enough for the LED for reliable communication. But again, this only affects the N32R8V board, which has the 1.8V flash chip.

PilnyTomas, 2022-06-24:
The simplest bypass is to connect pin 48 to any other GPIO in the 3.3V power domain and set the NeoPixel driver for the new pin. Accoring to the [documentation](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) pins 0-21 and 38-46 should be ok. I'm not sure about pins 33-37, but don't use 27-32 nor 47 and 48.

maned3v, 2022-07-15:
Hi everyone

I have the same problem with my ESP32-S3-DevKitC-1-N8R8 when using VsCode PlatformIO (Core 6.1.0·Home 3.4.2), but it works perfectly fine with ArduinoIDE (1.8.19).

Using the ArdionoIDE ESP32 > GPIO > Blink RGB example, it works (all colors are shown):
```
void setup() {
}

// the loop function runs over and over again forever
void loop() {
#ifdef BOARD_HAS_NEOPIXEL
  digitalWrite(LED_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);

  neopixelWrite(LED_BUILTIN,LED_BRIGHTNESS,0,0); // Red
  delay(1000);
  neopixelWrite(LED_BUILTIN,0,LED_BRIGHTNESS,0); // Green
  delay(1000);
  neopixelWrite(LED_BUILTIN,0,0,LED_BRIGHTNESS); // Blue
  delay(1000);
  neopixelWrite(LED_BUILTIN,0,0,0); // Off / black
  delay(1000);
#endif
}
```

while in Visual Studio Code PlatformIO

```
#include <Arduino.h>

#define LED_BUILTIN 48

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);  //Tried with and without it
}

void loop() {
  Serial.begin(115200);

  Serial.println("HIGH");
  digitalWrite(LED_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  Serial.println("LOW");
  digitalWrite(LED_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);
}
```
it's dead.

zjanosy, 2022-07-15:
@maned3v The LED on this board is not a simple LED, but it is a NeoPixel LED, which needs to be programmed via a serial interface. I guess the Arduino init code (which is not called from your VSC code) does the programming, that's why it works. You need to initialize the LED with the appropriate function, and use neopixelWrite instead of digitalWrite.

maned3v, 2022-07-15:
> @maned3v The LED on this board is not a simple LED, but it is a NeoPixel LED, which needs to be programmed via a serial interface. I guess the Arduino init code (which is not called from your VSC [...]

Ah ok, thanks. I never used the built in led in the ESP so after reading this i got curious and tried it and saw it working with Arduino IDE but not in VsCode, i thought it might be a sw problem.

zjanosy, 2022-07-15:
@maned3v Actually only the ESP32-S3-DevKitC-1-**N32R8V** board is affected, which has a 1.8V SPI flash on board, so your board should work. Try running the blink example from VSC.

maned3v, 2022-07-15:
> @maned3v Actually only the ESP32-S3-DevKitC-1-**N32R8V** board is affected, which has a 1.8V SPI flash on board, so your board should work. Try running the blink example from VSC.

I tried all 3 examples (arduino-link, espidf-arduino-blink and espidf-blink) and none worked. But it doesn't really matter, everything else works so i'm fine 😄 I'll give it another try in better times.

DavePlusPlus, 2022-10-10:
Hi, I just purchased a ESP32-S3 N32R8V and encountered the same problem, is there any way to access the source code of the preloaded firmware? as others have stated before, when the device first booted, the RGB LED was blinking super bright and changing colors.

ptrandem, 2022-11-22:
In case this helps others using the N32R8V variant of the ESP32-S3-DevKitC-1 (v 1.1), I found that the `CONFIG_BLINK_GPIO` needs to be using pin `38`, not the default `48`. You can set this using the terminal cmd `idf.py menuconfig`, selecting "Example Configuraiton" and changing "Blink GPIO Number" to `38`.

QnBarb, 2023-02-10:
Could someone advise on how I can remap the pin as suggest by ptrandem, but using Arduino IDE?

AlbCod, 2023-06-26:
If it helps, I had the same problem with a [LIOLIN S3 PRO](https://www.wemos.cc/en/latest/s3/s3_pro.html#s3-pro) , (which uses PIN38) and I solved it by setting on Arduino (IDE 2.1.0) ESP32S3 Dev Module instead of LOLIN S3 Pro as Board. Now the led works perfectly.
I think it's a configuration issue.

QnBarb, 2023-06-28:
> I have the N32R8V variant of the ESP32-S3-DevKitC-1 (v 1.1), using Arduino
> IDE 2.1.0 and I can't make the RGB to work, the board setting I'm using is
> ESP32S3 Dev Module, and I still can't make it blink's, I've tried putting
> it high (getting 3.3V) but nothing, I've should measure the signals when it
> arrived on pin 38

Did you follow the settings I had? Also, have you uncommented the
brightness code line? look for the line with brightness 64, uncomment that.

Andreipoe1, 2023-06-28:
Yes, even set it to 200

Andreipoe1, 2023-06-29:
> Here is the setting that works for this board (for me): ESP32-S3DEVKITC-1-N32R8V (32MB): USB CDC On Boot: Enabled CPU Frequency: 240MHz (WiFi) Core Debug Level: None USB DFU On Boot: Disabled Erase All Flash Before Sketch Upload: Disabled Events Run On: Core 1 Flash Mode: DIO 80MHz Flash Size: 4MB (32Mb) JTAG Adapter: Integrated USB JTAG Arduino Runs On: Core 1 USB Firmware MSC On Boot: Disabled Partition Scheme: Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS) PSRAM: OPI PSRAM Upload Mode: UART0/Hardware CDC Upload Speed: 921600 USB Mode: Hardware CDC/JTAG
> […](#)

Yep, I use the same settings

Andreipoe1, 2023-06-30:
[image]
Soo running the RGB code example on  ESP32-S3DEVKITC-1-N32R8V 
What is with all the noise?
[image]
Here's another one with 1 second per division
And I have 2 esp
[image]

Idk what to do

Andreipoe1, 2023-07-02:
I've figured out, I had to add an additional line #define RGB_BUILTIN 38
and now it works 👍

PatrickMons, 2023-07-06:
I buy a board esp32-s3 wroom-1 MON8R2
the led is not working because the resistor (R17) is missed and the circuit versus pin 48 is open now I will try to close and see what's happen...

levinhtin, 2023-08-28:
My board is not working too,
then I connect that, and now it's working.

[image]

code: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/BlinkRGB/BlinkRGB.ino

phillipblakesev1tech, 2023-11-05:
Still not working.  Have several of the ESP32S3-WOORM2 (ESP32-S3DEVKITC-1-N32R8V). Tried all of the IDE options and seem to be missing what you guys did to fix.  Anyone using Arduino IDE 2.x.x if you got working, please show your settings.  I read this entire string and seems like at least a few may have found the magic configuration sequence. Would apricate the help.

packetkiller, 2023-11-23:
Found the issue. There is nothing wrong with the board. Arduino seems to think there is only one ESP32S3 board configuration out there and refuses to change their board GPIO configurations.  In this case, the board is a ESP32-WROOM2 DevKitC v1.  This board uses GPIO38 to talk to the built-in addressable LED. If you use the Arduino libraries (or anyone else's libraries based on Arduino; like Adafruit) for the ESP32S3 GPIO definition, the you will get GPIO48 and some others instead.  Since there are several models of ESP32S3 from Espressif, there should be at least 3 board configurations in the libraries that I know of (maybe more).  To fix this you will need to clone any library you utilize in your sketches and fix all GPIO references to the lowest level.

Max-LAB, 2023-12-25:
If someone wants to run ESP-IDF blink example on [Adafruit ESP32-S3 Feather](https://learn.adafruit.com/adafruit-esp32-s3-feather) board with integrated NeoPixel integrated LED. As [documentation](https://learn.adafruit.com/adafruit-esp32-s3-feather/pinouts#neopixel-and-red-led-3115992) says there is a NeoPixel power pin that needs to be pulled high for the NeoPixel to work. 

[image]

Here we are talking about `NEOPIXEL_POWER` GPIO pin 21. 

[image]

Using `menuconfig` you can configure `sdconfig` to use GPIO pin 33 to control NeoPixel LED.

[image]

Good luck

[5 more replies]

closed 2024-08-08 by ESP-Marius