espressif/esp-idf #8550: MQTT client doesn't handle long data operations correctly

KaeLL, 2022-03-11:
### Problem Description
Data reception through MQTT on slow connections (or large enough data) monopolizes the mqtt task by repeatedly dispatching `MQTT_EVENT_DATA` events. This causes the client to stop sending PINGREQs, the first issue, and in turn, depending on the keepalive interval and/or connection speed and/or total data size, the broker may drop the client before it finishes receiving all data. When this happens, the client goes bonkers, endlessly dispatching the data event with the same payload and `TRANSPORT_BASE: tcp_read error, errno=Socket is not connected` right after it, which is the 2nd issue.

### Expected Behavior
Graceful handling of connection drop during data reception operation;
Prioritization of housekeeping chores over data reception;

### Steps to reproduce
mqtt/tcp example, only modification being keepalive set to 5 seconds (helps reproducing it faster) on wifi;
Big enough file to simulate a long data reception operation (I used a 255MiB file);
Publish the file to a topic the client is connected to and wait ~20s.

KaeLL, 2022-03-18:
2nd issue I mentioned is already tracked by #8482. I reopened this to track the 1st issue of mqtt task stop sending PINGREQs during data reception. #8587 seems to be related.

@david-cermak thoughts?

euripedesrocha, 2022-03-22:
@KaeLL thanks for reporting, we have this issue under investigation at the moment.