espressif/esp-idf #8551: Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

firstnamenie, 2022-03-11:
ELF file SHA256: b8b886e3804b1cdc

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
entry 0x40080618
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40145fc2  PS      : 0x00060233  A0      : 0x800d6f88  A1      : 0x3ffcee70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb78c8  A5      : 0x0000040e  
A6      : 0x3ffceea0  A7      : 0x3ffbf1a8  A8      : 0x00000000  A9      : 0x3ffcee40  
A10     : 0x00000000  A11     : 0x0000040e  A12     : 0x40083344  A13     : 0x00000000  
A14     : 0x3ffceea0  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0

Alvin1Zhang, 2022-03-11:
@firstnamenie Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

firstnamenie, 2022-03-11:
> @firstnamenie Thanks for reporting. Would you please help provide more details as suggested in the [issue [...]

I use ESP-IDF4.4 ,hardware-esp32-WROOM-32,Use micropython1.18 pack

Dazza0, 2022-03-11:
@firstnamenie 

> EXCVADDR: 0x00000008

There's an attempt to store at an illegal address (0x00000008). Looks like a NULL pointer access.

> Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0

Could you please decode the backtrace to find out where the illegal store is occurring?

firstnamenie, 2022-03-15:
> @firstnamenie
> 
> > EXCVADDR: 0x00000008
> 
> There's an attempt to store at an illegal address (0x00000008). Looks like a NULL pointer access.
> 
> > Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0
> 
> Could you please decode the backtrace to find out where the illegal store is oc

> @firstnamenie
> 
> > EXCVADDR: 0x00000008
> 
> There's an attempt to store at an illegal address (0x00000008). Looks like a NULL pointer access.
> 
> > Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0
> 
> Could you please decode the backtrace to find out where the illegal store is occurring?

ELF file SHA256: 2c1c417434bacd19

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
0x40080400: _init at ??:?

entry 0x40080618
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40145fc2  PS      : 0x00060233  A0      : 0x800d6f88  A1      : 0x3ffcee70  
0x40145fc2: uart_isr_register at /home/zhongshang/esp/esp-idf/components/driver/uart.c:594

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb78c8  A5      : 0x0000040e  
A6      : 0x3ffceea0  A7      : 0x3ffbf1a8  A8      : 0x00000000  A9      : 0x3ffcee40  
A10     : 0x00000000  A11     : 0x0000040e  A12     : 0x40083344  A13     : 0x00000000  
0x40083344: uart_irq_handler at /home/zhongshang/esp/micropython-1.18/ports/esp32/build/../uart.c:46

A14     : 0x3ffceea0  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0 
0x40145fbf: uart_isr_register at /home/zhongshang/esp/esp-idf/components/driver/uart.c:594

0x400d6f85: uart_init at /home/zhongshang/esp/micropython-1.18/ports/esp32/build/../uart.c:41

0x400d6e0d: mp_task at /home/zhongshang/esp/micropython-1.18/ports/esp32/build/../main.c:96

firstnamenie, 2022-03-15:
> @firstnamenie
> 
> > EXCVADDR: 0x00000008
> 
> There's an attempt to store at an illegal address (0x00000008). Looks like a NULL pointer access.
> 
> > Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0
> 
> Could you please decode the backtrace to find out where the illegal store is occurring?

ELF file SHA256: 2c1c417434bacd19

Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
0x40080400: _init at ??:?

entry 0x40080618
Guru Meditation Error: Core 0 panic'ed (StoreProhibited). Exception was unhandled.

Core 0 register dump:
PC : 0x40145fc2 PS : 0x00060233 A0 : 0x800d6f88 A1 : 0x3ffcee70
0x40145fc2: uart_isr_register at /home/zhongshang/esp/esp-idf/components/driver/uart.c:594

A2 : 0x00000000 A3 : 0x00000000 A4 : 0x3ffb78c8 A5 : 0x0000040e
A6 : 0x3ffceea0 A7 : 0x3ffbf1a8 A8 : 0x00000000 A9 : 0x3ffcee40
A10 : 0x00000000 A11 : 0x0000040e A12 : 0x40083344 A13 : 0x00000000
0x40083344: uart_irq_handler at /home/zhongshang/esp/micropython-1.18/ports/esp32/build/../uart.c:46

A14 : 0x3ffceea0 A15 : 0x00000000 SAR : 0x00000013 EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000008 LBEG : 0x00000000 LEND : 0x00000000 LCOUNT : 0x00000000

Backtrace:0x40145fbf:0x3ffcee700x400d6f85:0x3ffceea0 0x400d6e0d:0x3ffceed0
0x40145fbf: uart_isr_register at /home/zhongshang/esp/esp-idf/components/driver/uart.c:594

0x400d6f85: uart_init at /home/zhongshang/esp/micropython-1.18/ports/esp32/build/../uart.c:41

0x400d6e0d: mp_task at /home/zhongshang/esp/micropython-1.18/ports/esp32/build/../main.c:96

Dazza0, 2022-03-15:
@firstnamenie Seems like a micropython issue. As seen from their v1.18 source code, the `uart_init()` function ([link](https://github.com/micropython/micropython/blob/da4b38e7562dfa451917f9d7f344a7f26de8c7bd/ports/esp32/uart.c#L39)) attempts to register a UART ISR before calling `uart_driver_install()`.

However, they've already fixed it on their master branch ([commit here](https://github.com/micropython/micropython/commit/2cc9232781d818b8a317dc7b173ecd5b752867e7))

Please try updating your micro-python version. I'm closing this issue as the root cause isn't in ESP-IDF. Please feel free to open if the issue isn't resolved.

closed 2022-03-15 by Dazza0