espressif/esp-idf #8552: flash_download_tool  Error: Unable to set default locale: 'unsupported locale setting'

CAIJUN111, 2022-03-11:
The tool I downloaded doesn't work 
[image]
Can anyone help me？
I have never come across such a problem
The flash tool version is flash_download_tool_3.9.2

arosyc, 2022-03-11:
Maybe config file is broken.
Could you delete config folder and try again?

CAIJUN111, 2022-03-11:
I tried, but it didn't work. I can use the same file on another computer

arosyc, 2022-03-11:
okey, Could you download and try again. If it does not work, I will try to reproduce your issure.

CAIJUN111, 2022-03-14:
I tried, but it didn't work

arosyc, 2022-03-15:
It looks like the problem is caused by the version of the UI library the tool depends on. However, this problem should not affect the normal use of the download tool, and we will fix this problem in the next version.

Alvin1Zhang, 2024-12-17:
Thanks for reporting and sharing the updates, sorry for the very slow turnaround, the issue has been fixed, feel free to reopen if the issue still happens. Thanks.

closed 2024-12-17 by Alvin1Zhang