espressif/esp-idf #8554: esp_http_client_perform returns ESP_ERR_HTTP_FETCH_HEADER sometimes although server is proper

sduddell, 2022-03-11:
Hi,

I'm trying to post a HTTP request using esp_http_client_perform(), but it returns ESP_ERR_HTTP_FETCH_HEADER. 
I debugged further and found that, ESP_ERR_HTTP_FETCH_HEADER is being returned because of esp_transport_read() returning 0.

```
int esp_http_client_fetch_headers(esp_http_client_handle_t client)
{
    ESP_LOGI(TAG,"client state in fetching header: %d",client->state);
    if (client->state < HTTP_STATE_REQ_COMPLETE_HEADER) {
        return ESP_FAIL;
    }

    client->state = HTTP_STATE_REQ_COMPLETE_DATA;
    esp_http_buffer_t *buffer = client->response->buffer;
    client->response->status_code = -1;

    while (client->state < HTTP_STATE_RES_COMPLETE_HEADER) {
        buffer->len = esp_transport_read(client->transport, buffer->data, client->buffer_size_rx, client->timeout_ms);
        ESP_LOGI(TAG,"buffer size rx: %d , buffer len: %d",client->buffer_size_rx,buffer->len );
        if (buffer->len <= 0) {
            return ESP_FAIL;
        }
        http_parser_execute(client->parser, client->parser_settings, buffer->data, buffer->len);
    }
```
buffer length returned by esp_transport_read is 0 after 2 iterations in read function, is this not a valid case? should the condition `if (buffer->len <= 0)` need to be changed to `if (buffer->len <0)` , or please let me know if I'm missing anything here.
Thanks for the help!

shubhamkulkarni97, 2022-03-11:
Hi @sduddell, thanks for reporting this issue.

Can you please share URL which reproduces this issue (if it can be reproduced with public URL)?

esp_transport_read returning 0 signifies that the underlying connection is closed and ideally connection should not be closed in fetch header state.

Is your issue fixed if you change the check to `if (buffer->len <0)`?

sduddell, 2022-03-11:
Hi @shubhamkulkarni97 ,

Thanks for quick reply.
Yes, issue got fixed after changing to `if (buffer->len <0)` .
I haven't tried on public URL.

shubhamkulkarni97, 2022-03-14:
Hi @sduddell, I was not able to reproduce this issue and looking at the code I don't think `if (buffer->len <= 0)` should cause an issue.

It would be helpful if you could share sample code along with URL to reproduce this issue.

sduddell, 2022-03-18:
Hi Shubham,

Unfortunately, I cannot share the URL. 
I'm not sure how but with the change `if (buffer->len < 0)` I could not reproduce issue. I have a question, do you see any issue if `if (buffer->len < 0)` is retained in the code.

shubhamkulkarni97, 2022-04-25:
Hi @sduddell, I don't see any issue in making this change. But we need test code to verify the fix and check if it does not introduce any other issues.

AxelLin, 2022-04-27:
@sduddell 
1. Which esp-idf version are you using? (git describe --tags output)
2. Can you print errno when buffer->len is 0?

AxelLin, 2022-04-27:
> esp_transport_read returning 0 signifies that the underlying connection is closed and ideally connection should not be closed 
Or read timeout?

AxelLin, 2022-05-11:
> Hi Shubham,
> 
> Unfortunately, I cannot share the URL. I'm not sure how but with the change `if (buffer->len < 0)` I could not reproduce issue. I have a question, do you see any issue if `if (buffer->len < 0)` is retained in the code.

@sduddell 

Could you try reproduce the issue with public URL? Otherwise it's difficult to understand what could be wrong.

sduddell, 2022-05-11:
Hi @AxelLin ,

I'm using ESP-IDF version of 4.4.
I tried to reproduce the issue, but I cannot reproduce it now with our URL and public URL, suspecting if network/wifi issue was also a factor previously. 
I think we can close the thread, thanks much!!

Alvin1Zhang, 2022-05-11:
@sduddell Thanks for sharing the updates, thanks to @AxelLin for the notes, feel free to reopen.

closed 2022-05-11 by Alvin1Zhang

kzoltaan, 2024-03-20:
I have same problem when the target webserver is Apache. On Nginx looks okay. I any new idea?

kzoltaan, 2024-03-20:
okay. Increase the KeepAliveTimeout on Apache and the msg disappear.

dizcza, 2024-08-30:
> okay. Increase the KeepAliveTimeout on Apache and the msg disappear.

Thanks for the hint. Adding `--keep-alive 10` to `gunicorn` command solved my problem as well.