espressif/esp-idf #8555: BLE peripheral nimble unable to terminate connection

AngelVersa, 2022-03-11:
## Environment

- Development Kit:      [none]
- Module or chip used:  ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):  v4.3.2
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes] VS code
- Power Supply:         [external 3.3V]

## Problem Description
The ESP32 works as ble peripheral with nimble. When use ble_gap_terminate(connectionHandler, BLE_ERR_CONN_LIMIT) the connection is not terminated and returns 0x212 error.

### Expected Behavior
Close connection with the device with the connection handler passed to ble_gap_terminate()

### Actual Behavior
Returns 0x212 error, and not disconnect the device
### Steps to reproduce

1. Receive a connection event (BLE_GAP_EVENT_CONNECT) in the gap event callback
2. Call ble_gap_terminate() with the connection handler gave by the connection event

### Code to reproduce this issue

```cpp
//******************************************************************************
int BLE_GapEvent(bleGapEvent *event, void *arg) 
//******************************************************************************
{

  switch (event->type) 
  {
  case BLE_GAP_EVENT_CONNECT:
    // A new connection was established or a connection attempt failed 
    ESP_LOGI(TAG, "connection %s; connHnd=%u",
                event->connect.status == 0 ? "established" : "failed",
                event->connect.conn_handle);
    int ret = ble_gap_terminate(event->connect.conn_handle, BLE_ERR_CONN_LIMIT);
    if (ret != 0) { ESP_LOGE(TAG, "Error 0x%X disconnecting 0x%04X", ret, pendingCmd.connHnd);}
    break;
  }
```
Also, I tried the same in another thread that manage write request on characteristics with the same result.

## Debug Logs

```
I (73442) BLE_PER: connection established; connHnd=0
E (73445) BLE_PER: Error 0x212 disconnecting 0x0000
```

rahult-github, 2022-03-14:
Hi @AngelVersa ,

 The reason code passed to disconnect command in your example BLE_ERR_CONN_LIMIT, is not in the list of valid reason as specified by spec. 

In BT spec, Vol 4, Part E, you can refer to the values that the reason parameter can have.  In your above example, changing the BLE_ERR_CONN_LIMIT to any of the valid reason code should work.

[image]
 

Thanks,
 Rahul

AngelVersa, 2022-03-14:
Hi @rahult-github,

Thanks, this fixed my problem. 

But maybe I would suggest to update function description to include this limitation in both, in the code and in the documentation. As is difficult to found this limitation (for example this chapter is not present in BLE standard 4.2).

rahult-github, 2022-03-14:
Hi @AngelVersa ,
 
My apologies for not mentioning the BT version i was referring to. In BT 4.2, Vol2 , Part E  , section 7.1.6 has this information. Please check the Disconnect command description , under Link Control Commands . 

Thanks,
 Rahul

closed 2022-03-14 by rahult-github