espressif/esp-idf #8556: Invalid header error on the monitor

filgra96, 2022-03-11:
## Environment

- Development Kit: custom board
- Module or chip used:  ESP32-S2-MINI-1
- IDF version: v4.4
- Build System: idf.py
- Operating System: Windows
- Using an IDE?: VSCode (Espressif IDF extention)
- Power Supply: external 3.3V from another device, using USB.

## Problem Description

I'm using a custom board (I can't share schematics) that mounts an ESP32-S2-MINI-1 on a PCB. Power supply comes from an external unit via USB.

All was working well until we decided to mount also a 16MB flash memory (W25Q128JV) on the same PCB.
Before then I was able to use the internal partitions (SPIFFS, NVS, OTA), to communicate with the outer device via USB (it provides also the power supply), to connect to the WiFi, ecc... I made these tests by using the ESP32-S2 Dev board.

To use the flash I studied the example inside _storage/ext_flash_fatfs_. Every time I program it, I obtain the following output log.

> I (188) cpu_start: Pro cpu up.
> I (200) cpu_start: Pro cpu start user code
> I (200) cpu_start: cpu freq: 160000000
> I (200) cpu_start: Application information:
> I (205) cpu_start: Project name:     ext_flash_fatfs
> I (210) cpu_start: App version:      1
> I (215) cpu_start: Compile time:     Mar 11 2022 12:05:14
> I (221) cpu_start: ELF file SHA256:  d2f1cec4da9b421e...
> I (227) cpu_start: ESP-IDF:          v4.4
> I (232) heap_init: Initializing. RAM available for dynamic allocation:
> I (239) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
> I (245) heap_init: At 3FFBFEF0 len 0003C110 (240 KiB): DRAM
> I (251) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
> I (258) spi_flash: detected chip: generic
> I (262) spi_flash: flash io: dio
> I (267) cpu_start: Starting scheduler on PRO CPU.
> I (272) example: Initializing external SPI Flash
> I (272) example: Pin assignments:
> I (272) example: MOSI: 35   MISO: 37   SCLK: 36   CS: 38
> I (282) spi_flash: detected chip: generic
> I (282) spi_flash: flash io: dio
> I (292) example: Initialized external Flash, size=16384 KB, ID=0x204016
> I (302) example: Adding external Flash as a partition, label="storage", size=16384 KB
> I (302) example: Listing data partitions:
> I (312) example: - partition 'nvs', subtype 2, offset 0x9000, size 24 kB
> I (322) example: - partition 'phy_init', subtype 1, offset 0xf000, size 4 kB     
> I (322) example: - partition 'storage', subtype 129, offset 0x0, size 16384 kB   
> I (332) example: Mounting FAT filesystem
> W (342) vfs_fat_spiflash: f_mount failed (13)
> I (342) vfs_fat_spiflash: Formatting FATFS partition, allocation unit size=4096  
> I (722) vfs_fat_spiflash: Mounting again
> I (722) example: FAT FS: 16212 kB total, 16212 kB free
> I (722) example: Opening file
> I (952) example: File written
> I (952) example: Reading file
> I (952) example: Read from file: 'Written using ESP-IDF v4.4'

**This is true the first time I program it!!**. When I unplug and plug again the device the log became the following (I will attach only a little part of the thousand lines that appear all together):

> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalidESP-ROM:esp32s2-rc4-20191025
> Build:Oct 25 2019
> rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb
> invalid header: 0x4d90feeb

It starts instantly doing this, I don't know how to catch the first lines....

### Actual Behavior

I think it shouldn't format every time the external flash every time I program it.
Also, the major issue is that I can't do anything, obtaining the behavior of the "inifinite" invalid header error.
I can only re-programming it obtaining the first log I attached. And it continues..
I tried to use the power supply generated by the PC (instead of the other device): same issue.
I can try to switch to another identical device, but I can't until monday.
Finally, worth to mention it, uploading a different example (for example to test the WiFi), it works normally.

### Code to reproduce this issue

The official example: [ext_flash_fatfs](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/storage/ext_flash_fatfs).

## Other items if possible

My sdkconfig as textfile [sdkconfig.txt](https://github.com/espressif/esp-idf/files/8233563/sdkconfig.txt)

negativekelvin, 2022-03-11:
>partition 'storage', subtype 129, offset 0x0, size 16384 kB

Well if you configure your fat partition to be the whole chip it is going to overwrite the bootloader, partition table, app, etc...  maybe you didn't realize the address space is shared?

Really it should not

negativekelvin, 2022-03-11:
It seems like the partition is getting mounted on the main flash chip instead and corrupting it.

filgra96, 2022-03-14:
> It seems like the partition is getting mounted on the main flash chip instead and corrupting it.

Thank you, this is a first clue to understand the solution.
I admit I changed few lines in order to make the code working.

    static esp_flash_t* example_init_ext_flash(void)
    {
        const spi_bus_config_t bus_config = {
            .mosi_io_num = 35,//VSPI_IOMUX_PIN_NUM_MOSI, --Changed
            .miso_io_num = 37,//VSPI_IOMUX_PIN_NUM_MISO, --Changed
            .sclk_io_num = 36,//VSPI_IOMUX_PIN_NUM_CLK, --Changed
            .quadhd_io_num = -1,//VSPI_IOMUX_PIN_NUM_HD, --Changed
            .quadwp_io_num = -1,//VSPI_IOMUX_PIN_NUM_WP, --Changed
        };

        const esp_flash_spi_device_config_t device_config = {
            //.host_id = FSPI_HOST, --Commented
            .cs_id = 0,
            .cs_io_num = 38,//VSPI_IOMUX_PIN_NUM_CS, --Changed
            .io_mode = SPI_FLASH_DIO,
            .speed = ESP_FLASH_40MHZ,
        };

        ESP_LOGI(TAG, "Initializing external SPI Flash");
        ESP_LOGI(TAG, "Pin assignments:");
        ESP_LOGI(TAG, "MOSI: %2d   MISO: %2d   SCLK: %2d   CS: %2d",
            bus_config.mosi_io_num, bus_config.miso_io_num,
            bus_config.sclk_io_num, device_config.cs_io_num
        );

        // Initialize the SPI bus
        ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &bus_config, 1));

        // Add device to the SPI bus
        esp_flash_t* ext_flash;
        ESP_ERROR_CHECK(spi_bus_add_flash_device(&ext_flash, &device_config));

        ext_flash->size = 16*1024*1024;

        // Probe the Flash chip and initialize it
        esp_err_t err = esp_flash_init(ext_flash);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "Failed to initialize external Flash: %s (0x%x)", esp_err_to_name(err), err);
            return NULL;
        }

        // Print out the ID and size
        uint32_t id;
        ESP_ERROR_CHECK(esp_flash_read_id(ext_flash, &id));
        ESP_LOGI(TAG, "Initialized external Flash, size=%d KB, ID=0x%x", ext_flash->size / 1024, id);

        return ext_flash;
    }

I commented lines that are different in my case. Also, I removed the line of the *host_id* (I think it will use the default value).
All the remaining code is identical.

negativekelvin, 2022-03-14:
>Also, I removed the line of the host_id (I think it will use the default value).

I think that is the problem, you have to use SPI2_HOST or FSPI_HOST

filgra96, 2022-03-14:
You're right, now it's working fine! Using `FSPI_HOST` inside *host_id* the bug is solved.

negativekelvin, 2022-03-14:
@igrr I think this is a bug that this conflict is not detected by the driver and can result in a bricked device

closed 2022-04-26 by espressif-bot