espressif/esp-idf #8558: Race condition at startup corrupts memory

pickaxprograms, 2022-03-11:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER
- IDF version : v4.4-12-ga0404a9ed2
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: WSL 
- Using an IDE?: No

## Problem Description

Core panic when starting scheduler on APP CPU. main_task() resumes too quickly

### Expected Behavior

System to boot properly

### Actual Behavior

Reboot loop

### Steps to reproduce

Migrating project to v4.4. Project runs fine with v4.3.2

## Debug Logs
...
I (1858) spiram: Adding pool of 4082K of external SPI memory to heap allocator
Pause app (enter bootloader mode), press Ctrl-T Ctrl-R to restart
I (2410) spi_flash: detected chip: generic
I (2411) spi_flash: flash io: dio
I (2421) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x400803c0  PS      : 0x00060436  A0      : 0xfefefefe  A1      : 0x3ffe7d60  
0x400803c0: _DoubleExceptionVector at .../esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:461

A2      : 0x00005dc0  A3      : 0x11877cb5  A4      : 0x00000019  A5      : 0x00000036  
A6      : 0x00000000  A7      : 0x3ffe7cc4  A8      : 0x00000000  A9      : 0x3ffbe6b0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffbe690  
A14     : 0x400847fc  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
0x400847fc: ipc_task at .../esp-idf/components/esp_ipc/src/esp_ipc.c:52

EXCVADDR: 0xfefefede  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400803bd:0x3ffe7d600x7efefefb:0x3ffbe5b0  |<-CORRUPTED

## Other items if possible
```
static void main_task(void* args)
{
#if !CONFIG_FREERTOS_UNICORE
    // Wait for FreeRTOS initialization to finish on APP CPU, before replacing its startup stack
    while (port_xSchedulerRunning[1] == 0) {
        ;
    }

    // [refactor-todo] check if there is a way to move the following block to esp_system startup
    heap_caps_enable_nonos_stack_heaps();

    // Now we have startup stack RAM available for heap, enable any DMA pool memory
#if CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL
    if (g_spiram_ok) {
        esp_err_t r = esp_spiram_reserve_dma_pool(CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL);
        if (r != ESP_OK) {
            ESP_EARLY_LOGE(TAG, "Could not reserve internal/DMA pool (error 0x%x)", r);
            abort();
        }
    }
#endif
...
```
```
BaseType_t xPortStartScheduler( void )
{
    portDISABLE_INTERRUPTS();
    // Interrupts are disabled at this point and stack contains PS with enabled interrupts when task context is restored

#if XCHAL_CP_NUM > 0
    /* Initialize co-processor management for tasks. Leave CPENABLE alone. */
    _xt_coproc_init();
#endif

    /* Setup the hardware to generate the tick. */
    vPortSetupTimer();

    port_xSchedulerRunning[xPortGetCoreID()] = 1;

    // Cannot be directly called from C; never returns
    __asm__ volatile ("call0    _frxt_dispatch\n");

    /* Should not get here. */
    return pdTRUE;
}
```

The race condition is `port_xSchedulerRunning[1]` is set to one and then `_frxt_dispatch` is executed. Internal to `_frxt_dispatch` there are other C calls which need the stack. The startup stack for the App core is thus still in use when `esp_spiram_reserve_dma_pool()` is called. Instead of waiting on `port_xSchedulerRunning[1]`, `main_task()` should wait on `int_wdt_app_cpu_ticked` instead to make sure the start stack is indeed done with.

mahavirj, 2022-03-15:
@pickaxprograms 

Can you please share your `sdkconfig`?

> main_task() should wait on int_wdt_app_cpu_ticked instead to make sure the start stack is indeed done with.

This may not work if someone disables this `CONFIG_ESP_INT_WDT`. I think we can register `idle` hook for APP cpu core and wait till callback executes from `idle` task on APP cpu core to ensure that scheduler initialization has been completed. We will evaluate this approach.

pickaxprograms, 2022-03-16:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8261979/sdkconfig.txt)

laukik-hase, 2022-04-27:
Hello, @pickaxprograms and @winzkigermany!

Could you try out this patch - [startup_crash.zip](https://github.com/espressif/esp-idf/files/8568840/startup_crash.zip) and let us know if it works for you?

Also, the crash mentioned in the issue description occurs when `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` is set to low values (< 128).

> Does this race condition have anything to do with the XMC flash init workaround?

@winzkigermany, could you try the fix attached and evaluate this?

winzkigermany, 2022-04-27:
Hi @laukik-hase 
I've tried the fix, but it seems the callback is never called. Now I see the last log:
```
I (965) cpu_start: Starting scheduler on PRO CPU.
I (18) cpu_start: Starting scheduler on APP CPU.
```

and there execution stays forever (presumably staying in the while loop waiting for `_app_cpu_startup_done` to become true )

Now I'll look for your hint about `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL`

winzkigermany, 2022-04-28:
@laukik-hase Could it be that the problem pops up later?
I just added a vTaskDelay(1); in the while loop of your fix, in order not to choke the cpu.
As a result, the callback is called properly, and the code execution goes on.

```
I (965) cpu_start: Starting scheduler on PRO CPU.
D (0) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (976) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
 (982) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (988) cpu_start: Starting scheduler on APP CPU.
**I (970) cpu_start: registered esp_register_freertos_idle_hook result: 0**
**I (1002) cpu_start: esp_deregister_freertos_idle_hook result: 0**
D (1006) heap_init: New heap initialised at 0x3ffe0440
D (1012) heap_init: New heap initialised at 0x3ffe4350
I (1018) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (1026) spiram: Allocating block of size 32768 bytes
D (1030) intr_alloc: Connected src 16 to int 9 (cpu 0)
D (1036) intr_alloc: Connected src 46 to int 12 (cpu 0)
```

After the last line, the cpu resets, and the reboot loop starts. So probably it's not a racing condition around the time when your fix should wait for APP CPU to enter idle state the first time?

winzkigermany, 2022-04-28:
Indeed, that's it. I've been hunting a red herring. The problem appears later in the code, it's even in app_main.
Now I can follow the track on. Sorry I can't help you more with the original problem. (Possibly it's due to a problem later on in the code, too)

mahavirj, 2022-07-11:
Fix merged with https://github.com/espressif/esp-idf/commit/b38ba40d4b945ed0c630ee7b61bbe5dca07d57f0

closed 2022-07-11 by mahavirj