espressif/esp-idf #8560: ESP32S3 seemingly "bricked" after flash from the latest IDF

JadedTuna, 2022-03-11:
## Environment

- Development Kit:      `[ESP32-S3-DevKitC-1]`
- Kit version (for WroverKit/PicoKit/DevKitC): `[v1.0]`
- Module or chip used:  `[ESP32-S3-WROOM-1]`
- IDF version (run ``git describe --tags`` to find it):
    `// v5.0-dev-1996-ga8ba5a0264`
- Build System:         `[idf.py]`
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    `// xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0`
- Operating System:     `[Linux]`
- (Windows only) environment type: `[-]`.
- Using an IDE?: `[No]`
- Power Supply:         `[USB]`

## Problem Description

After updating to the latest IDF (master) and flashing the blink project, the module will no longer boot, no matter what else is flashed (different projects, configurations, etc). This behaviour persists after rolling back to the previous IDF version used (v5.0-dev-1452-g93106c9348). This has been reproduced on two modules.

**N.B**: I'm not sure if bricked is the right word here -- the module can still be flashed, but no matter what, it's just stuck in the same boot loop. Another thing that changed since the previous IDF version is the compiler -- the new one has `patch3` in the name, so possibly a compiler issue?

### Expected Behavior

Flashed applications work without issue.

### Actual Behavior

The following is output to the serial console on repeat:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16e0
load:0x3b6000ff,len:0xbb840
Invalid image block, can't boot.
ets_main.c 329 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16e0
load:0x3b6000ff,len:0xbb840
Invalid image block, can't boot.
ets_main.c 329 
```

### Steps to reproduce

1. `idf.py set-target esp32s3` in the blink example
2. Compile blink example with the mentioned IDF version
3. Flash to an ESP32-S3-DevKitC-1 kit (E2N8R2)

### Code to reproduce this issue

Included blink example, nothing changed.

## Debug Logs

No debug logs, but here is output from the flasher:
```
esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:41:45:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00041fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21104 bytes to 13184...
Writing at 0x00000000... (100 %)
Wrote 21104 bytes (13184 compressed) at 0x00000000 in 0.4 seconds (effective 417.8 kbit/s)...
Hash of data verified.
Compressed 204240 bytes to 107132...
Writing at 0x00010000... (14 %)
Writing at 0x0001c191... (28 %)
Writing at 0x00021b74... (42 %)
Writing at 0x00027d06... (57 %)
Writing at 0x0002e84b... (71 %)
Writing at 0x000388f7... (85 %)
Writing at 0x0003e6a5... (100 %)
Wrote 204240 bytes (107132 compressed) at 0x00010000 in 1.5 seconds (effective 1066.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 478.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

## Attachments
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8235222/sdkconfig.txt)
[blink.elf.tar.gz](https://github.com/espressif/esp-idf/files/8235227/blink.elf.tar.gz)

JadedTuna, 2022-03-11:
Update: This seems similar to the issue outlined in #7994. One of the suggestions actually fixed the boot loop:
```bash
$ esptool.py write_flash_status --bytes 3 --non-volatile 0
Initial flash status: 0x230200
Setting flash status: 0x000000
After flash status:   0x000200
```

I did this for both modules, and they are alive again. Flashing other projects works without any issues as well.

---

I've pinpointed the problem down further. The very first project I flashed with the new IDF version was configured for QIO 120 MHz flash mode (blink sketch had default configuration with DIO 80 MHz). Flashing any project in QIO 120 MHz _OR_ DIO 120 MHz mode results in exactly the same situation. The code runs until some issue is encountered, hangs for ~10 seconds, and then restarts into the boot loop. Here is the output (blink example, DIO at 120 MHz):
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40056f6f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16e8
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f78
entry 0x403b6254
I (24) boot: ESP-IDF v5.0-dev-1996-ga8ba5a0264-dirty 2nd stage bootloader
I (25) boot: compile time 22:33:54
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08a40h ( 35392) map
I (100) esp_image: segment 1: paddr=00018a68 vaddr=3fc92290 size=02c44h ( 11332) load
I (105) esp_image: segment 2: paddr=0001b6b4 vaddr=40374000 size=04964h ( 18788) load
I (115) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18b54h (101204) map
I (137) esp_image: segment 4: paddr=00038b7c vaddr=40378964 size=09928h ( 39208) load
I (146) esp_image: segment 5: paddr=000424ac vaddr=50000000 size=00010h (    16) load
I (152) boot: Loaded app from partition at offset 0x10000
I (153) boot: Disabling RNG early entropy source...
W (167) flash HPM: Enter HPM by reconfiguring dummy has not been fully tested
I (167) flash HPM: Enabling high speed mode for chip XMC
W (169) FLASH: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
```

closed 2022-03-18 by espressif-bot

rob-bits, 2022-08-08:
I have the same issue with ESP32-S3-WROOM-1 M0N16 modules. Randomly once it works once it does not. The boot process fails, I get plenty of different error messages, and sometimes it just run fine.

I got messages such as:

rst:0x3 (RTC_SW_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb872
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1668
load:0x403b6000,len:0xf48
load:0x403ba000,len:0x32d8
entry 0x403b6334
I (24) boot: ESP-IDF v4.4.1-149-g009e66e625 2nd stage bootloader
I (24) boot: compile time 17:17:09
I (24) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
E (52) flash_parts: Incorrect MD5 checksum
E (57) boot: Failed to verify partition table
E (62) boot: load partition table error!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb872
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1668
load:0x403b6000,len:0xf48
load:0x403ba000,len:0x32d8
entry 0x403b6334
I (24) boot: ESP-IDF v4.4.1-149-g009e66e625 2nd stage bootloader
I (24) boot: compile time 17:17:09
I (24) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 app0             OTA app          00 10 00010000 00300000
I (85) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000
I (93) boot: End of partition table
I (414) boot: Detect GPIO 0 active to enter UF2 bootloader
I (415) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2b8cch (178380) map
I (445) esp_image: segment 1: paddr=0003b8f4 vaddr=3fc93c80 size=03eb0h ( 16048) load
I (448) esp_image: segment 2: paddr=0003f7ac vaddr=40374000 size=0086ch (  2156) load
I (451) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9a214h (631316) map
I (554) esp_image: segment 4: paddr=000da23c vaddr=4037486c size=0f414h ( 62484) load
I (567) esp_image: segment 5: paddr=000e9658 vaddr=50000000 size=00010h (    16) load
E (567) esp_image: Checksum failed. Calculated 0x77 read 0x71
E (571) boot: OTA app partition slot 0 is not bootable
E (576) boot: No bootable app partitions in the partition table
rst:0x3 (RTC_SW_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb872
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1668
load:0x403b6000,len:0xf48
load:0x403ba000,len:0x32d8
entry 0x403b6334
I (24) boot: ESP-IDF v4.4.1-149-g009e66e625 2nd stage bootloader
I (24) boot: compile time 17:17:09
I (24) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 app0             OTA app          00 10 00010000 00300000
I (85) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000
I (93) boot: End of partition table
E (97) boot: ota data partition invalid and no factory, will try all partitions
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2b8cch (178380) map
I (141) esp_image: segment 1: paddr=0003b8f4 vaddr=3fc93c80 size=03eb0h ( 16048) load
I (144) esp_image: segment 2: paddr=0003f7ac vaddr=40374000 size=0086ch (  2156) load
I (147) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9a214h (631316) map
I (250) esp_image: segment 4: paddr=000da23c vaddr=4037486c size=0f414h ( 62484) load
I (263) esp_image: segment 5: paddr=000e9658 vaddr=50000000 size=00010h (    16) load
E (263) esp_image: Checksum failed. Calculated 0x67 read 0x71
E (266) boot: OTA app partition slot 0 is not bootable
E (272) boot: No bootable app partitions in the partition table
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021

miguelfernando55, 2023-07-24:
I was having a problem when I putted SPIFFS.begin() on another file and then included on the main. I putted where it was the #include "SPIFFS.h" and the reset problem disappeared.

michelegirolami, 2023-11-29:
HI all, i have a similar issue with lilyho ESP32S3. After flashing a very basic fw it seems the device is bricked. I can upload a new fw but the device does not load the fw. How to fix this? I read I should upload at 80MHZ... this is not clear for me. Any help is apprecioated
M

garvitkhera, 2024-01-23:
Hi @michelegirolami 
I'm also facing similar issues. Did you make it work by any chance?

michelegirolami, 2024-01-25:
Hi @garvitkhera , I fixed this issue and I report the procedure [here](https://github.com/mu-editor/mu/issues/2313#issuecomment-1875757624)