espressif/esp-idf #8566: Why some of the header files included in the IDF example have to be combined with ralative path?

yourmaddad, 2022-03-14:
[https://s3.us-west-2.amazonaws.com/secure.notion-static.com/6d027a4e-9e19-43b5-b26f-95116c522afd/Untitled.png?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20220314%2Fus-west-2%2Fs3%2Faws4_request&X-Amz-Date=20220314T120258Z&X-Amz-Expires=86400&X-Amz-Signature=2189e60a4e2ecba0522d6c69ea7f24cc46d3a71419e2e7805cbc10a7876506b0&X-Amz-SignedHeaders=host&response-content-disposition=filename%20%3D%22Untitled.png%22&x-id=GetObject](url)

while some other has no need to be included with relative path？ like "#include "esp_adc_cal.h""?

yourmaddad, 2022-03-14:
copy the url to web browser to see the screen cut from my IDF code.

KaeLL, 2022-03-14:
It's a poor's man namespace.

ginkgm, 2022-03-15:
Generally it's encouraged to have headers under a namespace like "driver/adc.h". The main purpose is to avoid conflicts of user API with IDF APIs.

But sometimes we didn't pay enough attention when reviewing new headers before they are merged.. And for back-compatibility reasons, it's not proper to change the path now.

Though "esp_adc_cal.h" doesn't have namespace, since it's started with `esp_`, it won't cause conflict probelm for the users, right?

yourmaddad, 2022-03-15:
> Generally it's encouraged to have headers under a namespace like "driver/adc.h". The main purpose is to avoid conflicts of user API with IDF APIs.
>
> But sometimes we didn't pay enough attention when [...]

yes, you are right! thanks for your answer! i got the point now~

and i registerd "include/driver" path in the Cmakelsits.txt file under the IDF_PATH/component/driver,  it will be OK to use "include "adc.h" ".

yourmaddad, 2022-03-15:
感觉EPS_IDF 配置起来对新手确实有点不太友好哈哈，期待后续乐鑫公司的手册更新！^_^ 。

ginkgm, 2022-03-15:
> yes, you are right! thanks for your answer! i got the point now~
>
> and i registerd "include/driver" path in the Cmakelsits.txt file under the IDF_PATH/component/driver, it will be OK to use "include [...]

Another way of doing this without modifying the IDF, is to create a component, with headers only. And include the original headers. For example

new `adc.h`:
```
#include "driver/adc.h"
```

And add this to project requirement, you will be able to use this `adc.h` everywhere.

But we cannot do this, because 3rd party libs may also want to have adc.h. This name is too common. Same for other drivers.

ginkgm, 2022-03-15:
Em.. which part of configuration do you feel not friendly? Could you provide some details so that we can improve it in the future?

yourmaddad, 2022-03-15:
> Another way of doing this without modifying the IDF, is to create a component, with headers only. And include the original headers. For example
>
> new `adc.h`:
>
> ```
> #include "driver/adc.h"
> ```
>
> And [...]

Excellent, now i totally understood the rule(habbit) to include header file from IDF component !

yourmaddad, 2022-03-15:
> Em.. which part of configuration do you feel unfriendly? Could you provide some details so that we can improve it in the future?

it is definitely the environment configuration of ESP_IDF!

when i need some help in  learning how IDF works, i will go to the web of [ESP_IDF programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/index.html), **BUT** the most severe problem is, the CONTENT in [ESP_IDF programming guide]((https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/index.html) is listed based on the alphabet, rather than a reasonable order suitable for beginners. **SO** it is more like a manul than a guide book.

Well, the engineer from espressif are no doubt friendly! just like you !  ^_^

yourmaddad, 2022-03-15:
ESP编程指南如果可以提供一定傻瓜式的对ESP的学习顺序的建议，那就最好了。这样可以方便例如使用stm芯片的庞大的学生用户群体顺利过渡到ESP来。

krzychb, 2022-03-16:
> when i need some help in learning how IDF works, i will go to the web of [ESP_IDF programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/index.html), BUT the most severe [...]

@yourmaddad that's true! 
Do you have some recommendations on how to better organize the content?
Which part of the guide requires attention first? [API Guides](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/index.html) or [API Reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/index.html)?

Alvin1Zhang, 2022-04-12:
Thanks for reporting, would you please share if any updates regarding https://github.com/espressif/esp-idf/issues/8566#issuecomment-1068949537? Thanks.

yourmaddad, 2022-04-13:
> 

抱歉！这段时间在忙毕业论文所以没有来得及回复。详细的我会近期整理一个思维导图在本话题下提交给你们！

yourmaddad, 2022-04-13:
> Thanks for reporting, would you please share if any updates regarding [#8566 (comment)](https://github.com/espressif/esp-idf/issues/8566#issuecomment-1068949537)? Thanks.

当然！我非常乐意为你们提供我的建议和想法，也非常感谢你们愿意倾听我的意见！我的详细的将在近期整理好上传至本话题下。谢谢！

Alvin1Zhang, 2022-04-14:
@yourmaddad 感谢回复，期待更新，谢谢。

Alvin1Zhang, 2023-02-20:
@yourmaddad 您好，请问有什么更新么？谢谢。

closed 2023-04-03 by Alvin1Zhang