espressif/esp-idf #8567: esp_log_level_set() can can create multiple inconsistent entries in cache

pickaxprograms, 2022-03-14:
## Environment

- Module or chip used: ESP32-WROVER
- IDF version :v4.4-12-ga0404a9ed2
  - Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: wsl
- Using an IDE?: No

## Problem Description

Calling esp_log_level_set() can create multiple cache entries in s_log_cache resulting in inconsistent behavior.

### Expected Behavior
Log level to match what is set

### Actual Behavior
Log level may be an older value.

### Steps to reproduce

```cpp
    {
        char string1[20];
        char string2[20];
        sprintf( string1, "WEB");
        sprintf( string2, "WEB");
        esp_log_level_set( string1, 1); 
        printf("Level%s = %d\r\n",  string1, esp_log_level_get(string1));
        esp_log_level_set( string2, 2);
        printf("Level%s = %d\r\n",  string1, esp_log_level_get(string1));
        esp_log_level_set( string1, 3);
        printf("Level%s = %d\r\n",  string1, esp_log_level_get(string1));
        printf("Level%s = %d\r\n",  string2, esp_log_level_get(string2));
    }

```

Output:
```
LevelWEB = 1
LevelWEB = 1 (should be 2)
LevelWEB = 1 (should be 3)
LevelWEB = 3
```

The problem is get_cached_log_level() is checking pointers rather than comparing strings
```
static inline bool get_cached_log_level(const char *tag, esp_log_level_t *level)
{
    // Look for `tag` in cache
    uint32_t i;
    for (i = 0; i < s_log_cache_entry_count; ++i) {
#ifdef LOG_BUILTIN_CHECKS
        assert(i == 0 || s_log_cache[(i - 1) / 2].generation < s_log_cache[i].generation);
#endif
        if (s_log_cache[i].tag == tag) {
            break;
        }
    }
```

This should be a `(strncmp( s_log_cache[i].tag, tag) ==0 )` like the rest of log.c

igrr, 2022-03-14:
Thanks for bringing this issue up @pickaxprograms!

I think the way the code was supposed to work was:

* whenever s_log_level_get_and_unlock is called, look if the cache contains the same tag pointer (`get_cached_log_level`, which is fast)
  * if not found, fall back to strcmp-ing the tag with every entry in `s_log_tags` (`get_uncached_log_level`, which is slower)
  * also add the tag to cache via `add_to_cache`. This should create one entry for each pointer, but since different pointers may point to the same string (in `strcmp` sense), there could be duplicates. All of them should have the same log level value, though.
* whenever esp_log_level_set is called, update the log level in `s_log_tags` and also update any value in cache which matches (in strcmp sense) the provided tag. So even if the pointer passed to esp_log_level_set is not the same as previously recorded in the cache, the log level for the same tags in cache should still be updated.

Since you observe this issue, there's obviously either a flaw in the above logic or the code doesn't actually do what I think it should. We'll have to look into the code in a bit more detail and try to find the issue.

That said on the two issues you note:

* a tag may point to a string on the stack; after the string goes out of scope the tag in the cache would point to some invalid memory
* multiple source files may use the same tag, resulting in duplicate cache entries (Note, according to my understanding this should still work correctly, in the sense that esp_log_level_set should update all such entries; that might be something to fix if it doesn't work)

I think both are not likely to happen if you consider regular ESP_LOG usage, which is:

* to define TAG statically in each translation unit. Usually you'd have a unique tag for each file / module.
* pass the static TAG to ESP_LOGX macros.

Another thing, I recall seeing a comment by someone who pointed out that the whole tag cache solution in esp_log is probably a premature optimization. I can't find the comment right now to give you a link. Indeed in case when we log at least something, a few extra hundred cycles doing strcmp have very little performance impact compared to the thousands of cycles we spend waiting for UART to emit the log output. So perhaps the smarter thing to do would be to remove all the complex cache logic and leave a simple linked list of log levels in place, with `strcmp` to find find the right tag... Given that most users wouldn't be calling `esp_log_level_set` too many times, the list is unlikely to get very long.

pickaxprograms, 2022-03-14:
I uncovered this by wanting to have a command line debug routine to get/set the log level for a passed in tag. When esp_log_level_get() is called with a previously unknown tag, it is added to the s_log_cache with just a pointer to the string that was passed in. If that memory goes out of scope, then s_log_cache will be pointing to memory that is something completely different.

moefear85, 2022-05-07:
could this possibly cause system crashes? I face crashes that happen immediately on boot, but only if I enable esp_logX statements by calling esp_log_level_set. The dump contains alot of lines, implicating locks and vfs and uart and so on. it makes it very hard to debug when the debug statements themselves are triggering a crash (or causing a double crash). I can't upload logs right now becuase it's on another pc.

KonstantinKondrashov, 2024-07-18:
The master now has new log Kconfigs that can be used for this use case - https://github.com/espressif/esp-idf/blob/master/components/log/Kconfig.level_settings#L30-L72. The cache can be disabled and use only the linked list. This approach compares the whole strings of log tags for finding the log level.

closed 2024-07-18 by KonstantinKondrashov

zzzh, 2024-12-06:
> The master now has new log Kconfigs that can be used for this use case - https://github.com/espressif/esp-idf/blob/master/components/log/Kconfig.level_settings#L30-L72. The cache can be disabled and [...]

Sounds good, does this apply to version 4.4.2?

KonstantinKondrashov, 2024-12-06:
No, according to our policy, we backport only fixes not new features. Sorry.