espressif/esp-idf #8568: Are

lukedukeus, 2022-03-14:
## Environment

- Development Kit:      ESP32-C3-DevkitM-1
- Kit version v3/ rev3
- Module or chip used:  ESP32-C3
- IDF version 4.4
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: VScode + esp idf addon
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.
I am setting up wifi like this:
```
#define WIFI_SSID "ESP32 C3"

    wifi_config_t wifi_config = {
        .ap = {
            .ssid = WIFI_SSID,
            .ssid_len = strlen(WIFI_SSID),
            .password = WIFI_PASS,
            .max_connection = MAX_STA_CONN,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
 
    };
    if (strlen(WIFI_PASS) == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
```
this starts up a softAP which works fine. Later on, I want to join a wifi network while still in APSTA mode, so I call:
```
wifi_config_t config;

esp_wifi_get_config(WIFI_IF_STA, &config);
esp_wifi_get_config(WIFI_IF_AP, &config);

ESP_LOGI(TAG, "AP is'%s'", config.ap.ssid);
ESP_LOGI(TAG, "STA is'%s'", config.sta.ssid);
```

My hope is that I can write to the sta prop, then call `esp_wifi_connect()`, while still serving a captive portal through softAP. However, this returns:
```
I (637) WIFI_CONNECT: AP is'ESP32 C3'
I (637) WIFI_CONNECT: STA is'ESP32 C3'
```

If I call only `esp_wifi_get_config(WIFI_IF_STA, &config);` then the sta prop is accurate, and if I call only `esp_wifi_get_config(WIFI_IF_AP, &config);` then the ap prop is accurate, but if I call both, then ap and sta have the same info. 

It seems like `esp_wifi_get_config()` overwrites the other field. Is it even possible to connect to another AP while the esp is serving a captive portal and is in APSTA mode? The idea is that the user can choose a wifi network and connect to it all through softAP/ captive portal.

atanisoft, 2022-03-14:
The `wifi_config_t` type is actually a union as seen in https://github.com/espressif/esp-idf/blob/5963de1caf284b14ddfed11e52730a55e3783a3d/components/esp_wifi/include/esp_wifi_types.h#L261-L264. Due to this you should use two different local variables in your call to `esp_wifi_get_config`, one for AP and the other for STA.

lukedukeus, 2022-03-14:
I realized that it was two other types, and using two different variables does fix the problem, but I still think that 

`esp_wifi_get_config(WIFI_IF_STA, &config);` should only fill the STA config
`esp_wifi_get_config(WIFI_IF_AP, &config);` should only fill the AP config

atanisoft, 2022-03-14:
> but I still think that
> 
> `esp_wifi_get_config(WIFI_IF_STA, &config);` should only fill the STA config
> `esp_wifi_get_config(WIFI_IF_AP, &config);` should only fill the AP config

That would be impossible due to `wifi_config_t` being a union of the two underlying structure, as a union there is *ONE* block of memory that can be either type in the union but can not be both simultaneously *AND* retain unique data in both types.

lukedukeus, 2022-03-14:
Makes sense. I was expecting that it would leave the other half null

closed 2022-03-14 by lukedukeus