espressif/esp-idf #8579: Can Esp32 support 802.11u interworking information element?

zhm-nico-nico-ni, 2022-03-16:
Under the wifi Ap mode.
The idf support edit vendor information element inside beacon,  I think it can support interworking information element too.

Can esp open this interface?

Thanks

MaxwellAlan, 2022-03-23:
Hi @zhm-nico-nico-ni , thanks for your feature request.
Can you share some more detail about this feature, such as, 
- Which specific scenario ?
- What interworking information elements you may need?
- etc, you may care about.
These information help us to evaluate this feature.
Thanks.

zhm-nico-nico-ni, 2022-03-23:
Thinks @MaxwellAlan,

The ESP may can support carplay as ap router.  but which need Interworking IE.

I found the wpa_supplicant is open source inside idf,  can directly add any Information Element there.  (Under 752 size).
What I want do show in the broadcast beacon.
Because this is not **standard** operation, i feel something wire, but can't tell where wrong. 

Hope esp can support the whole 802.11u stack. 

Is this possible?

sagb2015, 2022-05-03:
@zhm-nico-nico-ni - We do not support 802.11u as of now. We will consider this in future if enough demand is there.

kapilkedawat, 2024-03-20:
Apple car play devices need apple certification, its unlikely that adding one IE will do it. We won't be taking up this feature unless there is enough demand for 802.11u.

closed 2024-03-20 by Alvin1Zhang