espressif/esp-idf #8580: system reboot itself while reporting "gap_event_handler:157: fail reason = 0x66"

EWang78, 2022-03-16:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 3.3
    // v3.2-dev-1148-g96cd3b75c
- Build System:        :idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type:PowerShell
- Using an IDE?: NO
- Power Supply:         external 5V

## Problem Description

Hello Dear, 
I encountered a problem. Esp32 is often restarted itself
[ble_2022-03-09_17_29_48.txt](https://github.com/espressif/esp-idf/files/8258672/ble_2022-03-09_17_29_48.txt)
. The serial port prompts the following error when restarting：
what is the "fail reason = 0x66"? and when it will be reported?  Is this the key reason for the rebooting issue? and how can I solve this issue?

[ Debug]: 2022-03-09 17:30:11, hidd_event_callback:100: start_advertising......
 disc Mac:7C,89,CB,96,C0,B0,[ Debug]: 2022-03-09 17:30:11, hidd_event_callback:110: ESP_HIDD_EVENT_BLE_DISCONNECT
[ Debug]: 2022-03-09 17:30:11, APPMqtt_evtTask:203: HT_BLE_DISCONNECTED_BIT
[ Debug]: 2022-03-09 17:30:11, gap_event_handler:141: remote BD_ADDR: 7c89cb96c0b0
[ Debug]: 2022-03-09 17:30:11, gap_event_handler:142: address type = 1
[ Debug]: 2022-03-09 17:30:11, gap_event_handler:143: pair status = fail
[ble_2022-03-09_17_29_48.txt](https://github.com/espressif/esp-idf/files/8258680/ble_2022-03-09_17_29_48.txt)

[ Debug]: 2022-03-09 17:30:11, gap_event_handler:157: fail reason = 0x66
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c46c  PS      : 0x00060430  A0      : 0x800ebfc4  A1      : 0x3ffe8e40  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000148  A5      : 0x00000000  
A6      : 0x0000004b  A7      : 0x00000014  A8      : 0x800ebf77  A9      : 0x3ffe8e10  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ffe8ea0  A13     : 0x00000013  
A14     : 0x000000ce  A15     : 0x0000006a  SAR     : 0x0000000c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000013  

Backtrace:0x4000c469:0x3ffe8e40 0x400ebfc1:0x3ffe8e50 0x4010de81:0x3ffe8e70 0x4010dfed:0x3ffe8e90 0x400e73eb:0x3ffe8ef0 0x400e76a5:0x3ffe8f10 0x4010e0ae:0x3ffe8f30 0x400f2569:0x3ffe8f90 0x400f2f08:0x3ffe8fb0 0x400f3256:0x3ffe8fd0 0x40106e65:0x3ffe8ff0 0x40093d0d:0x3ffe9010

ELF file SHA256: 239023e171bd7cea

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6900
load:0x40078000,len:14308
load:0x40080400,len:3716
entry 0x40080680
I (27) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (27) boot: compile time 18:33:52
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
[ble_2022-03-09_17_29_48.txt](https://github.com/espressif/esp-idf/files/8258678/ble_2022-03-09_17_29_48.txt)

AxelLin, 2022-03-16:
> * IDF version (run `git describe --tags` to find it): 3.3

3.3 is already EOL, do you want to try reproducing the issue with newer sdk version?

EWang78, 2022-03-16:
which version is the latest?

AxelLin, 2022-03-16:
> which version is the latest?

https://github.com/espressif/esp-idf/releases/tag/v4.4

esp-cjh, 2024-02-27:
@EWang78 Does this problem still exist with the latest IDF version?

Alvin1Zhang, 2024-09-06:
Thanks for reporting, will close now, feel free to reopen with more updates.

closed 2024-09-06 by Alvin1Zhang