espressif/esp-idf #8581: Unable to perform advanced ota example

MaazSk, 2022-03-16:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-939-g952b4d848f
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description
I am trying to implement the advanced OTA example of esp-idf. I have saved by .bin file on google drive and the certificates for the drive were taken from the following command.
```openssl s_client -showcerts -connect www.drivegoogle.com:443 </dev/null```

But I am facing the error as below

> I (5060) esp_https_ota: Writing to partition subtype 16 at offset 0x120000
> E (5060) esp_https_ota: Complete headers were not received
> E (5070) wifi station: esp_https_ota_read_img_desc failed
> E (5080) wifi station: ESP_HTTPS_OTA upgrade failed
> 

I tried by hosting blink code also on google drive but I am getting the same error.

negativekelvin, 2022-03-16:
Google does not usually give direct links to the file, are you sure you are using a direct link?

MaazSk, 2022-03-16:
@negativekelvin,  yes I converted the link to get the direct download link

MaazSk, 2022-03-16:
@negativekelvin, When I was debugging I got the following message

> D (10560) HTTP_CLIENT: HEADER=Expires:Mon, 01 Jan 1990 00:00:00 GMT
> D (10570) HTTP_CLIENT: HEADER=Date:Wed, 16 Mar 2022 06:30:54 GMT

I can understand something has expired but what I don't know can you please help me out.

negativekelvin, 2022-03-16:
That is normal, Google pages always use that date

MaazSk, 2022-03-16:
Yes, I googled the same and came to know about it.

shubhamkulkarni97, 2022-03-17:
Hi @MaazSk, google is now returning HTTP 303 status code for redirecting the request to a new URL. This was not the case previously.

HTTP 303 is not handled in esp_http_client and esp_https_ota. Please try attached patch and let me know if it fixes your issue.
[0001-esp_http_client-Add-new-status-code-303.patch.zip](https://github.com/espressif/esp-idf/files/8283032/0001-esp_http_client-Add-new-status-code-303.patch.zip)

Thanks,
Shubham

MaazSk, 2022-03-17:
Hi @shubhamkulkarni97, I will try the patch and will let you know,

Thank you!

MaazSk, 2022-03-17:
Hi @shubhamkulkarni97 I was trying to apply the patch I am getting the following error

> maaz@maaz:~/esp/esp-idf$ git apply --check 0001-esp_http_client-Add-new-status-code-303.patch
> warning: components/esp_http_client/esp_http_client.c has type 100755, expected 100644
> error: patch failed: components/esp_http_client/esp_http_client.c:821
> error: components/esp_http_client/esp_http_client.c: patch does not apply
> warning: components/esp_http_client/include/esp_http_client.h has type 100755, expected 100644
> warning: components/esp_https_ota/src/esp_https_ota.c has type 100755, expected 100644
>

MaazSk, 2022-03-17:
Hi @shubhamkulkarni97, 

It is working now. I will just test it a little more and will let you know if there are any issues.

Thank you,
Maaz

MaazSk, 2022-03-17:
@shubhamkulkarni97, the patch is working. 

Thank you!

closed 2022-03-17 by MaazSk