espressif/esp-idf #8582: ESP32-S3 SDSPI timeout

mishaturnbull, 2022-03-16:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- Kit version (for WroverKit/PicoKit/DevKitC): `chip revision 0` printed on boot
- Module or chip used: ESP32-S3-WROOM-1
- IDF version `v5.0-dev-2047-g13f7dbad2d` (pulled earlier today, at time of writing)
- Build System: idf.py
- Compiler version `xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0`
- Operating System: Linux (debian-based)
- Using an IDE?: Vim (no)
- Power Supply: USB to the devkit, external bench PSU at 3.30V for the microSD card

## Problem Description

I am unable to get a MicroSD card to function correctly using SDSPI mode.  The card is physically connected via the following pins:
* MOSI: `GPIO11`
* MISO: `GPIO13`
* SCLK: `GPIO12`
* CS: `GPIO10`

The breakout board for the card adapter includes 10kohm pullup resistors on those lines.

### Expected Behavior

Running the sdspi example code (example/storage/sd_card/sdspi) results in a successful mounting and prints some information about file I/O.

### Actual Behavior

`esp_vfs_fat_sdspi_mount` is unable to mount the card.  Full `idf.py monitor` output:

```                                                        
E (37188) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107                                     
D (37198) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x107                                                                    
E (37198) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).                                                                                
I (37208) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0                                         
E (37218) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
```

### Steps to reproduce

1. Pull & install latest ESP-IDF master branch
2. Update pins on SDSPI example via `idf.py menuconfig`
3. Compile example code: `idf.py build`
4. Run: `idf.py -p /dev/ttyUSB0 flash monitor`

### Code to reproduce this issue

See https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdspi -- ran without modifications (other than the pin changes in menuconfig).

## Debug Logs

```
I (14928) example: Initializing SD card                             
I (14928) example: Using SPI peripheral                                                                                                 
D (30668) gdma: new group (0) at 0x3fce0878         
D (30678) gdma: new pair (0,0) at 0x3fce08b4
D (30678) gdma: new tx channel (0,0) at 0x3fce0850
D (30678) gdma: new rx channel (0,0) at 0x3fce08d4
D (30678) spi: SPI2 use iomux pins.         
I (30688) example: Mounting filesystem                        
D (36978) sdspi_host: sdspi_host_init_device: SPI2 cs=10 cd=-1 wp=-1                                                                    
D (36978) intr_alloc: Connected src 21 to int 13 (cpu 0)
D (36978) spi_hal: eff: 400, limit: 80000k(/0), 0 dummy, -1 delay
D (36978) spi_master: SPI2: New device added to CS5, effective clock: 400kHz                                                            
I (36988) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0                                         
D (36998) spi_master: device5 locked the bus                                                                                            
D (36998) spi_master: device5 release bus                                                                                               
D (37008) sdspi_host: sdspi_host_start_command: cmd=52 error=0x107                                                                      
D (37008) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x107                                                                                                                                                                                                                       
D (37018) spi_master: device5 locked the bus                        
D (37018) spi_master: device5 release bus                                                                                               
D (37048) spi_master: device5 locked the bus                     
D (37048) spi_master: device5 release bus                                                                                               
D (37068) spi_master: device5 locked the bus                                                                                            
D (37068) spi_master: device5 release bus                                                                                               
D (37068) sdmmc_sd: SDHC/SDXC card                                                                                                      
D (37068) spi_master: device5 locked the bus                                                                                            
D (37068) spi_master: device5 release bus                                                                                               
I (37068) sdspi_transaction: cmd=5, R1 response: command not supported                                                    
D (37078) sdmmc_cmd: cmd=5, sdmmc_req_run returned 0x106                                                                                
D (37088) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x106; not IO card                                                      
D (37088) spi_master: device5 locked the bus           
D (37098) spi_master: device5 release bus                           
D (37098) sdspi_host: data CRC set=1                                                                                                                                                                                                                                            
D (37108) spi_master: device5 locked the bus                        
D (37108) spi_master: device5 release bus                           
D (37118) spi_master: device5 locked the bus                        
D (37118) spi_master: device5 release bus                           
D (37128) spi_master: device5 locked the bus                        
D (37128) spi_master: device5 release bus                           
D (37128) sdspi_host: sdspi_host_start_command: cmd=55 error=0x107  
D (37138) sdmmc_cmd: cmd=55, sdmmc_req_run returned 0x107           
D (37138) spi_master: device5 locked the bus                       
D (37148) spi_master: device5 release bus
D (37148) sdspi_host: sdspi_host_start_command: cmd=55 error=0x107  
D (37158) sdmmc_cmd: cmd=55, sdmmc_req_run returned 0x107
D (37158) spi_master: device5 locked the bus
D (37168) spi_master: device5 release bus
D (37168) sdspi_host: sdspi_host_start_command: cmd=55 error=0x107  
D (37178) sdmmc_cmd: cmd=55, sdmmc_req_run returned 0x107                                                                                                                                                                                                                       
D (37178) sdmmc_cmd: sdmmc_send_cmd_send_op_cond: sdmmc_send_app_cmd err=0x107                                                          
E (37188) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107                                     
D (37198) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x107                                                                    
E (37198) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).                                                                                
I (37208) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0                                         
E (37218) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8259612/sdkconfig.txt)

I can get a picture of the wiring if needed, but it may include some other stuff on the breadboard.  Let me know!

mishaturnbull, 2022-03-16:
Just realized that commit ID won't make much sense -- that's with a patch file applied from over here: https://github.com/espressif/esp-idf/issues/8521.

The last commit from ESP-IDF's master branch is 5963de1caf284b14ddfed11e52730a55e3783a3d .

Sorry about that!

igrr, 2022-03-16:
From the log it looks like there is no reply from the SD card even to the first initialization command. Failure at such early stage usually means that there is a physical connection issue between ESP and the card. Please check the connections between ESP and the card. Please try probing the signals going between the ESP and the card using a logic analyser. Make sure MOSI signal is getting to the card and MISO from the card is getting to the ESP. If you see nothing on MISO line, please check the card power supply. Also if you are using a breakout board, make sure it's not doing any voltage level conversion.

mishaturnbull, 2022-03-16:
[image]

(The analog measurements are there since I was looking into clock timing -- read somewhere that if the rise/fall time of the MOSI/MISO lines isn't fast enough relative to the clock line rise/fall, there can be issues... but I'm not enough of an EE to fully understand what I'm looking at)

Hey @igrr , thanks for the quick response!  I actually did check that already, and have attached one of the screenshots from it.  This wasn't from the example code, rather from running some of my code but it made a call to the same `esp_vfs_fat_sdspi_mount` function.

Will check into the breakout board to be absolutely certain in the morning.

igrr, 2022-03-16:
The data from SD card looks okay, but the 1.8V high level on MISO is a bit low for 3.3V logic of the ESP. It might indeed be the case that the breakout board is doing some logic level translation. Please check the breakout board for any switches or jumpers which may correspond to voltage level selection. You need to set it up so that the breakout board uses 3.3V logic levels on both sides — ESP and the card.

closed 2022-03-16 by mishaturnbull

mishaturnbull, 2022-03-19:
Hey @igrr ,

Following up - you hit the nail on the head.  The breakout board was replace with one of these: https://learn.adafruit.com/adafruit-microsd-spi-sdio/overview with which the SDSPI example code works flawlessly.  I suppose if I have more problems, it's purely my fault :)

Thanks!