espressif/esp-idf #8584: vTaskDelay inaccurate

krupis, 2022-03-16:
I would like to understand how does vTaskDelay work exactly. I am having issues with time sensitive tasks. For my project, I need to create a task that would repeat itself precisely every 2 seconds. +-1ms is acceptable but not more than that.

I have configured the tick rate to be 1000hz which should allow me to achieve 1ms accuracy.

**EXPERIMENT1**
```
static void TIMER_TEST_TASK(void* parameters)
{       
    for(;;)
    {   
        int64_t task_execution_start = esp_timer_get_time()/1000; // get current time required for the task delay
        ESP_LOGW("TEST","time in miliseconds=%lli",task_execution_start);
        

        ESP_LOGW("TEST","PRINT STATEMENT FOR ADDITIONAL DELAY ");
        ESP_LOGW("TEST","PRINT STATEMENT FOR ADDITIONAL DELAY ");
        ESP_LOGW("TEST","PRINT STATEMENT FOR ADDITIONAL DELAY ");
        ESP_LOGW("TEST","PRINT STATEMENT FOR ADDITIONAL DELAY ");
        ESP_LOGW("TEST","PRINT STATEMENT FOR ADDITIONAL DELAY ");
        
        int64_t task_execution_end = esp_timer_get_time()/1000; 
        int64_t timestamp_difference = (task_execution_end - task_execution_start);
        ESP_LOGW("TEST","task execution end = %lli ms",task_execution_end);
        ESP_LOGW("TEST","timestamp difference = %lli \n",timestamp_difference);

        vTaskDelay(2000/portTICK_PERIOD_MS); 
    }
}
```
I have created a simple task with some print statements. I capture a time when the task is started and when it is finished. The are the logs:
```
W (3783) TEST: time in miliseconds=3073
W (3787) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3792) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3798) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3804) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3809) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3815) TEST: task execution end = 3105 ms
W (3820) TEST: timestamp difference = 32 

W (5825) TEST: time in miliseconds=5115
W (5825) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5825) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5829) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5835) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5841) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5846) TEST: task execution end = 5137 ms
W (5851) TEST: timestamp difference = 22 

W (7856) TEST: time in miliseconds=7146
W (7856) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (7856) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (7860) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (7866) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (7872) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (7877) TEST: task execution end = 7168 ms
W (7882) TEST: timestamp difference = 22 

W (9887) TEST: time in miliseconds=9177
W (9887) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (9887) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (9891) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (9897) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (9903) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (9908) TEST: task execution end = 9199 ms
W (9913) TEST: timestamp difference = 22 

W (11918) TEST: time in miliseconds=11208
W (11918) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (11918) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
[... 317 lines cut ...]
W (29926) TEST: timestamp difference = 22
```
Lets look at the first 2 blocks of logs:
```
W (3772) TEST: time in miliseconds=3061
W (3776) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3781) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3787) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3793) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3798) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (3804) TEST: task execution end = 3093 ms
W (3809) TEST: timestamp difference = 32 

W (5782) TEST: time in miliseconds=5071
W (5782) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5782) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5786) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5792) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5798) TEST: PRINT STATEMENT FOR ADDITIONAL DELAY 
W (5803) TEST: task execution end = 5093 ms
W (5808) TEST: timestamp difference = 22 
```
The task started at 3061ms and ended at 3093. Since I compensate for the task execution time, the next task should start precisely at 5061ms but that is not the case. I also see 10ms error.

Perhaps I am not fully understanding how the vTaskDelay works. Could someone help me out here?

0xjakob, 2022-03-16:
@krupis vTaskDelay is not very accurate. It depends on other tasks getting scheduled, interrupts, etc. If you want to be more accurate, you should use an actual timer:
* If you have some longer-running work to do, probably the [esp_timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#using-esp-timer-apis) component is best, it schedules work in it's own task according to a hardware timer. It is still some imprecise due to the scheduling of callbacks inside a task, but you won't have accumulating latency anymore.
* If you want to have the least latency, you should probably use the [General Purpose Timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gptimer.html). I haven't used it myself but according to the description, you register user interrupts with it, hence you would have instant response but have a lot of restrictions because your code runs inside an ISR.

krupis, 2022-03-17:
> @krupis vTaskDelay is not very accurate. It depends on other tasks getting scheduled, interrupts, etc. If you want to be more accurate, you should use an actual timer:
>
> * If you have some [...]

Thank you for pointing me in the right direction. I have copy and paste all my task functions inside the timer callback function and everything seemed to work fine at first. I have created a timer as following:

```
        const esp_timer_create_args_t periodic_timer_args = {
            .callback = &periodic_timer_callback,
            /* name is optional, but may help identify the timer when debugging */
            .arg = (void*) this,
            .name = "periodic"
    };
    esp_timer_handle_t periodic_timer;
    ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 2000000));
```
 I could see right away that the timestamp is much more accurate. And the my callback function is started every 2 seconds.

I would like to ask a few questions

1. Since using this timer and callback method is much more accurate, what is the difference between this method and using RTOS tasks? What are the drawbacks of using this periodic timer ?
2. Can I assign priorities to timer? What if I have multiple timers that all need to fire every 2 seconds
3.  Inside my timer callback, if certain scenario is met, I want to create an rtos task. For example. after 60 seconds upon power up, I want to start led_blink_task. I have noticed that tasks are not created when xTaskCreate is being called from within the timer_callback. Is there any specific reason?

tore-espressif, 2022-06-02:
1. The differences are described here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html
2. With `esp_timer` no, you can't. `esp_timer` uses one HW timer for all events. If you require multiple low-latency timers, you need to setup multiple HW timers ([General Purpose Timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gptimer.html)) and assign them different interrupt priorities.
3. In general, the timer callback should be fast - just passing signals to lower priority tasks that will do what you need. Anyway, it should be possible to create a task within timer callback, I'll have a look

Dazza0, 2023-03-08:
@krupis FreeRTOS delay accuracy can be affected by the FreeRTOS tick resolution, and the priority of the task executing the delay. For applications that require very accurate delays, please try esp_timer or one of the General Purpose HW timers.

closed 2023-03-08 by Dazza0