espressif/esp-idf #8587: MQTT with no connectivity

apiesse, 2022-03-16:
## Environment

- Development Kit: esp32-devkitc-v4
- IDF version: v4.4
- Build System: CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System: Windows
- (Windows only) environment type: idf.py
- Using an IDE?: Yes, Eclipse 2021-06 with esp-idf plugin

## Problem Description
During testing of one of our applications we found a strange behaviour where the esp-idf MQTT client does not detect internet connectivity absence if the upstream connection is detached from the WiFiAP the esp32 is connected to.

We managed to reproduce the situation on different esp-idf versions (3.3.6, 4.2, 4.4) and attached a slightly modified MQTT over TCP example.
Basically we configure the MQTT keepalive to 60 seconds and send data every 10 seconds with QoS0: in this situation the PING_REQ is never sent because the data send period is lower than the keepalive period, but the underlying sockets do not detect the lack of connectivity.

The result is a system thinking it's sending data when nothing is going to the MQTT broker actually. If we change to QoS1, of course, the situation is detected as soon as the first data packet is sent after upstream connection is detached because no acknowledgement from the broker is received

[mqtt_tcp.zip](https://github.com/espressif/esp-idf/files/8260827/mqtt_tcp.zip)
.

apiesse, 2022-03-16:
We've also tested enabling [CONFIG_MQTT_SKIP_PUBLISH_IF_DISCONNECTED](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mqtt-skip-publish-if-disconnected), but nothing changed.

euripedesrocha, 2022-03-21:
Hi @apiesse, I was able to reproduce the issue, but the disconnection is detected after some time. It just takes longer to detect. 
We are investigating solutions to improve the behavior. Could you elaborate on how this affects your tests?

apiesse, 2022-03-22:
Probem is this behaviour affects our application works because thinking we're connected we perform tasks which shiuld not run.

euripedesrocha, 2022-03-22:
@apiesse, the solution we found so far is to move the place where we update the keep alive for MQTT, making the ping message periodic. In our first tests, this reduced the time of disconnection detection. 

Even with this modification there will be a period of time before the disconnection detection, have you considered to have messages with higher QoS in order to guarantee your system's integrity?

apiesse, 2022-03-23:
@euripedesrocha we're using QoS1 for critical messages and QoS0 for no-critical messages because we're on a mobile connection and we're trying to save data. That's the reasone why I'm reporting the behaviour as a problem.
As I said in my initial report, QoS1 is keeping the time needed to detect the disconnetion really low.

For what I understand, moving to a periodic ping is "against" the MQTT specifications as these packets should be sent if there is no data for the keepalive period.
Apart from this, I'd expect the underlying TCP socket to detect the lack of connectivity even when sending the QoS0 messages. Did you manage to investigate this as well?

euripedesrocha, 2022-03-23:
@apiesse Thanks for clarifying your usage scenario.

From our understanding, it's not against the spec:

"It is the responsibility of the Client to ensure that the interval between Control Packets being sent does not exceed the Keep Alive value. In the absence of sending any other Control Packets, the Client MUST send a PINGREQ Packet [MQTT-3.1.2-23]." 

I'm still trying to avoid sending PINGREQ when not necessary, but for now, this is a quick fix for the issue. 

From the scenario you are working on, WIFI is available but no internet connection, the behavior is correct due to TCP nature. 

You may want to look into LWIP configuration to find the best configuration for your needs.

apiesse, 2022-03-24:
@euripedesrocha Any suggestion on what I can tune in LWIP configuration to have the TCP sockets reactive in these situations?

euripedesrocha, 2022-03-24:
@apiesse check:
TCP high speed retransmission - `LWIP_TCP_HIGH_SPEED_RETRANSMISSION`
`LWIP_TCP_MAXRTX`  - Maximum retransmission of data segments
`LWIP_TCP_RTO_TIME` -  Default TCP rto time

apiesse, 2022-04-12:
@Alvin1Zhang tried to play a little with the parameters suggested by @euripedesrocha , but the behaviours is still the same.
Unless I use QoS>0, the underlying TCP socket does not detect the loss of communication and the MQTT client still thinks it's connected.

euripedesrocha, 2022-04-12:
@apiesse could you please clarify if the disconnecting isn't detected, or it just takes longer than you were expecting?

I observed in the tests I made that the disconnection was detected after some time and this is correct from the configuration and scenario you described, due to TCP behavior. 

As I mentioned, to improve this in the MQTT level we changed the strategy on PINGREQ/PINGRESP processing and the modification will be merged soon.

I made a patch with the modifications made, could you apply it to esp-mqtt and test if the situation is improved on your application?

[0001-Changes-the-moment-we-update-keepalive_tick.zip](https://github.com/espressif/esp-idf/files/8471889/0001-Changes-the-moment-we-update-keepalive_tick.zip)

AxelLin, 2022-05-06:
@euripedesrocha 
Your fix has been in esp-mqtt for quite long time, but the esp-idf does not update the submodule.
When will the fix be available in esp-idf?

euripedesrocha, 2022-05-17:
@AxelLin, I missed your question. 
Yes, the fix will be available soon.

euripedesrocha, 2022-06-22:
@AxelLin the fix is about to be merged in the next couple days.

Alvin1Zhang, 2022-07-12:
Thanks for reporting, fix is available 33fccf6405a302e4f54913b55d7bd6299cfefd72, feel free to reopen.

closed 2022-07-12 by Alvin1Zhang

AxelLin, 2022-07-13:
@euripedesrocha 
This issue was reported on v4.4, but the fix is in master.
So I'm wondering if the stable branches also need fix?

euripedesrocha, 2022-07-20:
Hi @AxelLin, sorry for the late response. This issue triggered the modifications in esp-mqtt that are an improvement on the process but are not a direct fix for this issue, since it is related to the time it takes to detect a disconnection from MQTT client. From the tests, the disconnection is detected, it only takes more time than the application requires. After discussion, we decided to not port the modifications to older branches at the moment.

kriegste, 2023-05-23:
Also, when WiFi got disconnected and I call esp_mqtt_client_destroy() these misleading errors appear:

> MQTT_CLIENT: Poll read error: 119, aborting connection
> MQTT_CLIENT: Writing failed: errno=0
> MQTT_CLIENT: Error sending disconnect message